homieiot/homie-esp8266 #365: Incompatibility with latest ArduinoJson 5.11

chiefymuc, 2017-07-01:
I updated all my Arduino libraries and got compile errors:

`C:\Users\David\Documents\Arduino\libraries\homie-esp8266-2.0.0-beta.1\src\Homie\Boot\BootConfig.cpp: In member function 'void HomieInternals::BootConfig::_generateNetworksJson()':

C:\Users\David\Documents\Arduino\libraries\homie-esp8266-2.0.0-beta.1\src\Homie\Boot\BootConfig.cpp:156:147: error: use of deleted function 'ArduinoJson::DynamicJsonBufferBase<ArduinoJson::DefaultAllocator>::DynamicJsonBufferBase(const ArduinoJson::DynamicJsonBufferBase<ArduinoJson::DefaultAllocator>&)'

   DynamicJsonBuffer generatedJsonBuffer = DynamicJsonBuffer(JSON_OBJECT_SIZE(1) + JSON_ARRAY_SIZE(_ssidCount) + (_ssidCount * JSON_OBJECT_SIZE(3)));  // 1 at root, 3 in childrend

                                                                                                                                                   ^

In file included from C:\Users\David\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.hpp:10:0,

                 from C:\Users\David\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:10,

                 from C:\Users\David\Documents\Arduino\libraries\homie-esp8266-2.0.0-beta.1\src\Homie\Boot\BootConfig.hpp:10,

                 from C:\Users\David\Documents\Arduino\libraries\homie-esp8266-2.0.0-beta.1\src\Homie\Boot\BootConfig.cpp:1:

C:\Users\David\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/DynamicJsonBuffer.hpp:36:7: note: 'ArduinoJson::DynamicJsonBufferBase<ArduinoJson::DefaultAllocator>::DynamicJsonBufferBase(const ArduinoJson::DynamicJsonBufferBase<ArduinoJson::DefaultAllocator>&)' is implicitly deleted because the default definition would be ill-formed:

 class DynamicJsonBufferBase

       ^

C:\Users\David\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/DynamicJsonBuffer.hpp:36:7: error: use of deleted function 'ArduinoJson::JsonBufferBase<ArduinoJson::DynamicJsonBufferBase<ArduinoJson::DefaultAllocator> >::JsonBufferBase(const ArduinoJson::JsonBufferBase<ArduinoJson::DynamicJsonBufferBase<ArduinoJson::DefaultAllocator> >&)'

In file included from C:\Users\David\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/DynamicJsonBuffer.hpp:10:0,

                 from C:\Users\David\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.hpp:10,

                 from C:\Users\David\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:10,

                 from C:\Users\David\Documents\Arduino\libraries\homie-esp8266-2.0.0-beta.1\src\Homie\Boot\BootConfig.hpp:10,

                 from C:\Users\David\Documents\Arduino\libraries\homie-esp8266-2.0.0-beta.1\src\Homie\Boot\BootConfig.cpp:1:

C:\Users\David\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/JsonBufferBase.hpp:24:7: note: 'ArduinoJson::JsonBufferBase<ArduinoJson::DynamicJsonBufferBase<ArduinoJson::DefaultAllocator> >::JsonBufferBase(const ArduinoJson::JsonBufferBase<ArduinoJson::DynamicJsonBufferBase<ArduinoJson::DefaultAllocator> >&)' is implicitly deleted because the default definition would be ill-formed:

 class JsonBufferBase : public JsonBuffer {

       ^

C:\Users\David\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/JsonBufferBase.hpp:24:7: error: use of deleted function 'ArduinoJson::JsonBuffer::JsonBuffer(const ArduinoJson::JsonBuffer&)'

In file included from C:\Users\David\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Deserialization/JsonParser.hpp:10:0,

                 from C:\Users\David\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/JsonBufferBase.hpp:10,

                 from C:\Users\David\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/DynamicJsonBuffer.hpp:10,

                 from C:\Users\David\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.hpp:10,

                 from C:\Users\David\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:10,

                 from C:\Users\David\Documents\Arduino\libraries\homie-esp8266-2.0.0-beta.1\src\Homie\Boot\BootConfig.hpp:10,

                 from C:\Users\David\Documents\Arduino\libraries\homie-esp8266-2.0.0-beta.1\src\Homie\Boot\BootConfig.cpp:1:

C:\Users\David\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Deserialization/../JsonBuffer.hpp:38:7: note: 'ArduinoJson::JsonBuffer::JsonBuffer(const ArduinoJson::JsonBuffer&)' is implicitly deleted because the default definition would be ill-formed:

 class JsonBuffer : Internals::NonCopyable {

       ^

In file included from C:\Users\David\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Deserialization/../JsonBuffer.hpp:14:0,

                 from C:\Users\David\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Deserialization/JsonParser.hpp:10,

                 from C:\Users\David\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/JsonBufferBase.hpp:10,

[... 34 lines cut ...]
       ^

C:\Users\David\Documents\Arduino\libraries\homie-esp8266-2.0.0-beta.1\src\Homie\Boot\BootConfig.cpp: In member function 'void HomieInternals::BootConfig::_onDeviceInfoRequest()':

C:\Users\David\Documents\Arduino\libraries\homie-esp8266-2.0.0-beta.1\src\Homie\Boot\BootConfig.cpp:262:225: error: use of deleted function 'ArduinoJson::DynamicJsonBufferBase<ArduinoJson::DefaultAllocator>::DynamicJsonBufferBase(const ArduinoJson::DynamicJsonBufferBase<ArduinoJson::DefaultAllocator>&)'

   DynamicJsonBuffer jsonBuffer = DynamicJsonBuffer(JSON_OBJECT_SIZE(5) + JSON_OBJECT_SIZE(2) + JSON_ARRAY_SIZE(numNodes) + (numNodes * JSON_OBJECT_SIZE(2)) + JSON_ARRAY_SIZE(numSettings) + (numSettings * JSON_OBJECT_SIZE(5)));

                                                                                                                                                                                                                                 ^

Bibliothek homie-esp8266-2.0.0-beta.1 in Version 2.0.0 im Ordner: C:\Users\David\Documents\Arduino\libraries\homie-esp8266-2.0.0-beta.1  wird verwendet
Bibliothek async-mqtt-client-0.8.1 in Version 0.8.1 im Ordner: C:\Users\David\Documents\Arduino\libraries\async-mqtt-client-0.8.1  wird verwendet
Bibliothek ESPAsyncTCP-master in Version 1.0.0 im Ordner: C:\Users\David\Documents\Arduino\libraries\ESPAsyncTCP-master  wird verwendet
Bibliothek Ticker in Version 1.0 im Ordner: C:\Users\David\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\Ticker  wird verwendet
Bibliothek ESP8266WiFi in Version 1.0 im Ordner: C:\Users\David\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi  wird verwendet
Bibliothek ArduinoJson in Version 5.11.0 im Ordner: C:\Users\David\Documents\Arduino\libraries\ArduinoJson  wird verwendet
Bibliothek Bounce2-master in Version 2.21 im Ordner: C:\Users\David\Documents\Arduino\libraries\Bounce2-master  wird verwendet
Bibliothek ESP8266mDNS im Ordner: C:\Users\David\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266mDNS (legacy) wird verwendet
Bibliothek ESP8266WebServer in Version 1.0 im Ordner: C:\Users\David\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WebServer  wird verwendet
Bibliothek ESP8266HTTPClient in Version 1.1 im Ordner: C:\Users\David\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266HTTPClient  wird verwendet
Bibliothek DNSServer in Version 1.1.0 im Ordner: C:\Users\David\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\DNSServer  wird verwendet
exit status 1
Fehler beim Kompilieren für das Board NodeMCU 1.0 (ESP-12E Module).`

With ArduinoJson 5.10.1 everything still works.

Nzbuu, 2017-07-01:
I think this has already been fixed: https://github.com/marvinroger/homie-esp8266/issues/361

marvinroger, 2017-07-01:
Fixed in latest git revision.

closed 2017-07-01 by marvinroger

chiefymuc, 2017-07-01:
Sorry for that - I only looked into the open issues. Anyway thx for the fix! Homie 2.0 is working great!