homieiot/homie-esp8266 #367: Frequent WDT resets or exceptions when connecting to WiFi

luebbe, 2017-07-09:
This happens with homie 2.0 develop with an ESP 12F

When my device boots up, the following output is logged via serial several times unti Wifi finally connects successfully. There are several exceptions/WDT resets after the "Attempting to connect to Wi-Fi..." message. Sometimes they occur before the Wi-Fi connected message is shown, sometimes after, like in the attached log. As far as I have understood it, the setupHandler is called after the MQTT_CONNECTED event, so there's none of "my" code involved. This doesn't happen all the time, only very often, so I don't know where to look. Possibly an initalization problem (on my part)?

💡 Firmware ota-basic (1.0.0)
🔌 Booting into normal mode 🔌
{} Stored configuration
  • Hardware device ID: a020a6117237
  • Device ID: a020a6117237
  • Name: Developer
  • Wi-Fi:
    ◦ SSID: xxxxxx
    ◦ Password not shown
  • MQTT:
    ◦ Host: 192.168.0.10
    ◦ Port: 1883
    ◦ Base topic: homie/
    ◦ Auth? no
  • OTA:
    ◦ Enabled? no
↕ Attempting to connect to Wi-Fi...
✔ Wi-Fi connected, IP: 192.168.0.207
Tbiggering WIFI_CONNECTED event...

Exception (0):
epc1=0x402049b0 epc2=0x00000000 epc3=0x00000000 excvaddr=0x4024012a depc=0x00000000

ctx: sys
sp: 3ffffd60 end: 3fffffb0 offset: 01a0

>>>stack>>>
3fffff00:  3fff10d0 3fff1b54 3fffff50 40201400
3fffff10:  3fff10d0 3fff1b54 3fffff50 40208a95
3fffff20:  3ffe84d8 cf00a8c0 001a8ebc 4022bed1
3fffff30:  3fff2220 3fff3efc 3fff430c 4021fa87
3fffff40:  4010075a 00000000 60000600 40214d80
3fffff50:  3ffe84d8 cf00a8c0 3ffe84d8 00ffffff
3fffff60:  3ffe84d8 0100a8c0 00000000 00000005
3fffff70:  00000002 3fff430c 3fff3f24 40214ff2
3fffff80:  40104a1e 00000002 00000000 3ffea038
3fffff90:  3fffdc10 00000000 3fff430c 402250c0
3fffffa0:  40000f49 3fffdab0 3fffdab0 40000f49
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v3ffe8498
~ld
   ▒H▒H▒Device: ESP_117237
ChipID: 117237
Last reset reason: Exception
Memory size: 4194304 bytes
Free heap: 36920 bytes

💡 Firmware ota-basic (1.0.0)
🔌 Booting into normal mode 🔌
{} Stored configuration
  • Hardware device ID: a020a6117237
  • Device ID: a020a6117237
  • Name: Developer
  • Wi-Fi:
    ◦ SSID: xxxxxx
    ◦ Password not shown
  • MQTT:
    ◦ Host: 192.168.0.10
    ◦ Port: 1883
    ◦ Base topic: homie/
    ◦ Auth? no
  • OTA:
    ◦ Enabled? no
↕ Attempting to connect to Wi-Fi...
✔ Wi-Fi connected, IP: 192.168.0.207
Triggering WIFI_CONNECTED event...

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v3ffe8498
~ld
   j▒▒H▒Device: ESP_117237
ChipID: 117237
Last reset reason: Hardware Watchdog
Memory size: 4194304 bytes
Free heap: 37360 bytes

bertmelis, 2017-07-09:
ctx:sys means that the error occurred in the SDK stack (while ctx:cont would mean the sketch stack). So at first I would not seek an error in the Homie code. Is your SDK up-to-date? Can you decode the stack dump?

luebbe, 2017-07-10:
Thanks for the tips,
I'm working with PlatformIO and pio update shows everything in green, so I'd say the answer to your first question is yes. 
The stack dump is the "rare" event. 90% of the time it's a WDT reset.
How would I decode a stack dump with PIO? I've just found a plugin for the standard Arduino IDE.

bertmelis, 2017-07-10:
I wouldn't know. I had such a case and recompiled with the Arduino IDE.

luebbe, 2017-07-10:
thanks, will give it a shot.

luebbe, 2017-08-10:
I think this has been fixed with https://github.com/marvinroger/homie-esp8266/commit/57cb4d026ef912fcda01233384dd8e51588515ab as well. My "basic ota" firmware doesn't have any homie nodes. I flashed it as "clean homie init" many times since and the problem hasn't occured anymore.

closed 2017-08-10 by luebbe