homieiot/homie-esp8266 #368: READY_TO_SLEEP fires prematurely

hageltech, 2017-07-13:
I'm following the docs to go into the deep sleep. However, when HomieEventType::READY_TO_SLEEP event is being generated, it seems to me that the MQTT client is not cleanly disconnected yet. I'm looking a the network trace, and the TCP connection is being left open when the device goes to sleep (when I call ESP.deepSleep(secs) as a response to HomieEventType::READY_TO_SLEEP). 

If I just set a flag when handling the event, and call ESP.deepSleep from the main loop, 100ms later, I can see TCP connection being closed properly.

marvinroger, 2017-08-04:
Indeed, we don't wait for the ACK before firing the event.

stritti, 2019-04-17:
Seems to be stale issue.
Will be closed.
Feel free to reopen if there are more information.

closed 2019-04-17 by stritti