homieiot/homie-esp8266 #371: node.advertise seems to be broken in V2

jhergeth, 2017-07-21:
I downloaded the latest version of homie (d9a2760) and tried to implement my project with it. After some fiddeling I realized, that the device did not get any published messages from the broker. It could send messages, but it did not receive any messages.
I checked the mosquitto logs and found only SUBSCRIBE command from the homie device for topics like homie/18fe34cedb3e/+/+/set, but none for topics I created.
To check I loaded the LightOnOff example and found the same.
Working with platformIO I changed to Arduino 1.8.3, again with no success.

I used the current versions of all libraries.

For me it seems that the advertise() function in HomieNode is broken.

Any help is apreciated!

With regards 
J. Hergeth

p.s. the source files which are mentioned on the release page (https://github.com/marvinroger/homie-esp8266/releases) do not compile with the current versions of the other libraries (compiler error when using DynamicJson...). The revision from the main page (d9a2760) does compile.
I am working under W10

benzino77, 2017-07-22:
Could you post your code?

jhergeth, 2017-07-22:
I am currently on the road, but use just the LightOnOff example. This already does not work.

marvinroger, 2017-07-23:
Well, `homie/18fe34cedb3e/+/+/set` will catch everything, so it should work. What topic are you trying to publish to?

jhergeth, 2017-07-23:
I tried this code:
```
#include <Homie.h>

const int PIN_RELAY = 5;

HomieNode lightNode("light", "switch");

bool lightOnHandler(const HomieRange& range, const String& value) {
  if (value != "true" && value != "false") return false;

  bool on = (value == "true");
  digitalWrite(PIN_RELAY, on ? HIGH : LOW);
  lightNode.setProperty("on").send(value);
  lightNode.setProperty("status").send(on ? "on" : "off");
  Homie.getLogger() << "Light is " << (on ? "on" : "off") << endl;

  return true;
}

void setupHandler(){
  lightNode.advertise("on").settable(lightOnHandler);
  lightNode.setProperty("status").send("started");
}

void setup() {
  Serial.begin(115200);
  Serial << endl << endl;
  pinMode(PIN_RELAY, OUTPUT);
  digitalWrite(PIN_RELAY, LOW);

  Homie_setFirmware("awesome-relay", "1.0.0");
  Homie.setSetupFunction(setupHandler);

  Homie.setup();
}

void loop() {
  Homie.loop();
}
````
and got the message "started" on topic "status". When sending "true" to "homie/18fe34cedb3e/light/on", I get nothing... 
The mosquitto log for the starting phase of the esp8266 is:

```
1500814572: Received PINGREQ from Homie-18fe34cedb3e
1500814572: Sending PINGRESP to Homie-18fe34cedb3e
1500814572: Received PUBLISH from Node-BLUE (d0, q1, r0, m7308, 'HG-BW-Melder', ... (3 bytes))
1500814572: Sending PUBACK to Node-BLUE (Mid: 7308)
1500814584: New connection from 192.168.178.101 on port 1883.
1500814584: Client Homie-18fe34cedb3e already connected, closing old connection.
1500814584: Client Homie-18fe34cedb3e disconnected.
1500814584: New client connected from 192.168.178.101 as Homie-18fe34cedb3e (c1, k15).
1500814584: Sending CONNACK to Homie-18fe34cedb3e (0, 0)
1500814584: Received PUBLISH from Homie-18fe34cedb3e (d0, q1, r1, m1, 'homie/18fe34cedb3e/$homie', ... (5 bytes))
1500814584: Sending PUBACK to Homie-18fe34cedb3e (Mid: 1)
1500814584: Sending PUBLISH to homie-ota-704 (d0, q0, r0, m0, 'homie/18fe34cedb3e/$homie', ... (5 bytes))
1500814584: Received PUBLISH from Homie-18fe34cedb3e (d0, q1, r1, m2, 'homie/18fe34cedb3e/$implementation', ... (7 bytes))
1500814584: Sending PUBACK to Homie-18fe34cedb3e (Mid: 2)
1500814584: Sending PUBLISH to homie-ota-704 (d0, q0, r0, m0, 'homie/18fe34cedb3e/$implementation', ... (7 bytes))
1500814584: Received PUBLISH from Homie-18fe34cedb3e (d0, q1, r1, m3, 'homie/18fe34cedb3e/$mac', ... (17 bytes))
1500814584: Sending PUBACK to Homie-18fe34cedb3e (Mid: 3)
1500814584: Sending PUBLISH to homie-ota-704 (d0, q0, r0, m0, 'homie/18fe34cedb3e/$mac', ... (17 bytes))
1500814584: Received PUBLISH from Homie-18fe34cedb3e (d0, q1, r1, m4, 'homie/18fe34cedb3e/light/$type', ... (6 bytes))
1500814584: Sending PUBACK to Homie-18fe34cedb3e (Mid: 4)
1500814584: Sending PUBLISH to homie-ota-704 (d0, q0, r0, m0, 'homie/18fe34cedb3e/light/$type', ... (6 bytes))
1500814584: Received PUBLISH from Homie-18fe34cedb3e (d0, q1, r1, m5, 'homie/18fe34cedb3e/light/$properties', ... (0 bytes))
1500814584: Sending PUBACK to Homie-18fe34cedb3e (Mid: 5)
1500814584: Sending PUBLISH to homie-ota-704 (d0, q0, r0, m0, 'homie/18fe34cedb3e/light/$properties', ... (0 bytes))
1500814584: Received PUBLISH from Homie-18fe34cedb3e (d0, q1, r1, m6, 'homie/18fe34cedb3e/$name', ... (6 bytes))
[... 61 lines cut ...]
1500814584: Sending PUBACK to Homie-18fe34cedb3e (Mid: 24)
1500814584: Sending PUBLISH to homie-ota-704 (d0, q0, r0, m0, 'homie/18fe34cedb3e/$stats/uptime', ... (1 bytes))
1500814584: Received PUBACK from Homie-18fe34cedb3e (Mid: 1)`

```after that only
```
`1500814605: Received PINGREQ from Homie-18fe34cedb3e
1500814605: Sending PINGRESP to Homie-18fe34cedb3e`
```
and 
```
`1500814637: Received PINGREQ from Homie-18fe34cedb3e
1500814637: Sending PINGRESP to Homie-18fe34cedb3e
1500814644: Received PUBLISH from Homie-18fe34cedb3e (d0, q1, r1, m25, 'homie/18fe34cedb3e/$stats/signal', ... (2 bytes))
1500814644: Sending PUBACK to Homie-18fe34cedb3e (Mid: 25)
1500814644: Sending PUBLISH to homie-ota-704 (d0, q0, r0, m0, 'homie/18fe34cedb3e/$stats/signal', ... (2 bytes))
1500814644: Received PUBLISH from Homie-18fe34cedb3e (d0, q1, r1, m26, 'homie/18fe34cedb3e/$stats/uptime', ... (2 bytes))
1500814644: Sending PUBACK to Homie-18fe34cedb3e (Mid: 26)`
```
appear in the log with respect to that device.

I know several month ago everything worked. .....

Thanks for your help,
Joachim

marvinroger, 2017-07-23:
Well, it subscribes to `homie/18fe34cedb3e/+/+/set`, so as the Homie convention states, you have to publish to `homie/18fe34cedb3e/light/on/set`...

jhergeth, 2017-07-23:
F.....!
It works!

You are right! I knew it was something trivial.......

Thanks for your help and the outstanding project.

With regards
Joachim

marvinroger, 2017-07-23:
You're welcome!

closed 2017-07-23 by marvinroger