homieiot/homie-esp8266 #373: Reboot loop with current version

luebbe, 2017-07-23:
using platformio and homie minimum example, the device goes through an endless reboot loop after uploading. The crash always happens directly after the "Sending initial information..." message is shown.

The program:
```CPP
#define FW_NAME "homie-minimal"
#define FW_VERSION "1.0.0"

#include <Homie.h>

#define SERIAL_SPEED 115200

void setup() {
  Serial.begin(SERIAL_SPEED);

  Homie_setFirmware(FW_NAME, FW_VERSION);

  // before Homie.setup()
  Homie.disableResetTrigger();
  Homie.disableLedFeedback();

  Homie.setup();
}

void loop() {
   Homie.loop();
}
```

Platformio libraries in global storage, no lib_deps in plaformio.ini:
```
Library Manager
===============
Updating Adafruit BME280 Library         @ 1.0.4          [Up-to-date]
Updating Adafruit Unified Sensor         @ 1.0.2          [Up-to-date]
Updating ArduinoJson                     @ 5.11.1         [Up-to-date]
Updating AsyncMqttClient                 @ 0.8.1          [Up-to-date]
Updating Bounce2                         @ 2.1            [Up-to-date]
Updating DHT sensor library              @ 1.3.0          [Up-to-date]
Updating ESP8266_SSD1306                 @ 3.2.7          [Up-to-date]
Updating ESPAsyncTCP                     @ 1.0.1          [Up-to-date]
Updating Homie                           @ cd2fa9f        [Up-to-date]
Updating JsonStreamingParser             @ 1.0.5          [Up-to-date]
Updating NTPClient                       @ 3.1.0          [Up-to-date]
Updating PubSubClient                    @ 2.6            [Up-to-date]
```

The first two repititions of the serial output:
💡 Firmware homie-minimal (1.0.0)
🔌 Booting into normal mode 🔌
{} Stored configuration
  • Hardware device ID: a020a6117237
  • Device ID: a020a6117237
  • Name: Developer
  • Wi-Fi:
    ◦ SSID: onklan
    ◦ Password not shown
  • MQTT:
    ◦ Host: 192.168.0.10
    ◦ Port: 1883
    ◦ Base topic: homie/
    ◦ Auth? no
  • OTA:
    ◦ Enabled? no
↕ Attempting to connect to Wi-Fi...
✔ Wi-Fi connected, IP: 192.168.0.207
Triggering WIFI_CONNECTED event...
↕ Attempting to connect to MQTT...
Sending initial information...

Exception (28):
epc1=0x4020761c epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont
sp: 3fff04d0 end: 3fff0760 offset: 01a0

>>>stack>>>
3fff0670:  3fff279c 3fff1bd0 3fff0e98 40207600
3fff0680:  000000cf 00000000 00000000 fffffffe
3fff0690:  2e003036 2e383631 30322e30 3ffe0037
3fff06a0:  3ffe8590 cf00a8c0 3ffef738 00000030
3fff06b0:  00000000 3ffe87a8 40206d5c 4021cb64
3fff06c0:  00000000 00000000 3fff0e98 3ffef738
3fff06d0:  3fffdad0 00000000 3fff0e98 40208e24
3fff06e0:  3fff004a 3fff0075 000003e8 feefeffe
3fff06f0:  00000000 00000000 feefeffe 3ffef738
3fff0700:  00000000 3fff1dd0 3fff0e78 40203c0d
3fff0710:  00000001 feefeffe feefeffe 3ffef738
3fff0720:  3fffdad0 00000000 3fff1878 40203a29
3fff0730:  00000000 00000000 3ffef730 4021d828
3fff0740:  3fffdad0 00000000 3ffef730 4020256c
3fff0750:  feefeffe feefeffe 3ffef740 402036ec
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v3ffe8550
~ld
   ▒💡 Firmware homie-minimal (1.0.0)
🔌 Booting into normal mode 🔌
{} Stored configuration
  • Hardware device ID: a020a6117237
  • Device ID: a020a6117237
  • Name: Developer
  • Wi-Fi:
    ◦ SSID: onklan
    ◦ Password not shown
  • MQTT:
    ◦ Host: 192.168.0.10
    ◦ Port: 1883
    ◦ Base topic: homie/
    ◦ Auth? no
  • OTA:
    ◦ Enabled? no
↕ Attempting to connect to Wi-Fi...
✔ Wi-Fi connected, IP: 192.168.0.207
Triggering WIFI_CONNECTED event...
↕ Attempting to connect to MQTT...
Sending initial information...

Exception (28):
epc1=0x4020761c epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont
sp: 3fff04d0 end: 3fff0760 offset: 01a0

>>>stack>>>
3fff0670:  3fff3344 3fff1bd0 3fff0e98 40207600
3fff0680:  000000cf 3fffdad0 3ffef738 00000030
3fff0690:  2e003036 2e383631 30322e30 00000037
3fff06a0:  3ffe8590 cf00a8c0 3ffef738 00000030
3fff06b0:  00000000 3ffe87a8 40206d5c 4021cb64
3fff06c0:  00000000 00000000 3fff0e98 3ffef738
3fff06d0:  3fffdad0 00000000 3fff0e98 40208e24
3fff06e0:  3fff004a 3fff0075 000003e8 feefeffe
3fff06f0:  00000000 00000000 feefeffe 3ffef738
3fff0700:  00000000 3fff1dd0 3fff0e78 40203c0d
3fff0710:  00000001 feefeffe feefeffe 3ffef738
3fff0720:  3fffdad0 00000000 3fff1878 40203a29
3fff0730:  00000000 00000000 3ffef730 4021d828
3fff0740:  3fffdad0 00000000 3ffef730 4020256c
3fff0750:  feefeffe feefeffe 3ffef740 402036ec
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v3ffe8550
~ld

claybar, 2017-07-26:
I am getting something similar, but the `Exception (28)` is thrown just after `Attempting to connect to WI-Fi`

AllanMar, 2017-07-26:
I had the same, I downgraded Homie to commit d9a27603ccc4b4b4b6eea04ddfcc44966d6306bd and it's working fine now. I suspect this was the upgrade to asyncmqtt but haven't bothered to confirm yet.

claybar, 2017-07-27:
I'm struggling to get any combination to work now :( .  Always the same reboot loop after attempting to connect to WiFi.

Will try a stock example to see if that works.

n8twj, 2017-07-27:
If using Arduino IDE, try selecting the "Generic ESP8266" board... I have
had reboot loop issues with Sonoff's and standalone ESP-12 modules when I
selected NodeMCU "board".    Not certain how things work with PlatformIO

amayii0, 2017-07-27:
I'm currently on vacation but I'm pretty sure I had to use generic boards for D1 mini clones. 
As far as I can remember I had issues mainly related to memory size when doing various flashes both with Arduino IDE and PlatformIO 
Sometimes also due to cheap Chinese USB cables (grabbed a bundle of 6x 30cm cables from Amazon). Smartphones cables are very decent too.

AllanMar, 2017-07-27:
@claybar: I think that's a separate concern. That's the same kind of behavior I seen when accessing my Wifi when the signal isn't great (your log shows 42-46%).

luebbe, 2017-07-30:
Some more information: I decoded the stack trace with https://github.com/littleyoda/EspStackTraceDecoder.

here's the result:

```
Exception Cause: 28  [LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads]

0x4020761c: HomieInternals::BootNormal::_advertise() at ??:?
0x40207600: HomieInternals::BootNormal::_advertise() at ??:?
0x40206d5c: std::_Function_base::_Base_manager<std::_Bind<std::_Mem_fn<void (HomieInternals::BootNormal::*)(unsigned short)> (HomieInternals::BootNormal*, std::_Placeholder<1>)> >::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at ??:?
0x4021cb64: std::_Function_handler<void (unsigned short), std::_Bind<std::_Mem_fn<void (HomieInternals::BootNormal::*)(unsigned short)> (HomieInternals::BootNormal*, std::_Placeholder<1>)> >::_M_invoke(std::_Any_data const&, unsigned short) at ??:?
0x40208e24: HomieInternals::BootNormal::loop() at ??:?
0x40203c0d: HomieInternals::HomieClass::setup() at ??:?
0x40203a29: HomieInternals::HomieClass::loop() at ??:?
0x4021d828: loop at ??:?
0x4020256c: loop_wrapper() at core_esp8266_main.cpp:?
0x402036ec: cont_norm at cont.o:?
```

Unfortunately no line numbers, but maybe it helps in finding out, what is going wrong

luebbe, 2017-07-30:
Very strange. I uninstalled/reinstalled Homie in Platformio (https://github.com/marvinroger/homie-esp8266/commit/cd2fa9f88bbd9e2746901cf846e5894785663d79), recompiled, uploaded again and now everything is stable.

jhergeth, 2017-08-02:
Same exception for me. Even reinstalling Homie (bb7fbc9) does not help. Decoded stack trace:
```
Exception Cause: 28  [LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads]

0x40207ad8: HomieInternals::BootNormal::_advertise() at ??:?
0x40207abc: HomieInternals::BootNormal::_advertise() at ??:?
0x402092e0: HomieInternals::BootNormal::loop() at ??:?
0x40203c4d: HomieInternals::HomieClass::setup() at ??:?
0x40203828: std::_Function_base::_Base_manager<HomieInternals::HomieClass::HomieClass()::{lambda()#4}>::_M_manager(std::_Any_data&, std::_Function_base::_Base_manager<HomieInternals::HomieClass::HomieClass()::{lambda()#4}> const&, std::_Manager_operation) at Homie.cpp:?
0x4021d5a0: std::_Function_handler<void (), HomieInternals::HomieClass::HomieClass()::{lambda()#4}>::_M_invoke(std::_Any_data const&) at Homie.cpp:?
0x4021eedc: std::_Function_handler<void (), void (*)()>::_M_invoke(std::_Any_data const&) at ??:?
0x40203a69: HomieInternals::HomieClass::loop() at ??:?
0x4021ee44: loop at ??:?
0x4020257c: loop_wrapper() at core_esp8266_main.cpp:?
0x40203748: cont_norm at cont.o:?
0x4010f000: ?? ??:0
```

jbliesener, 2017-08-03:
Same thing here.
Startup messages:

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v3ffe8548
~ld
�

💡 Firmware bare-minimum (1.0.0)
🔌 Booting into normal mode 🔌

{} Stored configuration
  • Hardware device ID: 5ccf7f82f8e5
  • Device ID: 5ccf7f82f8e5
  • Name: Homie-1
  • Wi-Fi:
    ◦ SSID: JBWLAN
    ◦ Password not shown
  • MQTT:
    ◦ Host: 10.0.0.253
    ◦ Port: 1883
    ◦ Base topic: homie/
    ◦ Auth? no
  • OTA:
    ◦ Enabled? yes
↕ Attempting to connect to Wi-Fi...
✔ Wi-Fi connected, IP: 10.0.0.5
Triggering WIFI_CONNECTED event...
↕ Attempting to connect to MQTT...
Sending initial information...

Exception (28):
epc1=0x40207528 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont
sp: 3fff02c0 end: 3fff0550 offset: 01a0

>>>stack>>>
3fff0460:  3fff25a4 3fff19c0 3fff0c88 4020750c
3fff0470:  ffffffff 3fffc6fc 00000001 3ffef520
3fff0480:  00000000 3fffdad0 3ffef528 00000030
3fff0490:  3ffe8588 0500000a 3ffef528 00000030
3fff04a0:  00000000 3ffe8aa7 3fff0ce8 40216fce
3fff04b0:  3fffdad0 3fff0ce8 3fff0c88 3ffef528
3fff04c0:  3fffdad0 00000000 3fff0c88 40208d30
3fff04d0:  3ffefe3a 3ffefe72 000003e8 4021c524
3fff04e0:  00000000 00000000 feefeffe 3ffef528
3fff04f0:  00000000 3fff1bb0 3fff0c68 40203bbd
3fff0500:  3ffe858c feefeffe feefeffe 3ffef528
3fff0510:  3fffdad0 00000000 3fff1668 402039d9
3fff0520:  00000000 00000000 3ffef520 4021d7e0
3fff0530:  3fffdad0 00000000 3ffef520 4020256c
3fff0540:  feefeffe feefeffe 3ffef530 402036ec
<<<stack<<<
```

Mosquitto/DHCP server log:
```
Aug  3 19:29:28 linux3 dhcpd: DHCPDISCOVER from 5c:cf:7f:82:f8:e5 (5ccf7f82f8e5) via enp0s7
Aug  3 19:29:28 linux3 dhcpd: DHCPOFFER on 10.0.0.5 to 5c:cf:7f:82:f8:e5 (5ccf7f82f8e5) via enp0s7
Aug  3 19:29:28 linux3 dhcpd: DHCPREQUEST for 10.0.0.5 (10.0.0.253) from 5c:cf:7f:82:f8:e5 (5ccf7f82f8e5) via enp0s7
Aug  3 19:29:28 linux3 dhcpd: DHCPACK on 10.0.0.5 to 5c:cf:7f:82:f8:e5 (5ccf7f82f8e5) via enp0s7
Aug  3 19:29:29 linux3 mosquitto: 1501799369: New connection from 10.0.0.5 on port 1883.
Aug  3 19:29:29 linux3 mosquitto: 1501799369: Client Homie-5ccf7f82f8e5 already connected, closing old connection.
Aug  3 19:29:29 linux3 mosquitto: 1501799369: Client Homie-5ccf7f82f8e5 disconnected.
Aug  3 19:29:29 linux3 mosquitto: 1501799369: New client connected from 10.0.0.5 as Homie-5ccf7f82f8e5 (c1, k15).
```

Stack Trace:
```
Exception Cause: 28  [LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads]

0x40207528: HomieInternals::BootNormal::_advertise() at ??:?
0x4020750c: HomieInternals::BootNormal::_advertise() at ??:?
0x40216fce: Bounce::update() at ??:?
0x40208d30: HomieInternals::BootNormal::loop() at ??:?
0x4021c524: Print::write(unsigned char const*, unsigned int) at ??:?
0x40203bbd: HomieInternals::HomieClass::setup() at ??:?
0x402039d9: HomieInternals::HomieClass::loop() at ??:?
0x4021d7e0: loop at ??:?
0x4020256c: loop_wrapper() at core_esp8266_main.cpp:?
0x402036ec: cont_norm at cont.o:?
```

jbliesener, 2017-08-03:
Problem seems to be related to commit 182f15d9a67b9578c8a4e81dfb5d9f11602a22e4. Until commit 38933b64bd2e799139c11625ff992721a02bab10 things work fine.

jbliesener, 2017-08-03:
Problem found: Homie crashes on startup when not a single node is defined (as in the "bare minimum" sketch).

Pull request #379 fixes the problem.

marvinroger, 2017-08-04:
Merged! Guys, can you test with @jbliesener's fix?

luebbe, 2017-08-04:
Ouch, nice find. I'll test this evening when I'm back home.

closed 2017-08-25 by luebbe