homieiot/homie-esp8266 #374: homie for  board WeMos D1 R2 & mini

hakam80, 2017-07-25:
Hi,
Is Homie hardware dependent?
I am getting error when compiling the [Bare minimum sketch](http://marvinroger.github.io/homie-esp8266/2.0.0-beta.1/quickstart/getting-started/)

All instructions mentioned in Getting started page are followed

> Error compiling for board WeMos D1 R2 & mini. 

> 

Log are shown below (the error logs report something wrong about some Json functions):

> C:\Users\hakam\Downloads\arduino-1.8.3-windows\arduino-1.8.3\arduino-builder -dump-prefs -logger=machine -hardware C:\Users\hakam\Downloads\arduino-1.8.3-windows\arduino-1.8.3\hardware -hardware C:\Users\hakam\AppData\Local\Arduino15\packages -tools C:\Users\hakam\Downloads\arduino-1.8.3-windows\arduino-1.8.3\tools-builder -tools C:\Users\hakam\Downloads\arduino-1.8.3-windows\arduino-1.8.3\hardware\tools\avr -tools C:\Users\hakam\AppData\Local\Arduino15\packages -built-in-libraries C:\Users\hakam\Downloads\arduino-1.8.3-windows\arduino-1.8.3\libraries -libraries C:\Users\hakam\Documents\Arduino\libraries -fqbn=esp8266:esp8266:d1_mini:CpuFrequency=80,UploadSpeed=115200,FlashSize=4M3M -ide-version=10803 -build-path C:\Users\hakam\AppData\Local\Temp\arduino_build_660173 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool.path=C:\Users\hakam\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.9 -prefs=runtime.tools.mkspiffs.path=C:\Users\hakam\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\0.1.2 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\hakam\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -verbose C:\Users\hakam\Documents\Arduino\esp8266\Homie\Homie.ino
C:\Users\hakam\Downloads\arduino-1.8.3-windows\arduino-1.8.3\arduino-builder -compile -logger=machine -hardware C:\Users\hakam\Downloads\arduino-1.8.3-windows\arduino-1.8.3\hardware -hardware C:\Users\hakam\AppData\Local\Arduino15\packages -tools C:\Users\hakam\Downloads\arduino-1.8.3-windows\arduino-1.8.3\tools-builder -tools C:\Users\hakam\Downloads\arduino-1.8.3-windows\arduino-1.8.3\hardware\tools\avr -tools C:\Users\hakam\AppData\Local\Arduino15\packages -built-in-libraries C:\Users\hakam\Downloads\arduino-1.8.3-windows\arduino-1.8.3\libraries -libraries C:\Users\hakam\Documents\Arduino\libraries -fqbn=esp8266:esp8266:d1_mini:CpuFrequency=80,UploadSpeed=115200,FlashSize=4M3M -ide-version=10803 -build-path C:\Users\hakam\AppData\Local\Temp\arduino_build_660173 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool.path=C:\Users\hakam\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.9 -prefs=runtime.tools.mkspiffs.path=C:\Users\hakam\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\0.1.2 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\hakam\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -verbose C:\Users\hakam\Documents\Arduino\esp8266\Homie\Homie.ino
Using board 'd1_mini' from platform in folder: C:\Users\hakam\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0
Using core 'esp8266' from platform in folder: C:\Users\hakam\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0
Detecting libraries used...
"C:\Users\hakam\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\hakam\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include" "-IC:\Users\hakam\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include" "-IC:\Users\hakam\AppData\Local\Temp\arduino_build_660173/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10803 -DARDUINO_ESP8266_WEMOS_D1MINI -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_WEMOS_D1MINI"  -DESP8266 "-IC:\Users\hakam\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266" "-IC:\Users\hakam\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\variants\d1_mini" "C:\Users\hakam\AppData\Local\Temp\arduino_build_660173\sketch\Homie.ino.cpp" -o "nul"
"C:\Users\hakam\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\hakam\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include" "-IC:\Users\hakam\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include" "-IC:\Users\hakam\AppData\Local\Temp\arduino_build_660173/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10803 -DARDUINO_ESP8266_WEMOS_D1MINI -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_WEMOS_D1MINI"  -DESP8266 "-IC:\Users\hakam\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266" "-IC:\Users\hakam\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\variants\d1_mini" "-IC:\Users\hakam\Documents\Arduino\libraries\homie-esp8266-2.0.0-beta.1\src" "C:\Users\hakam\AppData\Local\Temp\arduino_build_660173\sketch\Homie.ino.cpp" -o "nul"
"C:\Users\hakam\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\hakam\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include" "-IC:\Users\hakam\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include" "-IC:\Users\hakam\AppData\Local\Temp\arduino_build_660173/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10803 -DARDUINO_ESP8266_WEMOS_D1MINI -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_WEMOS_D1MINI"  -DESP8266 "-IC:\Users\hakam\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266" "-IC:\Users\hakam\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\variants\d1_mini" "-IC:\Users\hakam\Documents\Arduino\libraries\homie-esp8266-2.0.0-beta.1\src" "-IC:\Users\hakam\Documents\Arduino\libraries\async-mqtt-client-master\src" "C:\Users\hakam\AppData\Local\Temp\arduino_build_660173\sketch\Homie.ino.cpp" -o "nul"
"C:\Users\hakam\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\hakam\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include" "-IC:\Users\hakam\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include" "-IC:\Users\hakam\AppData\Local\Temp\arduino_build_660173/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10803 -DARDUINO_ESP8266_WEMOS_D1MINI -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_WEMOS_D1MINI"  -DESP8266 "-IC:\Users\hakam\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266" "-IC:\Users\hakam\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\variants\d1_mini" "-IC:\Users\hakam\Documents\Arduino\libraries\homie-esp8266-2.0.0-beta.1\src" "-IC:\Users\hakam\Documents\Arduino\libraries\async-mqtt-client-master\src" "-IC:\Users\hakam\Documents\Arduino\libraries\ESPAsyncTCP-master\src" "C:\Users\hakam\AppData\Local\Temp\arduino_build_660173\sketch\Homie.ino.cpp" -o "nul"
"C:\Users\hakam\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\hakam\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include" "-IC:\Users\hakam\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include" "-IC:\Users\hakam\AppData\Local\Temp\arduino_build_660173/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10803 -DARD [... cut ...]
Using library ESP8266WebServer at version 1.0 in folder: C:\Users\hakam\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WebServer 
Using library ESP8266HTTPClient at version 1.1 in folder: C:\Users\hakam\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266HTTPClient 
Using library DNSServer at version 1.1.0 in folder: C:\Users\hakam\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\DNSServer 
exit status 1
Error compiling for board WeMos D1 R2 & mini.

>

marvinroger, 2017-07-25:
This commit solves this: https://github.com/marvinroger/homie-esp8266/commit/d9a27603ccc4b4b4b6eea04ddfcc44966d6306bd

Can you please update to this revision while waiting for the second beta?

hakam80, 2017-07-25:
It is possible now to compile the project.
Can you please clarify what was the problem?....hardware type has nothing to do with it, right?

marvinroger, 2017-07-25:
Sure, we used an non-conventional way of implementing the ArduinoJson
library, and that "non-conventional way" was disabled in the ArduinoJson
library, so it broke Homie. But implementing the library in a good way (as
we did) solves the issue.

closed 2017-07-26 by hakam80