homieiot/homie-esp8266 #376: Can't connect to Configurator

lnrobby, 2017-07-27:
Installed Homie 2 
Loaded the basic script onto a WEMOS D1 R2 
The serial console shows that it is in configuration mode.

⸮⸮⸮⸮⸮WeMos
✖ /homie/config.json doesn't exist
Configuration invalid. Using CONFIG MODE
💡 Firmware LenWater (1.0.0)
🔌 Booting into config mode 🔌
Device ID is a020a6194836
AP started as Homie-a020a6194836 with IP 192.168.123.1

Connected to the network on my Android V7
Ran up Homie Setup on my android
Just get "Waiting for device".

Have zapped the device again and tried again, still exactly the same.
Have back levelled the libraries to make sure they are on the minimum requirements for Homie 2.0.0 
Still the same

[image]
[image]
[image]

luebbe, 2017-07-27:
Waiting for the device can take more than a minute. How long did you wait?

lnrobby, 2017-07-27:
Over 10 minutes for the first attempt. Since then numerous attempts form 1 minute to 8 hours.

luebbe, 2017-07-27:
That should be long enough :)
with my devices I see different messages in the serial console, when something is connecting to the AP. 
Normally just "scanning for networks" repeatedly.

lnrobby, 2017-07-27:
I have since tried it on a WEMOS D1 Mini Pro. exactly the same situation.

marvinroger, 2017-07-27:
What browser do you use? Which OS? Also, is the configurator loaded over
HTTPS?

lnrobby, 2017-07-27:
Trying to connect to the Wemos to configure it using your app on an LG G5 running android Nougat (Homie Set Up Your Device).  Interestingly in V2 the device is not asking for a password when connecting through the android. I don't know where I find your setup-build for V2 to run as a web app.

I am now seeing a problem just trying to connect to the wifi AP in Nougat as it is saying "Obtaining IP Address" and then stops the connection. Again I have tried zapping and rebuilding it but no success.

lnrobby, 2017-07-27:
Tried yet again by completely zapping the device. Got as far as running an empty homie script and it says in Config mode. Connected via my Android and tried the app again . Nothing. Tried connecting to 192.168.123.1 and it said http://192.168.123.1/ is unreachable

jesserockz, 2017-07-28:
Try turning off your mobile data, Android will most of the time try to route all connections through that if the wifi has no internet access.

lnrobby, 2017-07-28:
Tried turning off mobile data. Still no difference

luebbe, 2017-07-28:
I could connect with browsers on mobile devices (iOS and Android) but had trouble entering or saving the configuration to the device.
Currently I use an old laptop (lubuntu, firefox) for the first time configuration of a homie device. Still needs some time to connect, but works reliably.

lnrobby, 2017-07-28:
I will try that. I wonder what part of this is going wrong.

lnrobby, 2017-07-31:
I have since set up this example from sparkfun https://learn.sparkfun.com/tutorials/esp8266-thing-hookup-guide/example-sketch-ap-web-server. It immediately connects and can talk to the AP so logic suggests that  there must be something stopping it in Homie or one of the libraries I would guess.

raliand, 2017-07-31:
The error is:
XMLHttpRequest cannot load http://192.168.123.1/heart. No 'Access-Control-Allow-Origin' header is present on the requested resource. Origin 'http://setup.homie-esp8266.marvinroger.fr' is therefore not allowed access.

lnrobby, 2017-07-31:
This is now getting even more confusing. Since I tried the Sparkfun example I have been able to successfully launch the V1.5 Homie configurator via the web on oth devices that were not working. This kind of suggests that something was missing in the Homie build but the Sparkfun test sorted it out. I know this doesn't really make much sense as I thought the build when loaded onto the 8266 it cleared all the original stuff out.

sirvaulterscoff, 2017-08-14:
The same issue for me. Had working prototype built with homie on Wemos D1mini.
Then i decided to update configuration of device, so i've sent a message to $implementation/reset endpoint.

After that i've had my device booting in configuration mode. But after AP appears i can't connect to it. Tried both mac and iphone, none of them connects. Tried erasing flash and using different sketches - no luck.

Is there a way to configure homie properties from the code itself?

raliand, 2017-08-14:
I found a chrome extension that did it for me: https://chrome.google.com/webstore/detail/allow-control-allow-origi/nlfbmbojpeacfghkpbjhddihlkkiljbi

benzino77, 2017-08-14:
You can always upload new SPIFFS image with new config.json via Arduino IDE/Platformio.

luebbe, 2017-08-14:
The quickest way for me to send the config to a device in configuration mode is to upload config.json via curl. But that means that your PC/Laptop has to be connected to the homie AP first

sirvaulterscoff, 2017-08-14:
@benzino77 thanks for you feedback. Never done this before, are there any guides on uploading config.json to esp ?

benzino77, 2017-08-14:
I'm using Platformio:
[image]

[image]

roidayan, 2017-10-13:
I also were unable to connect with the default ui_bundle.
I tried chrome/FF/IE11.

In IE11 there is an error Promise is undefined. need to add js to implement Promise if doesn't exists by the browser.

In chrome I got an error about missing js/css so I looked in the ui_bundle file and saw this:
```
<link href="./client.7178b4b2.js" rel="preload">
<link href="./client.9f23c3ce.css" rel="preload">
<link href="./vendor.0f6f9a63.js" rel="preload">
<link href="./manifest.2fe231a1.js" rel="preload">
```

I removed it as those do not exists really. they are inlined in the html already.
After removing this chrome worked fine without an issue.

marvinroger, 2017-10-13:
@roidayan I just fixed that "preload" issue for the latest bundle. 😉

roidayan, 2017-10-13:
Great thanks :-)

ortegafernando, 2017-12-22:
is this fixed? I can't setup Wemos D1 mini with Homie..

luebbe, 2017-12-22:
I always "push" my config.json with curl. It's a lot quicker to edit the config in a normal editor and cal a bat/bash script that uploads it.
The configurator is too slow for me and it is not consistent at some places.

timpur, 2017-12-22:
@luebbe if its not consistent then lets work on fixing that !! Any issues ?

luebbe, 2017-12-22:
Strange, I'm pretty sure that I submitted issues or commented on issues, but I can't find them.
From the top of my head:
- It didn't work with all browsers. FF on Linux was ok, chrome on android too, chrome on iOS not
- You can't enter a zero value in the configurator. 0 doesn't work, but 0.01 is accepted
- And another problem with entering values which I can't remember right now.

Admittedly I haven't used the web configurator for some months because I got fed up.

timpur, 2017-12-22:
okay well im going to be doing alot of work to fix these issues soon when im finished for the year :)

tripflex, 2018-01-20:
TBH I think majority of these issues should have been resolved by the CORS fix...

With that said though, Android Lollipop or newer, if wifi does not have internet it will not route any traffic through wifi (I can 100% confirm this on Android Oreo 8+) which has created a whole ton of fun for me regarding Cordova Wifiwizard plugin I forked https://github.com/tripflex/WifiWizard2/issues/7 

... even if you're attempting to connect to a PRIVATE IP address in PRIVATE IP range, on a network that *should* have a route already set when connecting to that wifi ... blehhh mehhhh rawrr

> Android's Marshmallow operating system introduced a non-forward compatible change that causes network traffic to sometimes use cellular data instead of WiFi when connected to a WiFi access point that cannot reach Google's servers. This is problematic for applications that need to talk to local devices on that access point (i.e., anything in the 10.0.0.0/8 or 192.168.0.0/16 ranges). 

Also, Promises have been around for a while now, so majority of browser versions 2014 or newer should support them (https://caniuse.com/#feat=promises)

Unless you're using Internet Exploder which ... you really shouldn't be using anyways ... but yeah I guess the config ui could have a promises polyfill included to be compatible with shatty browsers

stritti, 2020-01-04:
No progress on this issue. We will close.

closed 2020-01-04 by stritti