homieiot/homie-esp8266 #377: OTA stop when stats are reported

andreasgruener, 2017-07-31:
Hi,

I ran into the following problem:
When doing an OTA everything works fine until homie reports "stats" then the upload of the firmware stops and does not continue. When I try it several times and "stats" are not reported OTA works fine.

Steps for uploading:
**1) hodmin list**
MyDevice BadOben│a020a6174ad1 | edbc70ff287604f78684bc3d0de2c567 |
**2) Pushing Firmware with hodmin**
```$ hodmin -m a020a6174ad1 pushFW -c 6c4fab07beeac085b8b36415162ee13d
Device a020a6174ad1 is ONLINE. (installed FW-Checksum: edbc70ff287604f78684bc3d0de2c567)
New firmware: 6c4fab07beeac085b8b36415162ee13d. Start pushing? <Yn>:Y
202, pushing file
Pushing firmware, 81% -- hangs forever 
```
**3) FAILED Checking upload / status**
```
$mosquitto_sub -v -h 10.0.2.2 -t 'homie/a020a6174ad1/#'
homie/a020a6174ad1/$implementation/ota/status 206 287567/362064
homie/a020a6174ad1/$implementation/ota/status 206 289027/362064
homie/a020a6174ad1/$implementation/ota/status 206 290487/362064
homie/a020a6174ad1/$implementation/ota/status 206 291947/362064
homie/a020a6174ad1/$implementation/ota/status 206 293407/362064
homie/a020a6174ad1/$implementation/ota/status 206 294867/362064
---- > INTERRUPTED
homie/a020a6174ad1/$homie 2.0.0
homie/a020a6174ad1/$implementation esp8266
homie/a020a6174ad1/$mac A0:20:A6:17:4A:D1
homie/a020a6174ad1/light/$type switch
homie/a020a6174ad1/light/$properties on:settable
homie/a020a6174ad1/temperature/$type temperature
homie/a020a6174ad1/temperature/$properties delay:settable
homie/a020a6174ad1/$name BadOben
homie/a020a6174ad1/$localip 10.0.2.234
homie/a020a6174ad1/$stats/interval 60
homie/a020a6174ad1/$fw/name temperature-dallas
homie/a020a6174ad1/$fw/version 1.4.0
homie/a020a6174ad1/$fw/checksum edbc70ff287604f78684bc3d0de2c567
homie/a020a6174ad1/$implementation/config {"wifi":{"ssid":"Razan"},"mqtt":{"host":"10.0.2.2","port":1883,"auth":false},"name":"BadOben","ota":{"enabled":true}}
homie/a020a6174ad1/$implementation/version 2.0.0
homie/a020a6174ad1/$implementation/ota/enabled true
homie/a020a6174ad1/$online true
homie/a020a6174ad1/$stats/signal 100
homie/a020a6174ad1/$stats/uptime 47
homie/a020a6174ad1/temperature/degrees 29.44

homie/a020a6174ad1/temperature/degrees 29.44
homie/a020a6174ad1/temperature/degrees 29.44
homie/a020a6174ad1/temperature/degrees 29.44
```

**3) SUCCESS Checking upload / status**
 ```
206 354727/362064
206 356187/362064
206 357647/362064
206 359107/362064
206 360567/362064
206 362027/362064
206 362064/362064
200

Device reboot new firmware is installed
```

raiderj, 2017-08-02:
I'm seeing the same problem. At the moment I'm using homie-ota to update the firmware. Not sure if the issue is specific to that tool, but maybe not if others are seeing the same issue.

euphi, 2017-08-07:
The logfile you posted looks like the MQTT connection was lost. When it is available again, the complete stats are reported again.

So, the reported stats are **not** the cause of the problem, they are just another symptom.

I had a similar issue, if the Wifi connection was bad. Maybe there is too much MQTT data "queued" and thus the server(?) closes the connection?

If you can get a tcpdump (wireshark) of the MQTT connection, try to look who closes the MQTT connection.

andreasgruener, 2017-08-12:
Thanks for your help. I couldn't solve it yet.

I looked into the MQQ Conneciton. Since wireshark is not a tool I'm really good in. I did the following.
**0) I checked the WIFI Signal and Connectivity**
Couldn't be better is always close to 100%
**1) First upgraded MQQ to the acutal version 1.4.14**
**2) I tried it again, checked the MQQ log** for disconnects, I could not see a disconnect there
But i could see the following entry for the device always close to the upload fails

`1502545871: iot-a020a6080a1d 2 homie/a020a6080a1d/+/+/set`

**Could this be the problem / cause?**

But I kept checking and I did the following:
**3) I connect the Serial Monitor from the Arduino IDE**, there I checked 
   a) if the devices reboots 
   b) or the MQQ reconnects 
Neither did happen, the upload just stalls forever

**4) I tried it with a NodeMCU and NodeMCU v3** 
same behaviour

**ad 3)**
```Serial Log :
💡 Firmware temperature-dht11 (1.4.1)
🔌 Booting into normal mode 🔌
{} Stored configuration
  • Hardware device ID: a020a6174ad1
  • Device ID: a020a6174ad1
  • Name: BadOben
  • Wi-Fi: 
    ◦ SSID: myWIFI
    ◦ Password not shown
  • MQTT: 
    ◦ Host: 10.0.2.2
    ◦ Port: 1883
    ◦ Base topic: homie/
    ◦ Auth? no
  • OTA: 
    ◦ Enabled? yes
↕ Attempting to connect to Wi-Fi...
✔ Wi-Fi connected, IP: 10.0.2.234
Triggering WIFI_CONNECTED event...
↕ Attempting to connect to MQTT...
Sending initial information...
✔ MQTT ready
Triggering MQTT_CONNECTED event...
Calling setup function...
〽 Sending statistics...
  • Wi-Fi signal quality: 98%
  • Uptime: 26s
〽 Sending statistics...
  • Wi-Fi signal quality: 100%

====== About here the Upload stalled  ==== 

  • Uptime: 86s
〽 Sending statistics...
  • Wi-Fi signal quality: 100%
  • Uptime: 146s

=== I restarted the MQQ Server ==== 

✖ MQTT disconnected
Triggering MQTT_DISCONNECTED event...
↕ Attempting to connect to MQTT...
Retrying (500ms)...
↕ Attempting to connect to MQTT...
Sending initial information...
✔ MQTT ready
Triggering MQTT_CONNECTED event...
〽 Sending statistics...
  • Wi-Fi signal quality: 100%
  • Uptime: 204s
〽 Sending statistics...
  • Wi-Fi signal quality: 100%
  • Uptime: 264s
```

andreasgruener, 2017-09-01:
Tried it again today. 

It works now - I used the script provided with homie (scripts/ota_updater/ota_updater.py)
./ota_updater.py -l mqtt.local -i 5ccf7fa123be ~/homie-firmware/ESP8266-Homie-Temperature-BME-280-OLED-v2.ino.bin

Maybe some version conflict.

Thanks a lot for your support!

closed 2017-09-01 by andreasgruener