homieiot/homie-esp8266 #381: lightonhandler setting fine but no light

ramkithepower, 2017-08-07:
Hi ,

Am pretty new to home automation and ws8266. I have setup the following 

WS2812B based LED strip and has 60 LEDs
NodeMCU with WS8266
I have connected the 5v 4amp power supply to volt(red wire) and gnd(white) of LED strip
And from the same power supply i have fed the power to 3-5v logic shifter through the 5v input and gnd
using the power out at the end of the led strip i have powered the nodeMCU Vin and gnd pins
I have connected D4 of nodeMCU to DataIn(green cable) of the LEDstrip through the Logic shifter

The following is my code

```
#include <Homie.h>
#include <Adafruit_NeoPixel.h>

#define FW_NAME "led-control"
#define FW_VERSION "1.0.0"

#define PIN            4
#define LED_COUNT      60

Adafruit_NeoPixel strip = Adafruit_NeoPixel(LED_COUNT, PIN, NEO_GRB + NEO_KHZ800);
int delayval = 500; 
int showType = 0;
int SoffitR;
int SoffitG;
int SoffitB;
int numPixels = LED_COUNT;

HomieNode ledNode("led", "led");

bool lightOnHandler(HomieRange range, String value) {
  Serial.print("Light On Handler");
  if (value == "off") {
    colorWipe(strip.Color(0, 0, 0), 50);    // Black/off
    //Homie.setNodeProperty(ledNode, "color").send("0,0,0,0");
    ledNode.setProperty("color").send("0,0,0,0");
  } else if (value == "red") {
    colorWipe(strip.Color(255, 0, 0), 50);  // Red
    ledNode.setProperty("color").send("0,255,0,0");
  } else if (value == "green") {
    colorWipe(strip.Color(0, 255, 0), 50);  // Green
    ledNode.setProperty("color").send("0,0,255,0");
  } else if (value == "blue") {
    colorWipe(strip.Color(0, 0, 255), 50);  // Blue
    ledNode.setProperty("color").send("0,0,255,0");
  } else {
      //value.trim();
      Serial.print (value);
      //Serial.flush();
      // split string at every "," and store in proper variable
      // convert final result to integer
      SoffitR = value.substring(0,value.indexOf(',')).toInt();
      SoffitG = value.substring(value.indexOf(',')+1,value.lastIndexOf(',')).toInt();
      SoffitB = value.substring(value.lastIndexOf(',')+1).toInt();
      colorWipe(strip.Color(SoffitR, SoffitG, SoffitB), 50);
      ledNode.setProperty("color").send(value);
      Serial.print("Value Set to LED");
  }
  return true;
}

int i;

void colorWipe(uint32_t c, uint8_t wait) {
for(uint16_t i=0; i<strip.numPixels(); i++) { //<strip.numPixels
strip.setPixelColor(i, c);
strip.show();
delay(wait);
}
}

void setup() {
  Serial.begin(115200);
  Serial.println(FW_NAME FW_VERSION);
  
  strip.begin();

  Homie_setFirmware(FW_NAME, FW_VERSION);
  ledNode.advertise("color").settable(lightOnHandler);

  Homie.setup();

}

void loop() {
  Homie.loop();
}
```

I have uploaded the json config also through ESP8266 uploader
{
  "name": "LED for TV Stand",
  "device_id": "family-room-tv-led",
  "wifi": {
    "ssid": "ABCDEF",
    "password": "mypassword"
  },
  "mqtt": {
    "host": "192.168.0.123",
    "port": 1883,
    "base_topic": "devices/",
    "auth": true,
    "username": "username",
    "password": "password"
  },
  "ota": {
    "enabled": true
  }
}

this is my openhab config item
String TVLedLightColor (All) {mqtt=">[broker:devices/family-room-tv-led/led/color/set:command:*:default]"}

When i try to set a color this is what it shows in the Serial Monitor
 Hardware device ID: 6001942bdf7c
  • Device ID: family-room-tv-led
  • Name: LED for TV Stand
  • Wi-Fi: 
    ◦ SSID: ABCDEF
    ◦ Password not shown
  • MQTT: 
    ◦ Host: 192.168.0.123
    ◦ Port: 1883
    ◦ Base topic: devices/
    ◦ Auth? yes
    ◦ Username: username
    ◦ Password not shown
  • OTA: 
    ◦ Enabled? yes
↕ Attempting to connect to Wi-Fi...
✔ Wi-Fi connected, IP: 192.168.0.125
Triggering WIFI_CONNECTED event...
↕ Attempting to connect to MQTT...
Sending initial information...
✔ MQTT ready
Triggering MQTT_READY event...
Calling setup function...
〽 Sending statistics...
  • Wi-Fi signal quality: 76%
  • Uptime: 1s
Calling global input handler...
Calling node input handler...
Calling property input handler...
**Light On Handler**79,255,73**Value Set to LED**

The bolded text or my debug code
But still the light does not come on. LED strip is off. Please can you tell what else can i try?

euphi, 2017-08-07:
Your debug output shows that the value was correctly received by Homie and was propagated to your handler, so your problem is not really related to Homie.

The parsing of the value using substrings seems correct at a first glance, but maybe there is some subtle problem with it. I would use a debug output after parsing that sends the int values.

You can use the `Serial.printf()` method for this, that works really fine on ESP8266:

`Serial.printf("Color values received: r: %d, g: %d, b: %d\n", SoffitR, SoffitG, SoffitB);`

ramkithepower, 2017-08-07:
Is my assumption that PIN 4 outputs to PIN D4 of the NodeMCU correct?

luebbe, 2017-08-07:
No, the assumption is not correct. The pins are mapped differently. This is the first I could google up:
https://github.com/esp8266/Arduino/issues/584
But better google yourself :)

ramkithepower, 2017-08-07:
Yes it was the PIN. I had to prefix 'D' to pin number. That fixed the issue

closed 2017-08-07 by marvinroger