homieiot/homie-esp8266 #382: /homie/config.json doesn't exist after using “ESP8266 Sketch Data Upload”

Bee-Certain, 2017-08-09:
If I upload a configuration file with an error in the file, then Homie sees and reports the error.  If I fix the error and re-upload, Homie says it can not find the file. (???)

Environment: Arduino 1.8.3, Ubuntu 16.04, homie-esp8266-develop, ESP8266FS-0.2.0

I have a config.json file with a deliberate error for ota.enabled:
```
  "ota": {
    "enabled": xx
  },

```

When I do a “ESP8266 Sketch Data Upload”, I get the following:
```
[SPIFFS] data   : /home/bob/Arduino/OGS_TH_SHT30_0.2.0/data
[SPIFFS] size   : 3052
[SPIFFS] page   : 256
[SPIFFS] block  : 8192
/homie/config.json
[SPIFFS] upload : /tmp/buildf16290c8068cf9b1e4282e095ac64b98.spiffs/OGS_TH_SHT30_0.2.0.spiffs.bin
[SPIFFS] address: 0x100000
[SPIFFS] reset  : nodemcu
[SPIFFS] port   : /dev/ttyUSB0
[SPIFFS] speed  : 921600

Uploading 3125248 bytes from /tmp/buildf16290c8068cf9b1e4282e095ac64b98.spiffs/OGS_TH_SHT30_0.2.0.spiffs.bin to flash at 0x00100000
................................................................................ [  2% ]

```
Etc.

After the upload completes, I enable the serial monitor and reset the Wemos D1 Mini and get the following:
✖ **Config file is not valid, reason: ota.enabled is not a boolean**
Configuration invalid. Using CONFIG MODE
💡 Firmware bare-minimum (1.0.0)
🔌 Booting into config mode 🔌
Device ID is 5ccf7f2bb732
AP started as Homie-5ccf7f2bb732 with IP 192.168.123.1
Triggering Wi-Fi scan...
✔ Wi-Fi scan completed

So, clearly the file is being uploaded to the proper location and Homie sees the file. 

So, I go in and fix the config file and re-upload:
```
[SPIFFS] data   : /home/bob/Arduino/OGS_TH_SHT30_0.2.0/data
[SPIFFS] size   : 3052
[SPIFFS] page   : 256
[SPIFFS] block  : 8192
/homie/config.json
[SPIFFS] upload : /tmp/buildf16290c8068cf9b1e4282e095ac64b98.spiffs/OGS_TH_SHT30_0.2.0.spiffs.bin
[SPIFFS] address: 0x100000
[SPIFFS] reset  : nodemcu
[SPIFFS] port   : /dev/ttyUSB0
[SPIFFS] speed  : 921600

Uploading 3125248 bytes from /tmp/buildf16290c8068cf9b1e4282e095ac64b98.spiffs/OGS_TH_SHT30_0.2.0.spiffs.bin to flash at 0x00100000

```
And, after Reset:

✖ **/homie/config.json doesn't exist**
Configuration invalid. Using CONFIG MODE
💡 Firmware bare-minimum (1.0.0)
🔌 Booting into config mode 🔌
Device ID is 5ccf7f2bb732
AP started as Homie-5ccf7f2bb732 with IP 192.168.123.1
Triggering Wi-Fi scan...
✔ Wi-Fi scan completed

Here is the config.json file:
```

{
  "name": "The kitchen light",
  "wifi": {
    "ssid": "be-green-demo-02",
    "password": null
  },
  "mqtt": {
    "host": "192.168.1.10",
    "port": 1883,
    "base_topic": "devices/",
    "auth": false
  },
  "ota": {
    "enabled": false
  },
  "settings": {
    "percentage": 55
  }
}

```

Bee-Certain, 2017-08-10:
Never mind.  I traded out the board for a new one and everything seems to be working.

It would be nice to understand what was going on and why I was getting the strange error messages.

By the way, thanks, Marvin for your excellent work.

closed 2017-08-10 by Bee-Certain

euphi, 2017-08-11:
Are you sure you had set the correct flash memory layout? The file uploaded to SPIFFS has a size of 3M??

If memory size of flashes is configured wrongly, it may happen that it is uploaded to wrong region?
(See https://arduino-esp8266.readthedocs.io/en/latest/filesystem.html#flash-layout)

(I can't tell you where to set it in the arduino IDE, because I'm using platformio+Eclipse, there you can set it in the platformio.ini file with e.g. `build_flags = -Wl,-Tesp8266.flash.1m64.ld`

ech89899, 2018-10-19:
I have exactly the same issue with my ESP-01.
Checked flash memory layout and it is correct.
How can I fix this?

igrowing, 2018-11-03:
@ech89899 
I have the same problem of config.json after burning Homie firmware build on top of EspEasy. The EspEasy was burnt on the ESP-12 before Homie. 2 boards with the same behavior.

I tried following:
- SPIFFS.format(); on boot.
- Remove `/homie/config.json`.
- Create empty `/homie/config.json`.
- Create `/homie/config.json` with all valid fields empty.
- Create `/homie/config.json` with valid values in valid fields.
- Full erase the Flash and redoing the above.
No success so far.

As @Bee-Certain reported, burning the same firmware of Homie on new ESP-12 module works as charm.
I think there are some residues of old firmware somewhere beside Flash...