homieiot/homie-esp8266 #385: homie

neilbrookins, 2017-08-22:
The issue tracker is a great place to ask for enhancements and to report bugs.
If you have some questions or if you need help, some people might help you on the [Gitter room](https://gitter.im/homie-iot/ESP8266).

Before submitting your issue, make sure:

- [ ] You've read the documentation for *your* release (in the `docs/` folder or at http://marvinroger.github.io/homie-esp8266/) which contains some answsers to the most common problems (notably the `Limitations and know issues` and `Troubleshooting` pages)
- [ ] You're using the examples bundled in *your* release, which are in the `examples/` folder of the `.zip` of the release you're using. Examples might not be backward-compatible

Thanks!

amayii0, 2017-08-22:
WTF

neilbrookins, 2017-08-22:
Opened by mistake. Is there a way to close and delete?

closed 2017-08-22 by neilbrookins