homieiot/homie-esp8266 #386: Connection to WiFi fails when AP has special characters in password

neilbrookins, 2017-08-22:
I used the web based setup in config mode to enter the SSID and password. The ESP8266 successfully reboots, but fails to connect to the WiFi which has a password similar to this:    Hit=+NapHippy99
If I go back into config mode and use a different SSID with a password similar to this:   HungrySaab
Then, the WiFi connects successfully and it proceeds to the Mqtt connection as expected.

Question is it the special characters =+ that is breaking it? 
Or, is it the length of the password, 15 characters long that is breaking it?

I'm going to try different passwords and I'll let you know which change breaks it. I'll start with the password that works and add one character at a time from the other password to see which change causes it to stop working. Perhaps there needs to be some special quoting or escaping of the password in the homie code to protect the special characters?

neilbrookins, 2017-08-22:
My testing was with homie v2.0.0-beta.1
I set my NetGear router to use the following passwords and tested each one with Homie:
Hungry=Saab    WiFi connection fails after rebooting from configure mode to normal mode.
Hungry+Saab    WiFi connection fails after rebooting from configure mode to normal mode.
Hungry8Saab    WiFi connection succeeds and it proceeds to attempt MQTT connection as expected.

My conclusion is that Homie is not properly quoting the special characters, both '=' and '+' and probably many others, causing the WiFi connection to fail.

furyfire, 2017-08-23:
Confirmed
Embedding the config.json file using Platformios spiffs feature works no matter what characters you put in the wifi password. 
So it must be on the web API receiving end that special characters get dropped.

stritti, 2020-01-18:
Closing this issue because of no progress for long time.

closed 2020-01-18 by stritti