homieiot/homie-esp8266 #387: Web based configuration page opens new tab

neilbrookins, 2017-08-22:
When I'm using the web based configuration on the initial setup, right after I change the local WiFi client to point to the esp8266 and the browser makes its first communication to the new device running as an AP, the web browser unexpectedly opens a new tab, which then displays the text, "UI bundle not loaded..." which makes it look to the end user like something is wrong. But, actually, the previous browser tab still has the correct configuration setup screen and is working just fine. I'm able to enter the data into the web browser for the SSID, password, mqtt IP, etc. and its all just fine.  This is just confusing to an end user as to why the other browser tab pops open and therefore "hides" the more important tab that is really needed to proceed.  If I wasn't looking closely I would not see the other tab with the configuration setup. 

I was able to work around this issue by closing the extra tab in the browser.

I'm using Firefox on Windows 10 for the web-based configuration.
my device is esp12E and I'm using Homie 2.0.0-beta.1

neilbrookins, 2017-08-22:
I've been doing some more research into this issue. It appears to be related to Windows 10 doing a probe of the internet to determine if there is a captive portal. That's the kind of internet access that some hotels have where you need to log into the AP before you can get internet access. Well, Windows 10 is opening the new tab, trying to browse to www.msftconnecttest.com/connecttest.txt and then after that times out, it displays the page with the text "UI bundle not loaded...".  So, somehow the end user needs to know to ignore the error and look at the other tab. Is it possible to change the text of the error to help assist the user to look in the other tab to continue the setup process? Or, somehow to not open the new tab at all? I'm wondering if I'm the only one to see this behavior? Are most users not using Windows 10?

marvinroger, 2017-08-22:
This is intended. The captive portal is implemented so that users who have
loaded the UI bundle can configure the device without having to open the
browser. They only have to connect to the Wi-Fi network and the bundled
configurator opens.

Indeed, the text can be clearer about this. 😊

closed 2020-01-18 by stritti