homieiot/homie-esp8266 #390: Release of stable version and Implementation of Homie 2.1

euphi, 2017-08-28:
There was a large progress in the Homie-Convention in the last weeks.

While the convention is not released yet, it seems to be ready to have a reference implementation (e.g. see https://github.com/marvinroger/homie/issues/24)

Meanwhile, the Homie-es8266 2.0.0-beta1 runs quite stable, with only some non-critical open bugs.

However, some of the changes to homie-convention are breaking compatibility (e.g. Arrays have been moved from Property to Node-Level).

**So, to handle the progress  of Homie and Homie-esp8266 I propose to release Homie-esp8266-v2.0.0 _stable_ as soon as possible _without_ any of the new features of Homie-convention 2.1.0**

In detail:
* create branch 'dev-v2.0' and make it default branch (or merge 'development' to master)
* **tag the current state of homie-esp8266 as _v2.0.0-beta.2_** 
* **Release the Homie convention 2.0.0 as stable**
* identify bugs in the homie-esp8266 issue list that should be fixed before a final release and label them 'v2.0'.
* As soon as necessary, create branch 'dev-2.1' to implement Homie convention v2.1.0 **(or any other new feature)**
* Bug fixes now should be done in both 'dev-v2.0' and 'dev-v2.1' branches (there aren't that many)
* **If all important bugs have been fixed, release v2.0.0**
* If new bugs are found in homie-esp8266 v2.0.0, they can be fixed in the dev-2.0 branch. If necessary, release v2.0.1 etc.

Feel free to discuss and improve :-)

timpur, 2017-11-02:
lets get into this :)

stritti, 2020-01-18:
It is already done.

closed 2020-01-18 by stritti