homieiot/homie-esp8266 #391: Can't configure Homie when wifi password contains an ampersand (&)

fbjerggaard, 2017-08-31:
When trying to configure the homie device and trying to connect to a wifi with a password that contains an ampersand (&), configuration upload fails.

This JSON object fails:
`{"wifi":{"ssid":"MyWifiName","password":"MyWifi&Password"},"mqtt":{"host":"192.168.1.2","port":1883,"base_topic":"home/","auth":false},"name":"ledstrip 1","ota":{"enabled":false},"device_id":"ledstrip-1"}`

This JSON works:
`{"wifi":{"ssid":"MyWifiName","password":"MyWifiPassword"},"mqtt":{"host":"192.168.1.2","port":1883,"base_topic":"home/","auth":false},"name":"ledstrip 1","ota":{"enabled":false},"device_id":"ledstrip-1"}`

furyfire, 2017-09-01:
#386 Same issue

furyfire, 2017-09-01:
The underlaying Arduino library attempts to parse the body of the incomming request. 
Build project with -DDEBUG_ESP_HTTP_SERVER - It does not appear to be a problem with Homie, and changing the MIME type of the Post does not seem to have any effect either

georg90, 2020-06-12:
Can this get fixed? It's still an issue with the latest version of homie.

stritti, 2020-12-28:
@luebbe could we solve this old issue? I am not sure because of side effects.