homieiot/homie-esp8266 #392: Flashing SONOFF with Homie-Sketch does not work

gtom, 2017-09-03:
Trying to use a itead sonoff-switch. There are some issues down below, but my problem seems to be a little bit different.
Sometimes (very seldom) I am able to flash one of sonoff-homie-examples (sonoff), after flashing the normal configure-mode starts. But in most cases after flashing the example the led just remains off (IDE reports flashing OK). It looks like nothing is happen.

I read in some issues down below that esptool erase should help before flashing - unfortunatly not in my case. I also changed my Arduino-IDE-config from 1M-64K SPIFFS to 512K-64K SPIFFS. Didn't help either.
I tried platformio to flash same example with same result. Always it reports success in flashing, but config-mode doesn' t start.
I thought it could be a problem in flashing, so I tried another example (just a blink, without homie). In this case every flash is OK, the device reboots and LED starts blinking. 
I don' t think that this is a homie-bug, but it seems to be homie-related. Maybe someone has experience in flashing sonoff with homie-sketches and can point me in right direction?
Thanks gTom

n8twj, 2017-09-03:
Hi - Make sure you are using "Generic ESP8266" setting (at least in Arduino
IDE)

I have quite a few SONOFF's running Homie.

amayii0, 2017-09-03:
I second that, however I don't use that ATM, see:
https://github.com/amayii0/Homie-Sonoff
https://www.instructables.com/id/Use-Homie-Firmware-to-Drive-Sonoff-Switch-Module-E/

n8twj, 2017-09-03:
Ahh... I haven't take the time to dive into PlatformIO...Guess I should

gtom, 2017-09-04:
No, does not work: I tried the sketch from amayii0 above vai platformi0. This is my platform.ini:
`

[env:esp01]
;[env:esp01_1m]
platform = espressif8266
board = esp01
;board = esp01_1m
framework = arduino
lib_deps = git+https://github.com/marvinroger/homie-esp8266.git#develop

`
I tried both esp01 and esp01_1m with erasing flash between my trials. 

Maybe this info helps: I did catch messages during boot process with Arduino-IDE. This is what the device says (repeating):

`

ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v3ffe8580
~ld

`

kylegordon, 2017-09-05:
What sort of Sonoff switch? The touch switch uses the esp8285, which if memory serves will result in a the behaviour you describe if you compile and flash esp8266 code :-) I have working Sonoff touch switch code at https://github.com/kylegordon/homie-sonoff-touch if it helps.

gtom, 2017-09-07:
No, it is this one: [https://www.itead.cc/sonoff-wifi-wireless-switch.html](https://www.itead.cc/sonoff-wifi-wireless-switch.html)
And because some simple sketches work, it doesn' t seem to be related with cpu. Maybe it has something to do with used memory: including the homie-libs eats more memory. In the next day I will try to extend a simple blink stepwise to a more complex sketch and see what happens.

kylegordon, 2017-09-07:
Have you tried https://github.com/enc-X/sonoff-homie ?

It also uses Homie, v1 admittedly, but has been working fine for me on those devices for a while now. 

I'm actually in the process of trying to update that project to Homie 2

luebbe, 2017-09-08:
I own three of these and they work like a charm with my firmware at https://github.com/luebbe/homie-sonoff (Homie 2.0)

nerdfirefighter, 2018-01-14:
All the Sonoff that i have played with (TH10/16 ver2.1, Touch US ver1.0, and TH_v1.1)  are ESP8266. I brought all mine in the last few months.

I use the following settings in Ardiuno are:
//use the following settings when deploying to SONOFF Device
// Board: Generic ESP8266 Module
// Flash Mode: DOUT
// Flash Freq: 40Mhz
// CPU Freq: 80Mhz
// Flash Size: 1M (64K SPIFFS)
// Debug Port: disabled
// Debug Level: None
// reset Method: ck
// upload speed: 115200

I have tested with Homie-esp8266 2.0.0-beta.2 and 2.0.0-beta.3

Hope this helps.

n8twj, 2018-01-14:
Hi -  I too have Sonoff's and have experienced some very odd behavior.
One trick I learned by accident is when flashing the Sonoff's let them
reboot on their own after flashing...I suspect the SPIFFS task is either
hanging and/or takes quite a bit longer than "other" ESP8266s.

Also, make certain you have a solid ground between the Sonoff and the FTDI
and/or simply try a known good/reliable/not cloned FTDI board.... For some
reason the Sonoff's seem to hate my cheap/clone FTDIs but I never seem to
have any trouble with my [expensive] SparkFun FTDI.

Lastly, try using the NodeMCU 0.9 version target... Especially on very old
Sonoff's.

Good Luck - You can ping me directly if you get totally stuck.

dungtqa, 2018-07-22:
In file platformio.ini setting board = esp8285

stritti, 2020-12-28:
Closing issue: no progress/feedback.

closed 2020-12-28 by stritti