homieiot/homie-esp8266 #394: MQTT Client ID = Homie Brand + Homie Device Id

timpur, 2017-09-05:
Looking at wire shark the mqtt client id seems to be set as the brand + the device id??
Is there a reason for this?

furyfire, 2017-09-07:
the MQTT client ID has to be unique for every connection - The brand + device id seems like a fine way of doing that? 

> ClientId

> The client identifier (short ClientId) is an identifier of each MQTT client connecting to a MQTT broker. As the word identifier already suggests, it should be unique per broker. The broker uses it for identifying the client and the current state of the client. If you don’t need a state to be hold by the broker, in MQTT 3.1.1 (current standard) it is also possible to send an empty ClientId, which results in a connection without any state. A condition is that clean session is true, otherwise the connection will be rejected

timpur, 2017-09-07:
But shouldn't the Device ID be unique on its own? inst that the point of an id?

closed 2017-11-02 by timpur