homieiot/homie-esp8266 #395: Custom $stats items.

kylegordon, 2017-09-05:
Is it possible to add custom values to the $stats topic that Homie publishes? I use the following to get a report of free heap space, but it would be nice to publish is under the $stats topic rather than the node topic. 

ROM_Temp0.setProperty("freeheap").send(String(ESP.getFreeHeap(),DEC));

euphi, 2017-09-06:
It is already in the draft of the next version of homie-convention, see
https://github.com/marvinroger/homie/tree/redesign#device-statistics

Up to now there is no way to add own topics to the $stats hierarchy except chaning it directly in the homie-esp8266 sources.

kylegordon, 2017-09-06:
Ah, fantastic, good to know. Thank you :-)

closed 2017-09-06 by kylegordon