homieiot/homie-esp8266 #398: Homie crashes when running ESP.getFreeHeap()

johnjore, 2017-09-11:
I used to run this 

`Serial << "Heap size: " << ESP.getFreeHeap() << " bytes, ";`

over night to validate there were no memory leaks, and it used to work. I the version downloaded today (Sep 11) from develop branch, this happens:

```
Heap size: 
Exception (3):
epc1=0x402232b2 epc2=0x00000000 epc3=0x00000000 excvaddr=0x40032dc1 depc=0x00000000

ctx: cont 
sp: 3fff2290 end: 3fff25b0 offset: 01a0

>>>stack>>>
3fff2430:  00000003 00000001 3fff0f68 402148c0  
3fff2440:  00000003 00000001 3fff0f68 40212ee0  
3fff2450:  00000000 002300e0 0000001e 4010053d  
3fff2460:  00000000 0000000a 3ffe8b52 3fff0258  
3fff2470:  3fff01f4 3fff14a0 3fff14c0 40223368  
3fff2480:  3fff01f4 3fff10e8 3fff14c0 40201f08  
3fff2490:  3ffe8b48 3fff0cb4 3fff1338 40239ca9  
3fff24a0:  4021c128 3fff10e8 3fff14c0 4021c9ec  
3fff24b0:  3fff2530 3fff0cb4 3fff01a0 4020821f  
3fff24c0:  00000002 00000001 3fff0f68 402148c0  
3fff24d0:  00000002 00000001 3fff0f68 40212ee0  
3fff24e0:  3fff2933 002500e8 3fff0c23 4021f43c  
3fff24f0:  00000025 00000001 3fff2530 3fff0258  
3fff2500:  3fff0f68 3fff10e8 3fff0230 4021f126  
3fff2510:  40250020 3fff10e8 3fff0258 402103da  
3fff2520:  3fff0f68 3fff10e8 3fff0230 40210a79  
3fff2530:  3f003732 3ffe8384 3fff14c0 4021c909  
3fff2540:  3ffe9954 003638f4 000001f4 4021f11c  
3fff2550:  00000024 3ffe8384 3fff14c0 4021ca04  
3fff2560:  3fff00a4 3ffe8698 3fff14c0 3fff157c  
3fff2570:  3fffdad0 00000000 3fff0c10 40208cb5  
3fff2580:  00000000 00000000 3fff1575 40208324  
3fff2590:  3fffdad0 00000000 3fff1575 4021e1dc  
3fff25a0:  feefeffe feefeffe 3fff1590 40100718  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v3de0c112
~ld
µ
```

Any reason ESP.getFreeHeap() now crashes homie?

bertmelis, 2017-09-12:
What version did you use before?
To have the stack trace mean something useful, you'll have to [decode](https://github.com/me-no-dev/EspExceptionDecoder) it.
Do you call the function from within an interrupt (like using the Ticker lib?)

johnjore, 2017-09-12:
I think the previous version was from about 4 or 5 months ago. I'll take a look at the strack trace, thanks for the link, and no, its called from the loop.

Gulaschcowboy, 2017-10-03:
May I jump in...
I observe same.  ESP.getFreeHeap() is also called during OTA Updates and therefore the ESP crashes. On the other hand, I can also make it crash by including Homie.getLogger() << "Heap size: " << ESP.getFreeHeap() << " bytes, "; in my loop.

Decoded stack trace of an OTA attempt:
```
0x402142b5: AsyncClient::add(char const*, unsigned int, unsigned char) at /home/atze/Dokumente/arduino/libraries/ESPAsyncTCP/src/ESPAsyncTCP.cpp line 650
0x40104b30: ets_timer_arm_new at ?? line ?
0x40233b38: pm_set_sleep_type_from_upper at ?? line ?
0x4022f6ac: umm_free_heap_size at /home/atze/.arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc1/cores/esp8266/umm_malloc/umm_malloc.c line 1739
0x40254880: sleep_reset_analog_rtcreg_8266 at ?? line ?
0x40201880: xPortGetFreeHeapSize at /home/atze/.arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc1/cores/esp8266/heap.c line 73
0x40246e71: system_get_free_heap_size at ?? line ?
0x4021c500: EspClass::getFreeHeap() at /home/atze/.arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc1/cores/esp8266/Esp.cpp line 498
0x4021d424: UpdaterClass::begin(unsigned int, int) at /home/atze/.arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc1/cores/esp8266/Updater.cpp line 403
0x4021f826: HomieInternals::Logger::write(unsigned char) at /home/atze/Dokumente/arduino/libraries/homie-esp8266/src/Homie/Logger.cpp line 20
0x4020f254: HomieInternals::BootNormal::_onMqttMessage(char*, char*, AsyncMqttClientMessageProperties, unsigned int, unsigned int, unsigned int) at /home/atze/Dokumente/arduino/libraries/homie-esp8266/src/Homie/Boot/BootNormal.cpp line 472 (discriminator 3)
0x40238ab2: ieee80211_output_pbuf at ?? line ?
0x40100ec2: pp_post at ?? line ?
0x40100ec2: pp_post at ?? line ?
0x401041bf: lmacRxDone at ?? line ?
0x40101bd9: trc_NeedRTS at ?? line ?
0x40101d68: trc_NeedRTS at ?? line ?
0x4021f968: std::_Function_handler    (HomieInternals::BootNormal*, std::_Placeholder1>, std::_Placeholder2>, std::_Placeholder3>, std::_Placeholder4>, std::_Placeholder5>, std::_Placeholder6>)> >::_M_invoke(std::_Any_data const&, char*, char*, AsyncMqttClientMessageProperties, unsigned int, unsigned int, unsigned int) at /home/atze/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/functional line 2073
0x4020de77: std::_Function_base::_Base_manager    (HomieInternals::BootNormal*, std::_Placeholder1>, std::_Placeholder2>, std::_Placeholder3>, std::_Placeholder4>, std::_Placeholder5>, std::_Placeholder6>)> >::_M_clone(std::_Any_data&, std::_Any_data const&, std::integral_constant ) at /home/atze/Dokumente/arduino/libraries/homie-esp8266/src/Homie/Boot/BootNormal.cpp line 57
:  (inlined by) std::_Function_base::_Base_manager    (HomieInternals::BootNormal*, std::_Placeholder1>, std::_Placeholder2>, std::_Placeholder3>, std::_Placeholder4>, std::_Placeholder5>, std::_Placeholder6>)> >::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at /home/atze/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/functional line 1946
0x4021fc9e: std::function ::function(std::function  const&) at /home/atze/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/functional line 2439
0x40101f40: wDev_ProcessFiq at ?? line ?
0x40212af8: ~function at /home/atze/Dokumente/arduino/libraries/async-mqtt-client/src/AsyncMqttClient.cpp line 92
:  (inlined by) AsyncMqttClient::_onMessage(char*, char*, unsigned char, bool, bool, unsigned int, unsigned int, unsigned int, unsigned short) at /home/atze/Dokumente/arduino/libraries/async-mqtt-client/src/AsyncMqttClient.cpp line 515
0x4020de54: std::_Function_base::_Base_manager    (HomieInternals::BootNormal*, std::_Placeholder1>, std::_Placeholder2>, std::_Placeholder3>, std::_Placeholder4>, std::_Placeholder5>, std::_Placeholder6>)> >::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at /home/atze/Dokumente/arduino/libraries/homie-esp8266/src/Homie/Boot/BootNormal.cpp line 57
0x4021f940: std::_Function_handler    (HomieInternals::BootNormal*, std::_Placeholder1>, std::_Placeholder2>, std::_Placeholder3>, std::_Placeholder4>, std::_Placeholder5>, std::_Placeholder6>)> >::_M_invoke(std::_Any_data const&, char*, char*, AsyncMqttClientMessageProperties, unsigned int, unsigned int, unsigned int) at /home/atze/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/functional line 2069
0x4021fb39: std::_Function_handler    (AsyncMqttClient*, std::_Placeholder1>, std::_Placeholder2>, std::_Placeholder3>, std::_Placeholder4>, std::_Placeholder5>, std::_Placeholder6>, std::_Placeholder7>, std::_Placeholder8>, std::_Placeholder9>)> >::_M_invoke(std::_Any_data const&, char*, char*, unsigned char, bool, bool, unsigned int, unsigned int, unsigned int, unsigned short) at /home/atze/Dokumente/arduino/libraries/async-mqtt-client/src/AsyncMqttClient.cpp line 92
0x40213e2d: AsyncMqttClientInternals::PublishPacket::parsePayload(char*, unsigned int, unsigned int*) at /home/atze/Dokumente/arduino/libraries/async-mqtt-client/src/AsyncMqttClient/Packets/PublishPacket.cpp line 84
0x402128ed: AsyncMqttClient::_onData(AsyncClient*, char*, unsigned int) at /home/atze/Dokumente/arduino/libraries/async-mqtt-client/src/AsyncMqttClient.cpp line 92
[... cut ...]
0x4022ea48: tcp_input at /Users/ivan/e/arduino-esp8266/tools/sdk/lwip/src/core/tcp_in.c line 394 (discriminator 1)
0x40106748: pvPortMalloc at /home/atze/.arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc1/cores/esp8266/heap.c line 40
0x4022d109: ip_input at /Users/ivan/e/arduino-esp8266/tools/sdk/lwip/src/core/ipv4/ip.c line 559
0x4022c4d9: ethernet_input at /Users/ivan/e/arduino-esp8266/tools/sdk/lwip/src/netif/etharp.c line 1379
0x40230eaf: pp_tx_idle_timeout at ?? line ?
0x40230e22: pp_tx_idle_timeout at ?? line ?
0x40245b67: ets_snprintf at ?? line ?

```

johnjore, 2018-01-27:
Downloaded latest set of libraries and homie available as of now, (Jan 21st 2018), and problem has been resolved. Closing.

closed 2018-01-27 by johnjore