homieiot/homie-esp8266 #402: How to receive data from second Homie via MQTT broker?

petosak, 2017-09-24:
Hi,
anybody could help me?
I have this in my project:
RPi3 ( openhabian) + openhab + mosquitto + influxDB, etc
5x Wemos D1 - homie v2 and sensors like BME280, Dallas, etc.
This scenario works good: sensor -> Wemos -> Homie v2 -> Mosquitto -> openhab -> InfluxDB
But on one Wemos D1 mini device i would like to see temperature from other device.
One Wemos D1 i have for outside temperature. Second Wemos inside for inside temperature with OLED.
I would like to see on second device both temperatures. How to receive data from first device?
Thanks
Petr

LiyouZhou, 2017-09-24:
you need to subscribe on your second device the resources on the first device.

the syntax might look something like this: https://github.com/marvinroger/async-mqtt-client/blob/master/examples/FullyFeatured/FullyFeatured.ino#L43

petosak, 2017-09-24:
@LiyouZhou thanks, but it is not for homie...:(

bertmelis, 2017-09-24:
Homie for ESP8266 is a node implementation, not a server. If you just want to receive one node property from another device, create a MQTT client and subscribe to the topic. You can use the asyncmqtt library from marvinroger for this.

petosak, 2017-09-25:
@bertmelis I should use Homie2 and asyncmqtt library together on one ESP8266 ?

bertmelis, 2017-09-25:
That's one possibility.
I don't know if you can use Homie's underlying mqttclient `Homie.getMqttClient();`.
Another one is to "translate" the topic on a server and work on your ESP2 with Homie and a settable property. I'm using this one with node-red for the translation-step.

euphi, 2017-09-25:
@petosak Homie uses the AsyncMQTT library, so it is already available.
To get the MQTT client instance, use `Homie.getMqttClient();`.

benzino77, 2017-09-25:
You probably need to create another instance of  `AsyncMqttClient` because you will attach your own `onMessage` callback. Using underlying homie `AsyncMqttClient` instance may be tricky... I think. From my perspective @bertmelis approach with node-red is the coolest one.

euphi, 2017-09-25:
@benzino77 The AsyncMqttClient supports multiple callbacks, see https://github.com/marvinroger/async-mqtt-client/blob/master/src/AsyncMqttClient.cpp , line 135.

However, I also prefer to have a central server (combined with MQTT broker) to change message and resend them as this allows for better control and additional features.

benzino77, 2017-09-25:
@euphi Everyday I learn something ;)

furyfire, 2017-09-26:
You could use the broadcast feature?
http://marvinroger.github.io/homie-esp8266/develop/advanced-usage/broadcast/

Saves you from subscribing yourself to specific topics

petosak, 2017-09-28:
thanks a lot.
I really appreciate every advice.
once I have a moment, I will try every advice.

I thought it would be much easier;)

stritti, 2019-04-17:
Seems to be stale issue.
Will be closed.
Feel free to reopen if there are more information.

closed 2019-04-17 by stritti