homieiot/homie-esp8266 #404: BSSID / Channel doesn't seem to work

x821938, 2017-09-26:
When I try to add BSSID and channel to my config.json like this

    "bssid": "DE:AD:BE:EF:BA:BE",
    "channel": 1,

I can't connect to my wifi. I am pretty sure about my BSSID and channel.
I only get this:

```
↕ Attempting to connect to Wi-Fi...
2017-09-26 14:08:15	Message	✖ Wi-Fi disconnected
Triggering WIFI_DISCONNECTED event...
2017-09-26 14:08:15	Message	↕ Attempting to connect to Wi-Fi...
2017-09-26 14:08:17	Message	✖ Wi-Fi disconnected
Triggering WIFI_DISCONNECTED event...
↕ Attempting to connect to Wi-Fi...
2017-09-26 14:08:19	Message	✖ Wi-Fi disconnected
Triggering WIFI_DISCONNECTED event...
↕ Attempting to connect to Wi-Fi...
```

I am using the latest development version of Homie. 

If I remove the two lines of config it connects very fast. But I want to connect even faster for a Homie-battery project I am working on.

timpur, 2017-12-30:
still having this issue?

x821938, 2018-01-02:
I made my own solution, so I havn't really checked.
If you havn't changed to code, I will still have the problem. I tried on 3 different nodemcu's

Best regards,
   Alex

timpur, 2018-01-02:
Okay I'll test it out and fix if there is still an issue.

tripflex, 2018-01-20:
@x821938 did you try this with debug output enabled? I only see homie specific logging ...  just add this in setup `Serial.setDebugOutput(true);` and you should get more specific information from actual wifi connection in console

stritti, 2020-01-18:
Closing because of no more response.

closed 2020-01-18 by stritti