homieiot/homie-esp8266 #405: OTA Config updates - mqtt message stripping chars

psyciknz, 2017-09-28:
As per: http://marvinroger.github.io/homie-esp8266/develop/others/ota-configuration-updates/#ota-entities-projects

I'm trying ot play with updating the config file.  It seems to imply I send a json string to
`<topic>/<device>/$implmentation/config`

So in my case (since I've changed the topic to sensors/d1mini):
` mosquitto_pub -t sensors/d1mini/mqtt-test/$implementation/config -m <json data>`

But when I subscrive to sensors/d1mini/# I see the message come in as:
`sensors/d1mini/mqtt-test//config`

Am I missing something, or it this functionality still being developed?

elescondite, 2017-09-29:
Put the topic in quotes and all will be well (the shell is seeing $implementation as an empty variable)
```
mosquitto_pub -t "sensors/d1mini/mqtt-test/$implementation/config" -m <json data>

```

psyciknz, 2017-09-29:
Actually I did work it out, and it's single quotes not double, else it was ignored.

So I made it do something, but then it immediately fries the board 9gets a WDT reset and locks up.

psyciknz, 2017-11-12:
So been playing again, managing to set a json file to the config/set, and after a second I see the configt message get an update, but it doesn't seem to get a new value (I'm just trying to update the name with:
mosquitto_pub -t 'sensors/d1mini/mqtt-test/$implementation/config/set' -m '{ "name": "Andoni Sensor" }'

It does cause the device to reboot it seems (which is probably what the sub on /config is picking up.

timpur, 2017-12-30:
still having these issues ?

stritti, 2020-01-18:
Closing because of no response for long time.

closed 2020-01-18 by stritti