homieiot/homie-esp8266 #408: ESP doesn't connect to Wifi after Reset

luebbe, 2017-10-14:
Hi folks,
I have a strange problem with my ESP12F. Whenever I flash it or power-cycle it, it boots without problems about 95% of the time and connects to Wifi. When I pull the reset pin low, it boots, but Homie hangs in the "Attempting to connect to WiFi" state 100% of the time and never gets out of it. I then have to power cycle it.
Has anyone else observed this behaviour?
Using Homie@develop, Environment: VS Code + PlatformIO

yves-ledermann, 2017-10-17:
Hi @luebbe 
I think i have the same problem.

But i always thought its because of my "unifi" wifi setup.
If i hit the reconnect button on the wifi control interface (unifi) it reconnects wihout any problem.
I did not have enough time to test it with a different Wifi AccessPoint of a different Vendor.
I hope you are using a different wifi setup than me?

Im also using homie@develop with VS Code & PIO

luebbe, 2017-10-18:
Hi Yves,

I found the following threads, which indicate that there is indeed a problem with Wifi after a hard reset.

https://github.com/esp8266/Arduino/issues/2235
https://github.com/esp8266/Arduino/issues/1722

Very interesting is this comment in the second thread:
https://github.com/esp8266/Arduino/issues/1722#issuecomment-192624783

> ESP.reset() is a hard reset and can leave some of the registers in the old state which can lead to problems, its more or less like the reset button on the PC.

So I guess that's what we're experiencing.

I implemented an ESP.restart() via gpio and a button to get out of this situation. Can't power cycle the device in question. It's in a tiny 3D printed case and always on battery.

closed 2017-12-16 by timpur

kleini, 2019-02-03:
I encounter a similar problem. GPIO16 is connected to RESET and the device is put into deep sleep for 60 seconds. When the device is reset, reconnecting to WiFi does not work at all. It keeps blinking in WiFi connecting status.