homieiot/homie-esp8266 #409: Add homie nodes at runtime?

tkosig, 2017-10-17:
Hi there,

first let me say: great project! Just stumbled over it recently and it addresses lots of issues with using the ESP8266 in a home-automation environment.

After reading the docs and setting up a test device as a light switch I was wondering if homie could be used to cover a use case of mine which is a bit more complex.

I run an ESP8266 as a 868MHz-to-MQTT gateway to receive data from commercially available temperature/humidity sensors. My setup is working great, the sensors themselves are very battery efficient and update temp/humidity every 4 seconds.

My gateway forwards the received values to the MQTT broker using this topic structure:

    868mhz-gateway/out/<sensor ID>/temperature
    868mhz-gateway/out/<sensor ID>/humidity

The gateway does not know which sensor IDs it will receive upfront so this is something that can't be configured - as in homie's nodes.

I could publish the sensor ID together with temperature and humidity in the payload of a generic "sensor-data" topic but that would shift the responsibility of parsing the values to the consumer which is connected to the MQTT broker.

In my case this is OpenHAB and while it offers a JsonPath transformation it seems not the right thing to invoke 20+ item updates which will not get any new data anyways as one payload would only contain sensor data for one sensor at a time.

Does homie allow to just set a property if it hasn't been advertised during the setup process?
Or even to add another hierarchy level by adding a slash to the property?

    lightNode.setProperty("<deviceID>/temperature").send(value);

Or would you just recommend to use a range, even though most of the IDs in the range will not be sending data?

Best regards,
Timo

timpur, 2017-11-02:
Interesting problem.

Firstly you can do what you said by just adding a level when you set property values. Homie doesn't care. Ive done this my self as a one off. Note its not the best way because you'll loose discoverability (future apps might use this?.
`lightNode.setProperty("<deviceID>/temperature").send(value);`

Also if you using mysensors, which im assuming you are, why not just set the id of your nodes [(MySesnsorsConfig)](https://www.mysensors.org/download/sensor_api_20#configuration)?

That would solve this problem and probs best to do in the long run.... Changing ids ids -> changing topics doesn't make sense, you'll never know where you device is to bind for automation apps????

timpur, 2017-12-16:
Reply needed to reopen

closed 2017-12-16 by timpur

tkosig, 2017-12-16:
Thanks for your reply and sorry for my delayed response. I think I might either do what you suggested - or - put the data into a JSON structure that has to be interpreted / parsed on the consumer side.

The different sensors send their ID, temperature, humidity and if applicable a low battery signal - so there is quite a bit more information to convey anyways.

As to the issue of the changing IDs - it is an annoyance and it doesn't play well with other apps binding to the topic, you're quite right. A good solution would be possibly be to allow for setting an alias which would then be published such as ID43 = sensor_bedroom. One would need to add a bit of functionality to manage that on the gateway (including a nice web frontend for assigning aliases) but nothing that can't be done if time is readily available (that is more my problem at the moment ;)).

BTW, I'm not using mysensors or any other framework for that matter. Just MQTT.

timpur, 2017-12-17:
Okay well options are there, up to you.

And you should use my sensors for RF, great lib (never uses but looks great for that stuff), would solve some of your ID issues and they have mqtt gateways.

Don't know what your really doing but if you have more questions ask.