homieiot/homie-esp8266 #410: Proposal: Connect ESP8266 via WPS_PBC

ThesisJT, 2017-10-17:
Hi together,

by reading the docs and setting up the fist device the project looks and works great.
One enhancement proposal from my side, maybe it is also interesting for you.

How about to edit the "fixed" SSID & Password by a small script which gets the information automatically via WPS_PBC? 
So it is easy to add the ESP8266 in a home-automation environment without editing manually the SSID & Password.
I have done a project on the ESP8266 which is able to connect via WPS_PBC in combination with a double reset detection (which starts WiFi.beginWPSConfig();).

If this is considered useful I could share the code and work together with you on the implementation.

Best regards,
JT

bodiroga, 2019-04-26:
Ummmm... I'm not sure we should support this, although it sounds great. WPS is vulnerable and the recommended practice is to disable it in every router: https://nakedsecurity.sophos.com/2015/04/13/we-told-you-not-to-use-wps-on-your-wi-fi-router-we-told-you-not-to-knit-your-own-crypto/

I wouldn't add it to homie-esp8266 3.0.0, sorry 😞

CWempe, 2019-04-26:
Keep in mind that this article is not talking about the Push Button Configuration (WPS PBC).

bodiroga, 2019-04-26:
> Keep in mind that this article is not talking about the Push Button Configuration (WPS PBC).

Good point @CWempe! Anyway, is it possible to activate the Push Button Configuration on the router if the global WPS option is disabled? I always disable the WPS configuration on my router and I don't know how it works :wink: