homieiot/homie-esp8266 #411: Saving configuration fails due to WiFi password with special characters?

tkosig, 2017-10-18:
Hi there,

I tried to configure my device running the latest development branch of Homie with a WiFi password that contains special characters.

Here's the password (I added some extra plain characters to be able to post it on here :))
hJgRHGfd76d%#sS66ev!tt@gag

Configuration fails with an error message that parsing the JSON failed or the JSON was too big.

I think I successfully connected a devboard running Homie 1.5.0 to the WiFi network using this password so maybe it's a bug in 2.0.0.

timpur, 2017-12-16:
interesting, might be an issue with Arduino JSON ?? ill try get to the root of this issue.

timpur, 2017-12-27:
@tkosig 
try updating your arduino json >= 5.10.0
if that dosent work try using the dev version and get back to me i your shill having these issues.

tripflex, 2018-01-20:
@timpur seems most of these may be related #386 #391

timpur, 2018-01-20:
This might be fixed, homie moved over to a different web server ...., Confirmation would be nice.

stritti, 2019-04-17:
Seems to be stale issue.
Will be closed.
Feel free to reopen if there are more information.

closed 2019-04-17 by stritti