homieiot/homie-esp8266 #413: Property Support is Incomplete

weswitt, 2017-10-24:
The Homie convention (https://github.com/marvinroger/homie) states that all properties require attributes such as $datatype, $unit and $format, but the current source for homie-esp8266 has no support for property attributes.

timpur, 2017-11-02:
Note the homie convention has been updated to v2.1.0. I believe homie for esp8266 is still using homie v2.0.0 convention. I hope also this will be updated soon.

bodiroga, 2017-12-30:
Hi @timpur!

First of all, many thanks for taking the lead of the project now that Marvin is so busy with his real life tasks. Thank you!

As far as I now, the new Homie convention will be named "Homie 3.0", here you have a reference: https://github.com/marvinroger/homie/issues/46. So perhaps the future version of homie-esp8266 should be named 3.0, don't you think? What's your opinion @marvinroger?

As a side note, will this project be compatible with ESP32 boards?

Thanks again and best regards,

Aitor

timpur, 2017-12-30:
Love the comment.
Yes I'm thinking the same thing about aligning the versions of the convention major version if changed to 3.0 (that said minor version numbers might not always be the same)

Might discuss version naming actually with @marvinroger

ESP32 support is planed in my mind, just waiting for some core libs to come to Arduino esp32 before we can start the conversion. Also want to get homie for esp82266 up to date first since esp32 will be a fork.
See

stritti, 2020-12-28:
I think this issue is outdated and close it.

closed 2020-12-28 by stritti