homieiot/homie-esp8266 #415: Support ESP32

theinein, 2017-10-27:
The issue tracker is a great place to ask for enhancements and to report bugs.
If you have some questions or if you need help, some people might help you on the [Gitter room](https://gitter.im/homie-iot/ESP8266).

Before submitting your issue, make sure:

- [ ] You've read the documentation for *your* release (in the `docs/` folder or at http://marvinroger.github.io/homie-esp8266/) which contains some answsers to the most common problems (notably the `Limitations and know issues` and `Troubleshooting` pages)
- [ ] You're using the examples bundled in *your* release, which are in the `examples/` folder of the `.zip` of the release you're using. Examples might not be backward-compatible

Thanks!

theinein, 2017-10-27:
When do you plan to add support for ESP32 in Homie? I’m willing to contribute if I can find the time, but would need to understand what changes need to be made to Homie.

bertmelis, 2017-10-27:
1. async tcp library: see https://github.com/me-no-dev/AsyncTCP
2. check what it takes to get https://github.com/marvinroger/async-mqtt-client compatible with point 1.
3. check what it takes to make Homie compatible with point 2

apo-mak, 2017-10-29:
there is a similar topic here https://github.com/marvinroger/homie-esp8266/issues/353

i will also love to see and help with support for esp32 on homie.
 Because i have upgrade most of my esp from 8266 to 32 an i cant use homie any more

bertmelis, 2017-10-29:
Let's first try to make the asyncmqtt compatible.  Then a ticker lib (requires some FreeRTOS knowledge). For the ticker, me-no-dev also did some work for it, but not yet release-ready. You you replace this by a millis()-polling design meanwhile. Homie's code itself is largely genetica.

theinein, 2017-10-30:
Is anyone aware if Espressif plans to contribute to either of these libraries?

bertmelis, 2017-11-06:
another roadblock: there's no webserver yet for the ESP32. So config-mode is not working. There's a ticket for at the Arduino-esp32-repo and as I understand we'll have to wait for a seperate repo for a (async?) webserver compatible with both esp8266 and esp32. Once it's there we can a have another look at it.

Anyway, an eventual esp32 implementation should belong in it's own repo as it is another implementation ,which will be largely the same but not quite.

theinein, 2017-11-06:
Thanks for all the great discussion. Clearly larger than I thought.

timpur, 2017-11-07:
It would be nice to start working on these issues :)

timpur, 2017-11-08:
@bertmelis Why doesn't homie use [ESPAsyncWebServer](https://github.com/me-no-dev/ESPAsyncWebServer) ?? Works for both esp32 and esp8266 ??

bertmelis, 2017-11-08:
I didn't know it was already available. It's on @marvinroger 's to do list I think.

But again, the esp32 version should belong in it's own repo IHMO. Now the codebase is still largely the same but maybe it will not be the case eventually (TLS/SSL for example).

timpur, 2017-11-08:
No i totally agree, it should be in a different repo !!! (most definitely should be). 

Just thinking if homie was changed to use ESPAsyncWebServer then porting homie to esp32 would be that much easier. Additionally it makes sense to use async where possible. 

We should try keep the same featured for both.

timpur, 2017-11-13:
I have moved homie over to `AsyncWebServer`. [https://github.com/timpur/homie-esp8266](https://github.com/timpur/homie-esp8266). Still testing and fixing up doc related stuff. When done ill make a pull request 

@marvinroger hope you think this is the right progress for homie.

bertmelis, 2017-11-13:
is _sendCROS() a typo? Did you mean CORS or is it on purpose?

bertmelis, 2017-11-13:
@timpur: you did the changes on the develop branch right? (so I clone the right repo)

timpur, 2017-11-13:
Yep all on the dev branch. I was going to test more before making a Pull request?

timpur, 2018-04-26:
@bertmelis @tripflex News !!!!
Last night at 1am (for me), i successfully ported the Homie to the esp32, Since it was so late for me i havnt commited any code, as i want to fix up, what is propably quite dogy code (beeing done at 1am....). Inital tests at 1am show that all seems to work as you'd expect. Still some stablity issues that need ironing out, but had the config mode work with the ui, and could successfully connect to mqtt... so a good start. Will do some cleaning up over the weekend and need to push some things to ESP32 Arduino that was missing, but im exited .....

There acutally wernt that many changes need to port homie over (more a couple of esentials from the arduino framwork), this begs the question, can we have one port for both esp8266 and esp32 ? This woiuld make mangaing code easier .... lets discus.

ESP32 here we come :)

timpur, 2018-04-26:
Update, done some testing and things seem quite stable. A couple of minor glitches but only in config mode... 

Will try to fix and get it ready for you guys to play with :)

nicola-lunghi, 2018-05-08:
https://github.com/craftmetrics/esp32-homie

maurostaiti, 2018-05-16:
Hi Timpur

thanks for your work

today i'm tryng to compile an homie example for esp32 , but the errors are 

ATTENZIONE: Cartella spuria .circleci nella libreria 'Homie'
Opzioni di compilazione cambiate, ricompilo tutto
ATTENZIONE: la libreria homie-esp8266-esp32 dichiara di funzionare sulle architetture (esp8266) e potrebbe non essere compatibile con la tua scheda che utilizza l'architettura (esp32)
ATTENZIONE: la libreria async-mqtt-client-master dichiara di funzionare sulle architetture (esp8266) e potrebbe non essere compatibile con la tua scheda che utilizza l'architettura (esp32)
In file included from /home/mauro/Scaricati/arduino-1.8.5/libraries/homie-esp8266-esp32/src/Homie/Datatypes/.././Utils/../.././Homie/Datatypes/Callbacks.hpp:4:0,
                 from /home/mauro/Scaricati/arduino-1.8.5/libraries/homie-esp8266-esp32/src/Homie/Datatypes/.././Utils/../../HomieSetting.hpp:6,
                 from /home/mauro/Scaricati/arduino-1.8.5/libraries/homie-esp8266-esp32/src/Homie/Datatypes/.././Utils/Validation.hpp:8,
                 from /home/mauro/Scaricati/arduino-1.8.5/libraries/homie-esp8266-esp32/src/Homie/Datatypes/../Config.hpp:11,
                 from /home/mauro/Scaricati/arduino-1.8.5/libraries/homie-esp8266-esp32/src/Homie/Datatypes/Interface.hpp:7,
                 from /home/mauro/Scaricati/arduino-1.8.5/libraries/homie-esp8266-esp32/src/Homie.hpp:6,
                 from /home/mauro/Scaricati/arduino-1.8.5/libraries/homie-esp8266-esp32/src/Homie.h:4,
                 from /home/mauro/Scaricati/arduino-1.8.5/libraries/homie-esp8266-esp32/examples/LightOnOff/LightOnOff.ino:1:
/home/mauro/Scaricati/arduino-1.8.5/libraries/homie-esp8266-esp32/src/Homie/Datatypes/.././Utils/../.././Homie/Datatypes/../../HomieEvent.hpp:31:3: error: 'WiFiDisconnectReason' does not name a type
   WiFiDisconnectReason wifiReason;
   ^
In file included from /home/mauro/Scaricati/arduino-1.8.5/libraries/homie-esp8266-esp32/src/Homie.hpp:12:0,
                 from /home/mauro/Scaricati/arduino-1.8.5/libraries/homie-esp8266-esp32/src/Homie.h:4,
                 from /home/mauro/Scaricati/arduino-1.8.5/libraries/homie-esp8266-esp32/examples/LightOnOff/LightOnOff.ino:1:
/home/mauro/Scaricati/arduino-1.8.5/libraries/homie-esp8266-esp32/src/Homie/Boot/BootNormal.hpp:74:3: error: 'WiFiEventHandler' does not name a type
   WiFiEventHandler _wifiGotIpHandler;
   ^
/home/mauro/Scaricati/arduino-1.8.5/libraries/homie-esp8266-esp32/src/Homie/Boot/BootNormal.hpp:75:3: error: 'WiFiEventHandler' does not name a type
   WiFiEventHandler _wifiDisconnectedHandler;
   ^
/home/mauro/Scaricati/arduino-1.8.5/libraries/homie-esp8266-esp32/src/Homie/Boot/BootNormal.hpp:96:27: error: 'WiFiEventStationModeGotIP' does not name a type
   void _onWifiGotIp(const WiFiEventStationModeGotIP& event);
                           ^
/home/mauro/Scaricati/arduino-1.8.5/libraries/homie-esp8266-esp32/src/Homie/Boot/BootNormal.hpp:97:34: error: 'WiFiEventStationModeDisconnected' does not name a type
   void _onWifiDisconnected(const WiFiEventStationModeDisconnected& event);
                                  ^
In file included from /home/mauro/Scaricati/arduino-1.8.5/libraries/homie-esp8266-esp32/src/Homie.hpp:12:0,
                 from /home/mauro/Scaricati/arduino-1.8.5/libraries/homie-esp8266-esp32/src/Homie.h:4,
                 from /home/mauro/Scaricati/arduino-1.8.5/libraries/homie-esp8266-esp32/examples/LightOnOff/LightOnOff.ino:1:
/home/mauro/Scaricati/arduino-1.8.5/libraries/homie-esp8266-esp32/src/Homie/Boot/BootNormal.hpp:107:59: error: 'UPDATE_ERROR_OK' was not declared in this scope
   void _endOtaUpdate(bool success, uint8_t update_error = UPDATE_ERROR_OK);
                                                           ^
In file included from /home/mauro/Scaricati/arduino-1.8.5/libraries/homie-esp8266-esp32/src/Homie.h:4:0,
                 from /home/mauro/Scaricati/arduino-1.8.5/libraries/homie-esp8266-esp32/examples/LightOnOff/LightOnOff.ino:1:
/home/mauro/Scaricati/arduino-1.8.5/libraries/homie-esp8266-esp32/src/Homie.hpp:73:49: error: 'RFMode' has not been declared
   static void doDeepSleep(uint32_t time_us = 0, RFMode mode = RF_DEFAULT);
                                                 ^
/home/mauro/Scaricati/arduino-1.8.5/libraries/homie-esp8266-esp32/src/Homie.hpp:73:63: error: 'RF_DEFAULT' was not declared in this scope
   static void doDeepSleep(uint32_t time_us = 0, RFMode mode = RF_DEFAULT);
                                                               ^
exit status 1
Errore durante la compilazione per la scheda ESP32 Dev Module.

I'm using 
this https://github.com/timpur/homie-esp8266/tree/esp32 
and this https://github.com/espressif/arduino-esp32
thanks in advance!

timpur, 2018-05-16:
@maurostaiti and everyone else, so sorry i some how forgot to push my changes to the esp32 branch ... awks. 

@maurostaiti, as your comment contains a long stack trace im going to remove, please repost if you have anymore issues.

maurostaiti, 2018-05-16:
hi

Unfortunately other errors occurs:

`ATTENZIONE: Cartella spuria .circleci nella libreria 'Homie'
ATTENZIONE: la libreria homie-esp8266-esp32 dichiara di funzionare sulle architetture (esp8266) e potrebbe non essere compatibile con la tua scheda che utilizza l'architettura (esp32)
ATTENZIONE: la libreria async-mqtt-client-master dichiara di funzionare sulle architetture (esp8266) e potrebbe non essere compatibile con la tua scheda che utilizza l'architettura (esp32)
In file included from /home/mauro/Scaricati/arduino-1.8.5/libraries/homie-esp8266-esp32/src/./Homie/Datatypes/Interface.hpp:7:0,
                 from /home/mauro/Scaricati/arduino-1.8.5/libraries/homie-esp8266-esp32/src/HomieSetting.hpp:7,
                 from /home/mauro/Scaricati/arduino-1.8.5/libraries/homie-esp8266-esp32/src/HomieSetting.cpp:1:
/home/mauro/Scaricati/arduino-1.8.5/libraries/homie-esp8266-esp32/src/./Homie/Datatypes/../Config.hpp: In instantiation of 'HomieInternals::ValidationResult HomieInternals::Config::saveSetting(const char*, T) [with T = bool]':
/home/mauro/Scaricati/arduino-1.8.5/libraries/homie-esp8266-esp32/src/HomieSetting.cpp:54:89:   required from 'bool HomieSetting<T>::set(T, bool) [with T = bool]'
/home/mauro/Scaricati/arduino-1.8.5/libraries/homie-esp8266-esp32/src/HomieSetting.cpp:121:16:   required from here
/home/mauro/Scaricati/arduino-1.8.5/libraries/homie-esp8266-esp32/src/./Homie/Datatypes/../Config.hpp:82:66: error: invalid initialization of non-const reference of type 'ArduinoJson::JsonObject&' from an rvalue of type 'ArduinoJson::JsonObject'
   JsonObject& settings = configObject["settings"].as<JsonObject>();
                                                                  ^
/home/mauro/Scaricati/arduino-1.8.5/libraries/homie-esp8266-esp32/src/./Homie/Datatypes/../Config.hpp: In instantiation of 'HomieInternals::ValidationResult HomieInternals::Config::saveSetting(const char*, T) [with T = long int]':
/home/mauro/Scaricati/arduino-1.8.5/libraries/homie-esp8266-esp32/src/HomieSetting.cpp:54:89:   required from 'bool HomieSetting<T>::set(T, bool) [with T = long int]'
/home/mauro/Scaricati/arduino-1.8.5/libraries/homie-esp8266-esp32/src/HomieSetting.cpp:122:16:   required from here
/home/mauro/Scaricati/arduino-1.8.5/libraries/homie-esp8266-esp32/src/./Homie/Datatypes/../Config.hpp:82:66: error: invalid initialization of non-const reference of type 'ArduinoJson::JsonObject&' from an rvalue of type 'ArduinoJson::JsonObject'
/home/mauro/Scaricati/arduino-1.8.5/libraries/homie-esp8266-esp32/src/./Homie/Datatypes/../Config.hpp: In instantiation of 'HomieInternals::ValidationResult HomieInternals::Config::saveSetting(const char*, T) [with T = double]':
/home/mauro/Scaricati/arduino-1.8.5/libraries/homie-esp8266-esp32/src/HomieSetting.cpp:54:89:   required from 'bool HomieSetting<T>::set(T, bool) [with T = double]'
/home/mauro/Scaricati/arduino-1.8.5/libraries/homie-esp8266-esp32/src/HomieSetting.cpp:123:16:   required from here
/home/mauro/Scaricati/arduino-1.8.5/libraries/homie-esp8266-esp32/src/./Homie/Datatypes/../Config.hpp:82:66: error: invalid initialization of non-const reference of type 'ArduinoJson::JsonObject&' from an rvalue of type 'ArduinoJson::JsonObject'
/home/mauro/Scaricati/arduino-1.8.5/libraries/homie-esp8266-esp32/src/./Homie/Datatypes/../Config.hpp: In instantiation of 'HomieInternals::ValidationResult HomieInternals::Config::saveSetting(const char*, T) [with T = const char*]':
/home/mauro/Scaricati/arduino-1.8.5/libraries/homie-esp8266-esp32/src/HomieSetting.cpp:54:89:   required from 'bool HomieSetting<T>::set(T, bool) [with T = const char*]'
/home/mauro/Scaricati/arduino-1.8.5/libraries/homie-esp8266-esp32/src/HomieSetting.cpp:124:16:   required from here
/home/mauro/Scaricati/arduino-1.8.5/libraries/homie-esp8266-esp32/src/./Homie/Datatypes/../Config.hpp:82:66: error: invalid initialization of non-const reference of type 'ArduinoJson::JsonObject&' from an rvalue of type 'ArduinoJson::JsonObject'
In file included from /home/mauro/Scaricati/arduino-1.8.5/libraries/ArduinoJson/src/../include/ArduinoJson/JsonObject.ipp:11:0,
                 from /home/mauro/Scaricati/arduino-1.8.5/libraries/ArduinoJson/src/../include/ArduinoJson/JsonObject.hpp:152,
                 from /home/mauro/Scaricati/arduino-1.8.5/libraries/ArduinoJson/src/../include/ArduinoJson.h:10,
                 from /home/mauro/Scaricati/arduino-1.8.5/libraries/ArduinoJson/src/ArduinoJson.h:14,
                 from /home/mauro/Scaricati/arduino-1.8.5/libraries/homie-esp8266-esp32/src/./Homie/Datatypes/../Config.hpp:4,
                 from /home/mauro/Scaricati/arduino-1.8.5/libraries/homie-esp8266-esp32/src/./Homie/Datatypes/Interface.hpp:7,
                 from /home/mauro/Scaricati/arduino-1.8.5/libraries/homie-esp8266-esp32/src/HomieSetting.hpp:7,
                 from /home/mauro/Scaricati/arduino-1.8.5/libraries/homie-esp8266-esp32/src/HomieSetting.cpp:1:
/home/mauro/Scaricati/arduino-1.8.5/libraries/ArduinoJson/src/../include/ArduinoJson/JsonObjectSubscript.hpp: In instantiation of 'TValue ArduinoJson::JsonObjectSubscript<TKey>::as() const [with TValue = ArduinoJson::JsonObject; TKey = const char*]':
/home/mauro/Scaricati/arduino-1.8.5/libraries/homie-esp8266-esp32/src/./Homie/Datatypes/../Config.hpp:82:65:   required from 'HomieInternals::ValidationResult HomieInternals::Config::saveSetting(const char*, T) [with T = bool]'
[... cut ...]
/home/mauro/Scaricati/arduino-1.8.5/libraries/ArduinoJson/src/../include/ArduinoJson/JsonObject.hpp:42:25: note: candidate: ArduinoJson::JsonObject::JsonObject(ArduinoJson::JsonBuffer*)
   FORCE_INLINE explicit JsonObject(JsonBuffer* buffer)

                         ^
/home/mauro/Scaricati/arduino-1.8.5/libraries/ArduinoJson/src/../include/ArduinoJson/JsonObject.hpp:42:25: note:   candidate expects 1 argument, 0 provided
exit status 1
Errore durante la compilazione per la scheda Adafruit ESP32 Feather.``

maurostaiti, 2018-05-16:
Hi Tim
Please ignore my previous post. i've managet to solve them , was my json lib out of date.

thanks for your hard work
Mauro

justClouds, 2018-11-24:
Any updates on the status of the ESP32 implementation? I noticed the https://github.com/timpur/homie-esp8266/tree/esp32 branch hasn't been updated since May, so just wondering if it is still maintained?

timpur, 2018-11-26:
sadly not, im no longer really working on this project sadly, hopefully some one will like to pick it up ?

dmartinpro, 2019-03-18:
Also really/very interested in ESP32 support as an ESP32 board makes more sense now than a less versatile ESP8266 board.
Is it possible to merge the Timpur's work once for all or is there any blocking things here and there?
Could be really terrific to have Homie 2.0/3.0 on these boards.

euphi, 2019-03-18:
The [#develop-v3](https://github.com/homieiot/homie-esp8266/tree/develop-v3) branch supports ESP32 since some days.

euphi, 2019-03-18:
The develop-v3 branch supports the convention 3.0.1. The convention is not completely backward-compatible, but for almost all relevant usecases.
There are also some API changes in the homie-esp8266 implementation. However, it is already quite stable, so you can give it a try.

closed 2019-03-18 by euphi

lorenwest, 2019-03-18:
As a major version change, maybe it's time to merge into master?

mike-yesware, 2019-03-18:
I've been lurking here meaning to post a question like @lorenwest's for a while. Seems like:

* There hasn't been a release for quite a while, so something like Platform.io is quite out of date
* `master` is quite stable, and can be used without much worry
* `develop` is ahead of `master`, but compatible with the API there?
* develop-v3` is a new API, so not compatible?

TBH, the last two are most confusing to me as it seems like `homie-esp8266` should be versioned itself, and it also wants to support two different versions of the Homie spec. At least, I think that's the current intention of the two branches.

Would there be interest in `stable-v2` and a `stable-v3` (or, just `master`) branches that releases can be cut from? I'm quite happy to help with any of the release engineering work if that's agreeable.

Please let me know if I've gotten any of this wrong

lorenwest, 2019-03-18:
@mike-yesware Thanks for bringing this up. While I don't have a ton of time to invest in this I'd be happy to work with you as another pair of eyes, and to help with validation.

mike-yesware, 2019-03-18:
Actually, my comment feels like a hijack of this thread, so I moved it to: https://github.com/homieiot/homie-esp8266/issues/587