homieiot/homie-esp8266 #417: Problems parsing preloaded config.json

duke1swd, 2017-11-01:
I'm trying to configure my homie 2.0 beta v1 project by creating a SPIFFS file system with a config.json in it an uploading it using the Arduino SPIFFS uploader.

If I use the example config.json [here](http://marvinroger.github.io/homie-esp8266/2.0.0-beta.1/configuration/json-configuration-file/) everything works.   However, if I remove any line from that file, it fails with this on the serial port:

`✖ Invalid JSON in the config file`
`Configuration invalid. Using CONFIG MODE`

I confess that I've not yet tried uploading these via the config API.

Two requests:
1) I'd like to be able to remove the parts of the file that I don't need or want, such as the static IP address.

2) I'd like the Invalid JSON message to be more descriptive.

Thanks!

n8twj, 2017-11-02:
Hi - have you ran your resulting JSON thru validation to make sure it's
actually valid?
(/me ducks)

duke1swd, 2017-11-02:
OK.  Now I feel like an idiot.
My sincere apologies for wasting people's time.
It is working perfectly now that I removed one dangling comma.

duke1swd, 2017-11-02:
UBD

closed 2017-11-02 by duke1swd