homieiot/homie-esp8266 #418: Project - HomieManagement

timpur, 2017-11-03:
Hi community, ive been working on this in my spare time. Its a manager for Homie, based on v 2.0 and esp8266....

So far all it can do is view current device stats, change device config and flash firmware, all over mqtt.

Have a look, its a work in progress. Don't expect much.

[https://github.com/timpur/HomieManagement](https://github.com/timpur/HomieManagement)

amayii0, 2017-11-07:
(bookmarked :) in todo list of things to check)

closed 2017-11-30 by timpur

marvinroger, 2017-11-30:
Oh, @timpur, feel free to add that to the https://github.com/marvinroger/homie/blob/redesign/implementations. 😉

timpur, 2017-11-30:
yeah i think i will when i work on it a bit more and there arnt so many bugs :P (soon)