homieiot/homie-esp8266 #423: Question on code: online/offline

psyciknz, 2017-11-13:
While not an issue, hoping someone can help.

How is the $online false set if not driven use user code.  Ie what sets the online state from true to false when the device has disconnected/lost power?  Can someone explain?  I'd like to replicate in code for something else, so it uses the homie model of showing is a device is connected.

n8twj, 2017-11-13:
Hi - I believe it has something to do with the "Last Will and Testament"
concept from MQTT, but I cannot quickly locate an example

Typos courtesy of my iPhone

psyciknz, 2017-11-13:
I saw the posting of the true status in the boot-normal.cpp.   And it looked like it was retained.  The 1 before retained I thought was the at least once.

But I replicated that in c# and the true was set but it didn’t set to false when I exited.

psyciknz, 2017-11-13:
Figured it out, you helped with the LWT guidance.  So for the client I was using, setting the last will and testament was part of the connection.  And you set the $online to false as LWT.

Then as part of connecting you create a $online true.  And so on disconnection (forcibly) the LWT takes effect.

Thanks, interesting exercise.

closed 2017-11-13 by psyciknz