homieiot/homie-esp8266 #427: Simple bash script to clean homie device from MQTT (Mosquitto)

andreasgruener, 2017-11-18:
Works great for me to get rid of devices in MQTT, which are not active anymore.
Maybe it is helpful for someone else...

Output: 

> 
> Clearing for device a020a617ea1d all messages, last clean will block kill with Ctrl+C when nothing happens anymore.
> 
> STARTING :: Checking for valid topics one after the other....
>   0 Checking for Topic ....
>     homie/a020a617ea1d/temperature/$type >> sending empty payload to topic with retained flag
>   1 Checking for Topic ....
>     homie/a020a617ea1d/temperature/$properties >> sending empty payload to topic with retained flag
>   2 Checking for Topic ....
>     homie/a020a617ea1d/temperature/unit >> sending empty payload to topic with retained flag
> 

```
#!/bin/bash
#
# Simple Script to remove all reatined messages for a homie device

echo 
echo "Clearing for device $1 all messages, last clean will block kill with Ctrl+C when nothing happens anymore."
echo 
#mosquitto_sub -v -h 10.0.2.2 -t "homie/$1/#" -C $2
echo "STARTING :: Checking for valid topics one after the other...."

MAX_TOPICS_TO_CHECK=100

for c in `seq 0 $MAX_TOPICS_TO_CHECK`
    do
    echo "  $c Checking for Topic ...."
    for i in `mosquitto_sub -v -h 10.0.2.2 -t "homie/$1/#" -C 1|cut -d' ' -f1`
        do  
            # echo "mosquitto_pub -h 10.0.2.2 -r -n -t \"`echo $i | sed s#'\\$'#'\\\\\\\$'#g`\"" 
            echo "    $i >> sending empty payload to topic with retained flag"
            mosquitto_pub -h 10.0.2.2 -r -n -t "`echo $i | sed s#'\\$'#'\\\$'#g`" 
        done
done
```

timpur, 2017-12-16:
I like this, ill include it in my homie management project

luebbe, 2017-12-16:
There is also this project: https://github.com/contactless/mqtt-tools, which offers a similar function

timpur, 2017-12-16:
@andreasgruener if you want, make a little mark down page about this script and ill include it in the docs ?
how to use and the code ?

timpur, 2017-12-16:
@luebbe , worth also including in the docs ? maybe write a little doc page for it (gist, like ^) ?

luebbe, 2017-12-18:
@timpur do you mean: "add a reference to the tool to the homie docs? or "Please write some documentation about this tool"?.
It's not my tool, the "docs" there are only the command line help and that's it. It's a bit trial and error when you use the scripts...

andreasgruener, 2018-02-12:
@timpur sorry for the delay!
I added some comments to the code and as comment on the [gist](https://gist.github.com/andreasgruener/9935ffecbc59404375e7a9cb4a9a8812)

timpur, 2018-02-12:
On my to do list.

stritti, 2019-03-30:
I added reference within trouble shooting section. 
Thanks @andreasgruener for the script.

closed 2019-03-30 by stritti