homieiot/homie-esp8266 #428: Not able to compile with develop branch

johnmay1, 2017-11-18:
Hello,

I am trying to build homie node from the "develop" branch using getting started document for platformio.But the code is not compiling and throwing following error. Looks like latest version of "ESPAsyncTCP" is broken. Please advise. 

odemcuv2 (platform: espressif8266; lib_deps: git+https://github.com/marvinroger/homie-esp8266.git#develop; board: nodemcuv2; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Collected 34 compatible libraries
Looking for dependencies...
Library Dependency Graph
|-- <Homie> v2.0.0
|   |-- <ArduinoJson> v5.11.2
|   |-- <AsyncMqttClient> v0.8.1
|   |   |-- <ESPAsyncTCP> v1.1.2
|   |   |   |-- <ESP8266WiFi> v1.0
|   |-- <Bounce2> v2.3
|   |-- <ESPAsyncWebServer> v1.1.0
|   |   |-- <ESPAsyncTCP> v1.1.2
|   |   |   |-- <ESP8266WiFi> v1.0
|   |   |-- <ESP8266WiFi> v1.0
|   |   |-- <ArduinoJson> v5.11.2
|   |-- <ESP8266mDNS>
|   |   |-- <ESP8266WiFi> v1.0
|   |-- <ESP8266HTTPClient> v1.1
|   |   |-- <ESP8266WiFi> v1.0
|   |-- <ESPAsyncTCP> v1.1.2
|   |   |-- <ESP8266WiFi> v1.0
|   |-- <DNSServer> v1.1.0
|   |   |-- <ESP8266WiFi> v1.0
|   |-- <Ticker> v1.0
|   |-- <ESP8266WiFi> v1.0
Compiling .pioenvs/nodemcuv2/src/main.o
Archiving .pioenvs/nodemcuv2/libFrameworkArduinoVariant.a
Compiling .pioenvs/nodemcuv2/FrameworkArduino/Esp.o
Compiling .pioenvs/nodemcuv2/FrameworkArduino/FS.o
Compiling .pioenvs/nodemcuv2/FrameworkArduino/HardwareSerial.o
Compiling .pioenvs/nodemcuv2/FrameworkArduino/IPAddress.o
Compiling .pioenvs/nodemcuv2/FrameworkArduino/MD5Builder.o
Compiling .pioenvs/nodemcuv2/FrameworkArduino/Print.o
Indexing .pioenvs/nodemcuv2/libFrameworkArduinoVariant.a
Compiling .pioenvs/nodemcuv2/FrameworkArduino/Schedule.o
Compiling .pioenvs/nodemcuv2/FrameworkArduino/Stream.o
Compiling .pioenvs/nodemcuv2/FrameworkArduino/StreamString.o
Compiling .pioenvs/nodemcuv2/FrameworkArduino/Tone.o
Compiling .pioenvs/nodemcuv2/FrameworkArduino/Updater.o
Compiling .pioenvs/nodemcuv2/FrameworkArduino/WMath.o
Compiling .pioenvs/nodemcuv2/FrameworkArduino/WString.o
Compiling .pioenvs/nodemcuv2/FrameworkArduino/abi.o
Compiling .pioenvs/nodemcuv2/FrameworkArduino/base64.o
Compiling .pioenvs/nodemcuv2/FrameworkArduino/cbuf.o
Compiling .pioenvs/nodemcuv2/FrameworkArduino/cont.o
Compiling .pioenvs/nodemcuv2/FrameworkArduino/cont_util.o
Compiling .pioenvs/nodemcuv2/FrameworkArduino/core_esp8266_eboot_command.o
Compiling .pioenvs/nodemcuv2/FrameworkArduino/core_esp8266_flash_utils.o
Compiling .pioenvs/nodemcuv2/FrameworkArduino/core_esp8266_i2s.o
Compiling .pioenvs/nodemcuv2/FrameworkArduino/core_esp8266_main.o
Compiling .pioenvs/nodemcuv2/FrameworkArduino/core_esp8266_noniso.o
Compiling .pioenvs/nodemcuv2/FrameworkArduino/core_esp8266_phy.o
Compiling .pioenvs/nodemcuv2/FrameworkArduino/core_esp8266_postmortem.o
Compiling .pioenvs/nodemcuv2/FrameworkArduino/core_esp8266_si2c.o
Compiling .pioenvs/nodemcuv2/FrameworkArduino/core_esp8266_timer.o
Compiling .pioenvs/nodemcuv2/FrameworkArduino/core_esp8266_wiring.o
Compiling .pioenvs/nodemcuv2/FrameworkArduino/core_esp8266_wiring_analog.o
Compiling .pioenvs/nodemcuv2/FrameworkArduino/core_esp8266_wiring_digital.o
Compiling .pioenvs/nodemcuv2/FrameworkArduino/core_esp8266_wiring_pulse.o
Compiling .pioenvs/nodemcuv2/FrameworkArduino/core_esp8266_wiring_pwm.o
Compiling .pioenvs/nodemcuv2/FrameworkArduino/core_esp8266_wiring_shift.o
Compiling .pioenvs/nodemcuv2/FrameworkArduino/debug.o
Compiling .pioenvs/nodemcuv2/FrameworkArduino/heap.o
Compiling .pioenvs/nodemcuv2/FrameworkArduino/libb64/cdecode.o
[... 109 lines cut ...]
In file included from .piolibdeps/ESPAsyncTCP_ID305/src/ESPAsyncTCP.cpp:27:0:
/Users/hemaltandel/.platformio/packages/framework-arduinoespressif8266/tools/sdk/lwip/include/lwip/tcp.h:319:18: error:   initializing argument 2 of 'void tcp_sent(tcp_pcb*, tcp_sent_fn)' [-fpermissive]
void             tcp_sent    (struct tcp_pcb *pcb, tcp_sent_fn sent)ICACHE_FLASH_ATTR;
^
.piolibdeps/ESPAsyncTCP_ID305/src/ESPAsyncTCP.cpp:294:31: error: invalid conversion from 'long int (*)(void*, tcp_pcb*)' to 'tcp_poll_fn {aka signed char (*)(void*, tcp_pcb*)}' [-fpermissive]
tcp_poll(_pcb, &_s_poll, 1);
^
In file included from .piolibdeps/ESPAsyncTCP_ID305/src/ESPAsyncTCP.cpp:27:0:
/Users/hemaltandel/.platformio/packages/framework-arduinoespressif8266/tools/sdk/lwip/include/lwip/tcp.h:320:18: error:   initializing argument 2 of 'void tcp_poll(tcp_pcb*, tcp_poll_fn, u8_t)' [-fpermissive]
void             tcp_poll    (struct tcp_pcb *pcb, tcp_poll_fn poll, u8_t interval)ICACHE_FLASH_ATTR;
^
.piolibdeps/ESPAsyncTCP_ID305/src/ESPAsyncTCP.cpp: In member function 'void AsyncServer::begin()':
.piolibdeps/ESPAsyncTCP_ID305/src/ESPAsyncTCP.cpp:848:30: error: invalid conversion from 'long int (*)(void*, tcp_pcb*, long int)' to 'tcp_accept_fn {aka signed char (*)(void*, tcp_pcb*, signed char)}' [-fpermissive]
tcp_accept(_pcb, &_s_accept);
^
In file included from .piolibdeps/ESPAsyncTCP_ID305/src/ESPAsyncTCP.cpp:27:0:
/Users/hemaltandel/.platformio/packages/framework-arduinoespressif8266/tools/sdk/lwip/include/lwip/tcp.h:317:18: error:   initializing argument 2 of 'void tcp_accept(tcp_pcb*, tcp_accept_fn)' [-fpermissive]
void             tcp_accept  (struct tcp_pcb *pcb, tcp_accept_fn accept) ICACHE_FLASH_ATTR;
^
Compiling .pioenvs/nodemcuv2/lib/ESPAsyncTCP_ID305/tcp_axtls.o
*** [.pioenvs/nodemcuv2/lib/ESPAsyncTCP_ID305/ESPAsyncTCP.o] Error 1
============================================================= [ERROR] Took 10.49 seconds =============================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

bertmelis, 2017-11-18:
you can try using the staging version of the esp8266 platform:
put = https://github.com/platformio/platform-espressif8266.git#feature/stage in your platformio.ini

johnmay1, 2017-11-18:
@bertmelis  Thanks for quick response. Now i am getting following error,

ompiling .pioenvs/nodemcuv2/lib/ESPAsyncWebServer_ID306/AsyncEventSource.o
Compiling .pioenvs/nodemcuv2/lib/ESPAsyncWebServer_ID306/AsyncWebSocket.o
.piolibdeps/ESPAsyncWebServer_ID306/src/AsyncWebSocket.cpp:40:18: fatal error: Hash.h: No such file or directory
#include <Hash.h>
^
compilation terminated.
*** [.pioenvs/nodemcuv2/lib/ESPAsyncWebServer_ID306/AsyncWebSocket.o] Error 1
============================================================= [ERROR] Took 18.46 seconds =============================================================
The terminal process terminated with exit code: 1

bertmelis, 2017-11-18:
Strange, because Hash.h is part of the Arduino core. So it is not some exotische dependency.

Maybe a clean&build?
Or try empty the platforms folder in the .platformio folder.

You're on Windows?

johnmay1, 2017-11-18:
I did clean and build doesn't help. Uninstalled platformio and re-install, still same error.

I am on Mac.

johnmay1, 2017-11-18:
Added Hash as one of the dependency,
lib_deps = git+https://github.com/marvinroger/homie-esp8266.git#develop
  Hash

It fixed the issue.

Found from [https://github.com/me-no-dev/ESPAsyncWebServer/issues/237](https://github.com/me-no-dev/ESPAsyncWebServer/issues/237)

closed 2017-11-18 by johnmay1

kylegordon, 2017-11-24:
I'm also having difficulty getting it to compile with the latest changes in either ESPAsyncTCP or Core...

On PIO 3.4.1, pio run installs the libs and platform, but results in this...

> espressif8266_stage @ 9628ad3 is already installed
The platform 'https://github.com/platformio/platform-espressif8266.git#feature/stage' has been successfully installed!
The rest of packages will be installed automatically depending on your build environment.
Error: Unknown development platform 'https://github.com/platformio/platform-espressif8266.git#feature/stage'

Entire contents of platformio.ini file is this

> [env:nodemcu]
; platform = espressif8266
platform = https://github.com/platformio/platform-espressif8266.git#feature/stage
framework = arduino
board = nodemcu
lib_extra_dirs = src/
; lib_deps = Homie
lib_deps = git+https://github.com/marvinroger/homie-esp8266.git#develop

At a bit of a loss now :-)

bertmelis, 2017-11-24:
What error do you see?

kylegordon, 2017-11-24:
Just what was posted.

> PlatformManager: Installing platform-espressif8266
espressif8266_stage @ 9628ad3 is already installed
The platform 'https://github.com/platformio/platform-espressif8266.git#feature/stage' has been successfully installed!
The rest of packages will be installed automatically depending on your build environment.
Error: Unknown development platform 'https://github.com/platformio/platform-espressif8266.git#feature/stage'

marvinroger, 2017-11-29:
pio build is now fixed on develop. Thanks for the report. 👍