homieiot/homie-esp8266 #429: Unable to compile (Wifi plug firmware)

Dwgystyl, 2017-11-18:
I'm working through http://www.hagensieker.com/blog/page/?post_id=44&title=esp8266-hack-of-inexpensive-wifi-outlet

and when it tries to compile i get the following. 

C:\Users\Username\Documents\Arduino\libraries\homie-esp8266-develop\src\Homie\Boot\BootConfig.cpp: In member function 'virtual void HomieInternals::BootConfig::setup()':
C:\Users\Username\Documents\Arduino\libraries\homie-esp8266-develop\src\Homie\Boot\BootConfig.cpp:48:39: error: passing 'const IPAddress' as 'this' argument of 'String IPAddress::toString()' discards qualifiers [-fpermissive]

   _apIpStr = ACCESS_POINT_IP.toString();

exit status 1
Error compiling for board Generic ESP8266 Module.

I can track down the exact line it is calling too, but I am not familiar enough to figure out what or why the problem is.

timpur, 2017-11-19:
Thats interesting have what version of esp8266 Arduino core are you using ?

ngalfas, 2017-11-20:
I have exaxtly the same error.    I have Arduino IDE 1.8.2   

I downloaded the Homie zip from https://github.com/marvinroger/homie-esp8266/releases   which points to file https://github.com/marvinroger/homie-esp8266/archive/v2.0.0-beta.1.zip

Then i had the problem with latest ArduinoJSON incopatibility, the i downloaded the latest github version which corrected those errrors.

Now, i tried some older versions of ArduinoJSON but no luck, i still getting this error.
C:\Users\.................\Documents\Arduino\libraries\homie-esp8266-develop\src\Homie\Boot\BootConfig.cpp: In member function 'virtual void HomieInternals::BootConfig::setup()':

C:\Users\..............\Documents\Arduino\libraries\homie-esp8266-develop\src\Homie\Boot\BootConfig.cpp:48:39: error: passing 'const IPAddress' as 'this' argument of 'String IPAddress::toString()' discards qualifiers [-fpermissive]

   _apIpStr = ACCESS_POINT_IP.toString();

exit status 1
Error compiling for board WeMos D1 R2 & mini.

timpur, 2017-11-20:
@ngalfas if you are truly using "https://github.com/marvinroger/homie-esp8266/archive/v2.0.0-beta.1.zip" then `_apIpStr = ACCESS_POINT_IP.toString();` can not be a problem since this is from a recent commit. v2.0.0-beta.1.zip does not contain this change. so you must be using a recent version.

@Dwgystyl can you try using is fork (https://github.com/timpur/homie-esp8266) of homie as it undoes the change that you are having an error with. Just note there are other changes in this fork. So if you have other errors it would be good to know. note youll need https://github.com/me-no-dev/ESPAsyncWebServer

also if you want to use the latest most stable version of homie (dev), use the commit before these breaking changes (https://github.com/marvinroger/homie-esp8266/tree/715ad3f83f2c0667916a3373bd46adc3eedeb88c) sorry for this it will be fixed soon.

ngalfas, 2017-11-20:
As i said, i pulled the develop later and that's when i see this error.   After your remarks i went back to the beta-1  and now it's working.

So, to sumarize my exprerience.
The v2.0.0-beta.1.zip  is the best "stable" version,
BUT YOU MUST STICK TO 5.10 version of ArduinoJSON library.  The newer version gives another compilation error that you know of.

timpur, 2017-11-20:
Interesting to see this issue with esp8266 core 2.3.0, there is a fix coming. Sorry for the inconvenience, but seems updating your core will get around this for now.

ngalfas, 2017-11-20:
no, the core version has nothing to do with the first problem.  my bad for the confusion, i only said that i also tested with 2.4.0-rc2.

The main thing to remember and to help other is only this.

> The v2.0.0-beta.1.zip is the best "stable" version, BUT YOU MUST STICK TO 5.10 version of ArduinoJSON library. The newer 5.11 ArduinoJSON gives another compilation error that you know of.

Dwgystyl, 2017-11-30:
@timpur -  fork was successful,  I was able to compile the code and upload it to the esp without any other issues..  Thanks.

timpur, 2017-11-30:
Good to hear. About to commit the pull request for this fork. Adds a new deep sleep function you can use as a replacement for `esp.deepsleep()` recommend as this does some stuff to finalise the sleep like flush the serial so everything is printed before deep sleep happens. Mode details part of #432 `void doDeepSleep(uint32_t time_us = 0, RFMode mode = RF_DEFAULT);`

marvinroger, 2017-12-05:
Fixed in https://github.com/marvinroger/homie-esp8266/releases/tag/v2.0.0-beta.3
Thanks @timpur !

closed 2017-12-05 by marvinroger

euphi, 2017-12-06:
With beta.3 I get:

```
Compiling .pioenvs/esp01/lib/homie-esp8266/Homie.o
In file included from /home/ian/Coding/ESP_Libs/homie-esp8266/src/Homie/Datatypes/../Config.hpp:8:0,
from /home/ian/Coding/ESP_Libs/homie-esp8266/src/Homie/Datatypes/Interface.hpp:8,
from /home/ian/Coding/ESP_Libs/homie-esp8266/src/Homie.hpp:7,
from /home/ian/Coding/ESP_Libs/homie-esp8266/src/Homie.cpp:1:
/home/ian/Coding/ESP_Libs/homie-esp8266/src/Homie/Datatypes/../Utils/Validation.hpp:13:22: fatal error: Limits.hpp: No such file or directory
#include "Limits.hpp"
^
compilation terminated.
*** [.pioenvs/esp01/lib/homie-esp8266/Homie.o] Error 1
==================================================== [ERROR] Took 15.74 seconds ====================================================

```

timpur, 2017-12-07:
@euphi Dont know whats wrong there. Everything looks fine in dev source

euphi, 2017-12-07:
Oh.. my bad. I found a symlink to an old homie-version in the lib subdirectory.... :scream: 

However, now I have the #444 issue....