homieiot/homie-esp8266 #430: Problem with MQTT connection

bobkummerfeld, 2017-11-19:
I am attempting to use Homie on a Wemos D1 mini.  On my work computer (Mac) I installed the latest Arduino IDE and all the required libraries for Homie 2.0.0. Everything compiles but with some warnings (or "notes"). The LightOnOff program compiles and downloads ok and I can configure it via the web interface. It then works correctly, connecting to wifi and MQTT server.

So, I decide to continue working at home on a similar Mac running the same version of MacOS. I installed the latest Arduino IDE, libraries etc. All looks good until the Wemos attempts to connect to the MQTT server and just hangs.

I've checked that all libraries are the same version on each machine. I've taken the same Wemos D1 to work and installed the LightOnOff program, taken it back home and it works. I can't find the difference between the two systems.

Is there more detailed logging info I could enable to see why the MQTT connection is hanging?
Last message on the serial port is "Attempting to connect to MQTT..."

bobkummerfeld, 2017-11-19:
Update:  it works ok if I use an IP address rather than a domain name for the MQTT server name.

Does this point to a problem with ESP AsyncTCP? or lower level?

timpur, 2017-11-19:
That's not really enough to go off by. We need to know if you were using a different network? Different location? The difference between the 2 tests ( the one that worked and the one that didn't)

bobkummerfeld, 2017-11-19:
Different network and different location. Same configuration.  Only difference was that code compiled/loaded on one system doesn't work (ie MQTT connection hangs) in either place, and code compiled/loaded on the other system works in both places.  I've compared all library versions and they are the same.  So now I'm in the process of installing Platformio.

timpur, 2017-11-19:
Same Arduino core version ?
Verry intresting.

bobkummerfeld, 2017-11-20:
More info:  the call to WiFi.hostByName(host, remote_addr) in ESPAsyncTCP fails, so the connect call returns false.  Also, in a small test prog I tested name resolution using the ESP8266WiFi connect call and it worked ok.

timpur, 2017-11-20:
try an EspAsyncTCP example from https://github.com/me-no-dev/ESPAsyncTCP and see if your getting the same problem, try just connecting to website or something. if your still getting an issue then it might be a problem with EspAsyncTCP, in which you'll need to raise the issue there.

timpur, 2017-12-16:
any activity? will close soon.

closed 2019-03-30 by stritti