homieiot/homie-esp8266 #431: Not able to reset config

mishoboss, 2017-11-19:
Hi guys,
I don't know if its a bug or its me unable to figure out how to reset config and boot ESP into configuration mode from normal mode.

Here's what I do:
- I have a dedicated button and I use a button library that calls events on buttons presses, hold, long holds, etc. So I don't use the `Homie.setResetTrigger()` function in setup.
- This button press just sets a boolean flag `resetFlag`
- This is my loop
```if(resetFlag){
    Homie.reset();
  }
  Homie.loop();
  ... other stuff in loop
  if(resetFlag){
    ESP.restart(); 
  }```

The device restarts, but it doesn't boot in configuration mode. It is in normal mode and connects successfully to WiFi again. I tried some kund-fu with `Homie.setIdle(false/true)` but its still the same result. I'm using latest git code from yesterday, 18th Nov.

bertmelis, 2017-11-19:
no need to call ESP.restart(). Homie wil do that for you afaik.

mishoboss, 2017-11-19:
Well, it doesn't. :(

mishoboss, 2017-11-19:
It works! Sorry about that. I removed the `ESP.restart(); ` and all works good now. Please remove this issue.

closed 2017-11-19 by mishoboss