homieiot/homie-esp8266 #433: Feature Request +Config Mode can also edit current config

timpur, 2017-11-20:
Wondering if you guys think it would be nice to also use config mode to edit current configuration. Basically allow incremental json updates from api or web ui. This would be done by using the incremental json to update settings which already exists (from mqtt), so thats easy. 

Also thinking of changing the button to have two states to trigger a reset of config (current) or a restart into config mode to update current config. 
I was thinking of using https://github.com/mathertel/OneButton so we can use double click to put into update mode and long press like current to do full reset of config. This would basically be replacing https://github.com/thomasfredericks/Bounce2

Thoughts?

(Willing to implement by my self, just want to get ideas from you guys. Any additions?)

timpur, 2017-11-20:
@marvinroger @bertmelis @euphi @LiyouZhou @scharph @jesserockz

Gulaschcowboy, 2017-11-20:
Not being asked, but as a heavy homie-user: I vote +1

bertmelis, 2017-11-20:
I'm not waiting for it, but I not against neither.

But would it be still possible to run a webserver (on port 80) next to Homie?

timpur, 2017-11-20:
@Gulaschcowboy soz didn't mean to make it feel like i only wanted people who i tagged, not the case, just tagged them because they're contributes that i saw in recent commits. Anyone is always welcome to voice!!

@bertmelis not sure what you mean by this. Id be adding this feature to the homie web server and not subtracting anything from it? 

> But would it be still possible to run a webserver (on port 80) next to Homie?

bertmelis, 2017-11-21:
Maybe I don't get it completely.

Will the config-ui be available in config mode only or also in normal mode? As the latter will block any custom webserver that I may have implemented.

Or is the change only affecting config mode? (and the way you're getting into config mode)

marvinroger, 2017-11-21:
@timpur definitely, that's a great idea. Thank you so much for supporting this project, it's sad that I don't have enough time to work on it, but thanks to everyone involved in keeping this project alive. 😇

euphi, 2017-11-21:
OTA support would be nice, too. So you can update a firmware that crashes in NormalMode.

timpur, 2017-11-21:
@bertmelis change only to config mode to enhance the config web ui.

@euphi Use the config web ui to upload a new firmware? is what what you were thinking ?

luebbe, 2017-11-21:
How about removing config entries? Would this work (via incremental json) as well?

timpur, 2017-11-21:
@luebbe are you referring to #420, yeah id like to fix that issue for both mqtt and config mode.

timpur, 2017-11-29:
#432 
> Whats Next:
> - Add incremental config updates via config mode (Web UI and API) (#433).
> - Add ability to remove settings via incremental updates.
> - Fix incremental update trying to add settings when previously no settings object exists.
> - Better event buss handler for internal events and external. Hopefully this will make long async possesses that occur in homie more readable, like the prep for deep sleep procedure.
> - Better Task Manger for single shot async tasks + ongoing loop async tasks that occur throughout homie. Might also expose this for external use for the user to register tasks and loops. It will be something like the [TickerScheduler](https://github.com/Toshik/TickerScheduler) but custom to fit homie's needs.
> - More refactoring.
> - Possibly Upload Firmware via config mode (Web UI and API).

tripflex, 2018-01-20:
@timpur I do like this idea, mainly because of one issue that I came up with, imagine when the end user changes wifi creds, or router dies and they replace it but have different creds ... it would be nice to have some quick and easy way to update wifi credentials without having to do a full reset

sillyfrog, 2018-02-04:
Something that Sonoff-Tasmota does when it can't connect to WiFi is to go back into configuration mode for a few minutes before attempting to reconnect again. This is _very_ handy should your WiFi setup change, and you just want to change a password/SSID, without changing anything else. Tied with the ability to flip modes with the button (but not a full reset), and the upcoming features to keep the web server running, I think it would be a great addition!

ingoogni, 2018-02-04:
> go back into configuration mode for a few minutes before attempting to reconnect again
Several of my sensors have their own displays and work with or without wifi. I wouldn't like it at all when something, by default, interferes with this behaviour.

labodj, 2026-10-27:
I maintain a [standalone fork](https://github.com/labodj/homie-esp8266/) where [v4.1.0](https://github.com/labodj/homie-esp8266/releases/tag/v4.1.0) implements the distinction discussed here between reopening configuration and performing a factory reset.

With the default button timing, holding for 5 seconds and releasing before 15 seconds reopens setup while preserving the saved configuration and logical device ID.
Holding for at least 15 seconds and releasing performs a factory reset.
Homie.reset() also remains a factory reset.

The built-in setup page loads the existing configuration, preserves unchanged credentials and firmware settings, and tests the proposed Wi-Fi/MQTT connection before saving.
This allows a Wi-Fi credential change without starting configuration from scratch.

The implementation is available in the linked release for anyone still working on this workflow.