homieiot/homie-esp8266 #436: Experiment - Homie over RF

timpur, 2017-11-27:
Been thinking about how to get great range for IOT devices and that's the biggest problem with wifi is range.

So I was thinking if we could manage to get TCP over RF then we could use mqtt over RF module thus we could use Homie as a hybrid wifi for config more and what else you want and RF for normal mode and mqtt. 

It goal would be able to use the current Homie with no change to normal mode and build a lower level lib that the mqtt client can use to talk over RF or something like that. 

Wondering on your thoughts about this idea. Still researching into it, but seems there already exist TCP libs for RF modules so hopefully it would be just a matter of making a translation lib to convert these existing libs to a form that is the same as Async TCP lib.

Nzbuu, 2017-11-28:
Might be better to consider using MQTT-SN over an RF link since it doesn't require TCP

stritti, 2019-04-17:
Seems to be stale issue.
Please reopen if there are new progress or suggestions.

closed 2019-04-17 by stritti