homieiot/homie-esp8266 #441: No Access Point

jannnfe, 2017-11-30:
Hello,
I can flash my ESP8266 07 but no wifi AP will be create.
I'm using the beta 1 and Arduino setting up like in the doc.
I tested uploading a config.json over Arduino IDE but its not working too.
The light of the ESP lit red no matter GPIO0 ist connected or not
Thanks!

timpur, 2017-11-30:
Note with the ESP07 the led pin is on GPIO 2 not 1. Serial uses 1, thus if your arduino esp sdk defaults to builtin led to 1 this will mess up the serial connection, if you led is in use unless. You change this. 
One way could be to tell homie to use pin 2 for the led. 
Or personally i do this, since i only use esp07, change the line [static const uint8_t BUILTIN_LED = 1;](https://github.com/esp8266/Arduino/blob/eb494065b43de9bbf5850c061e46e149d3f41f12/variants/generic/pins_arduino.h#L37) to resemble using pin 2 not 1, and the line below this. This means arduino will use pin 2 not 1 for default builtin led.

So if your not getting any serial output from your ESP07 remember this, change your default builtin led definition.

jannnfe, 2017-12-01:
Where I have to change that file or the line? i can't find it on my Mac

euphi, 2017-12-01:
You can use 
`Homie.setLedPin(16, HIGH); // before Homie.setup() `

to change the LED to be used (see http://marvinroger.github.io/homie-esp8266/develop/advanced-usage/built-in-led/).

However, Homie sets the LED in setup(), so you should see the serial output before (bootloader etc.).

Bootloard baudrate is 74880. I recommend to set your application baudrate to 74880, too. That way you can see bootloader and application output, so use

```C++
Serial.begin(74880);
Serial.println("Application startup");
Serial.flush();
Homie.setLedPin(16, HIGH);  // or whatever LED pin you want to use.
```

before `Homie.setup();`

With the serial output you will see, why Homie does not open an access point. Maybe there is a valid config.json, but with wrong SSID or password?

jannnfe, 2017-12-01:
Maybe I do something wrong.
I Wire my ESP-07 like this:
VCC, CH_PD, GPIO2 -> VCC on FTDI
GND, GPIO15, GPIO0 -> GND on FTDI
RX -> TX on FTDI
TX -> RX on FTDI

VCC is 3,3V. I use [this](https://github.com/marvinroger/homie-esp8266/blob/develop/examples/DoorSensor/DoorSensor.ino) Sketch. I can compile it and upload it without any problems.
But the LED lit even when i change the tips above. No AP is generated. 

How is the right order to flash the ESP8266 ESP-07?
How can i check if the software is really uploaded?

euphi, 2017-12-01:
Have you checked the output on Serial during startup?

jannnfe, 2017-12-01:
Do you mean the Serial Monitor in Arduino IDE?
Yes its saying something like:
`<⸮`
and
`<⸮⸮c_⸮`

jannnfe, 2017-12-01:
Okay now i get this
`Application startup
"@⸮rjrA(!⸮S⸮u⸮P⸮⸮X⸮(⸮⸮P⸮⸮eh)⸮8⸮tm⸮*N⸮`

my void setup() looks like:
`void setup() {
  Serial.begin(74880);
  Serial.println("Application startup");
  Serial.flush();
  pinMode(PIN_DOOR, INPUT);
  digitalWrite(PIN_DOOR, HIGH);
  debouncer.attach(PIN_DOOR);
  debouncer.interval(50);

  Homie_setFirmware("awesome-door", "1.0.0");
  Homie.setLoopFunction(loopHandler);

  doorNode.advertise("open");

  Homie.disableLedFeedback();
  Homie.setup();
}`

jannnfe, 2017-12-01:
Update:
Now i get this: 

> ets Jan  8 2013,rst cause:4, boot mode:(1,6)
> wdt reset

It has something to do with that: http://arduino-esp8266.readthedocs.io/en/latest/boards.html#boot-messages-and-modes

euphi, 2017-12-01:
Can you upload your source and compete Serial ouput to https://gist.github.com/ ?

"wdt reset" happens if you application is "blocked" (e.g stuck in a loop)

amayii0, 2017-12-01:
(not in front of my code)
I noticed that for loops raising WDT it helps to use delay(0); within loop

euphi, 2017-12-01:
The ESP implementation of `delay(int)` calls `yield()`. 

`yield()` switches context back to the "Networking task", so the ESP is able to handle pending network (Wifi) requests and to reset the watchdog (WDT).

So, this helps for long running loops. However, I don't think that the problem here is caused by a intentionally long running loop.

timpur, 2017-12-01:
@jannnfe 
1. You said `"VCC, CH_PD, GPIO2 -> VCC on FTDI"` which means GPIO2(LED) is always connected to VCC which means the led will be stuck and not controlled by the mcu, since as i said ESP07 have the led connected to pin 2. Furthermore if the led in inverted, so pulling GPIO2 down to ground will activate the led, thus if GPIO2 is connected to VCC then when homie goes to flash the led it will be sinking VCC to GND which can't be good for the chip. So let GPIO2 be open (not connected) and add this line before homie setup for ESP07 `Homie.setLedPin(2, LOW);`
If led was the issue then you shouldn't get garbage after `"Application startup"`
2. Please send us your script and serial output if you still has issue through [https://gist.github.com/](https://gist.github.com/)

jannnfe, 2017-12-03:
Sorry for my late response.
I got a new situation. If I disconnect GPIO2 i am noch able to upload the Homie firmware. If I disconnect GPIO2 and GPIO15 I am able to upload the Firmware and while uploading there is a blue LED flashing and the red LED is lit red as usually.
But my Serial Monitor displays and no AP is created:

> Application startup
"@⸮rjrA(⸮⸮S⸮u⸮⸮v⸮X⸮(⸮⸮⸮y⸮eh⸮⸮"8⸮tmݪN⸮

timpur, 2017-12-04:
I think your just having an issue with general use of the ESP07, maybe read the doc (http://arduino-esp8266.readthedocs.io/en/latest/boards.html#minimal-hardware-setup-for-bootloading-and-usage). Try arduino examples (built in examples) for serial and check they work as expected. When you have it all working try homie again. Report back on any issues.

jannnfe, 2017-12-04:
I got the basic Arduino Blink running on ESP07:
> void setup() {
>   pinMode(2, OUTPUT);
> }
> 
> void loop() {
>   digitalWrite(2, HIGH);
>   delay(1000);
>   digitalWrite(2, LOW);
>   delay(1000);
> }

I connected VCC, CH_PD to VCC on FTDI
and GND, GPIO0 to GND on FTDI
and RX to TX and TX to RX

Arduino examples work but if i try the same setup on Homie Serial output will be:
> Application startup
"@⸮rjrA(!⸮SKu⸮КvX%(⸮⸮Py⸮eh)"8⸮tmݪN⸮

n8twj, 2017-12-04:
That looks like baud rate issues... try 74880 and put Serial.begin(74880)
as first line in setup()

closed 2018-01-25 by jannnfe