homieiot/homie-esp8266 #442: Error passing 'const IPAddress' as 'this' argument of 'String IPAddress::toString()'

stocarul, 2017-12-02:
I've been using the version `1.5` for a while and it works magically. You really did a great job! Today I decided to test the `2.0.0-beta2` version of this library and for some reason I get the following compilation error:

```
/Users/tutty/Documents/Arduino/libraries/homie-esp8266-2.0.0-beta.2/src/Homie/Boot/BootConfig.cpp:48:39: error: passing 'const IPAddress' as 'this' argument of 'String IPAddress::toString()' discards qualifiers [-fpermissive]
   _apIpStr = ACCESS_POINT_IP.toString();
                                       ^
```

Of course, if I change the line 48 from:
```
 _apIpStr = ACCESS_POINT_IP.toString();
```
into:
```
 _apIpStr = "192.168.123.1"; //ACCESS_POINT_IP.toString();
```
the compilation succeeds. Is this something expected or am I doing something wrong?

You can also check the full output bellow:

```
WARNING: Spurious .circleci folder in 'Homie' library
In file included from /Users/tutty/Documents/Arduino/libraries/homie-esp8266-2.0.0-beta.2/src/Homie/Datatypes/../Utils/Helpers.hpp:4:0,
                 from /Users/tutty/Documents/Arduino/libraries/homie-esp8266-2.0.0-beta.2/src/Homie/Datatypes/../Utils/Validation.hpp:7,
                 from /Users/tutty/Documents/Arduino/libraries/homie-esp8266-2.0.0-beta.2/src/Homie/Datatypes/../Config.hpp:10,
                 from /Users/tutty/Documents/Arduino/libraries/homie-esp8266-2.0.0-beta.2/src/Homie/Datatypes/Interface.hpp:7,
                 from /Users/tutty/Documents/Arduino/libraries/homie-esp8266-2.0.0-beta.2/src/Homie.hpp:6,
                 from /Users/tutty/Documents/Arduino/libraries/homie-esp8266-2.0.0-beta.2/src/Homie.h:4,
                 from /Users/tutty/Documents/Arduino/homie-test/homie-test.ino:1:
/Users/tutty/Documents/Arduino/libraries/homie-esp8266-2.0.0-beta.2/src/Homie/Datatypes/../Utils/../../StreamingOperator.hpp:9:15: warning: unused parameter 'arg' [-Wunused-parameter]
 inline Print &operator <<(Print &stream, _EndLineCode arg)
               ^
In file included from /Users/tutty/Documents/Arduino/libraries/homie-esp8266-2.0.0-beta.2/src/Homie/Datatypes/Interface.hpp:11:0,
                 from /Users/tutty/Documents/Arduino/libraries/homie-esp8266-2.0.0-beta.2/src/Homie.hpp:6,
                 from /Users/tutty/Documents/Arduino/libraries/homie-esp8266-2.0.0-beta.2/src/Homie.h:4,
                 from /Users/tutty/Documents/Arduino/homie-test/homie-test.ino:1:
/Users/tutty/Documents/Arduino/libraries/homie-esp8266-2.0.0-beta.2/src/Homie/Datatypes/../../HomieNode.hpp:27:107: warning: unused parameter 'range' [-Wunused-parameter]
   void settable(const PropertyInputHandler& inputHandler = [](const HomieRange& range, const String& value) { return false; });
                                                                                                           ^
/Users/tutty/Documents/Arduino/libraries/homie-esp8266-2.0.0-beta.2/src/Homie/Datatypes/../../HomieNode.hpp:27:107: warning: unused parameter 'value' [-Wunused-parameter]
/Users/tutty/Documents/Arduino/libraries/homie-esp8266-2.0.0-beta.2/src/Homie/Datatypes/../../HomieNode.hpp:27:107: warning: unused parameter 'range' [-Wunused-parameter]
/Users/tutty/Documents/Arduino/libraries/homie-esp8266-2.0.0-beta.2/src/Homie/Datatypes/../../HomieNode.hpp:27:107: warning: unused parameter 'value' [-Wunused-parameter]
In file included from /Users/tutty/Documents/Arduino/libraries/homie-esp8266-2.0.0-beta.2/src/Homie/Datatypes/Interface.hpp:11:0,
                 from /Users/tutty/Documents/Arduino/libraries/homie-esp8266-2.0.0-beta.2/src/Homie.hpp:6,
                 from /Users/tutty/Documents/Arduino/libraries/homie-esp8266-2.0.0-beta.2/src/Homie.h:4,
                 from /Users/tutty/Documents/Arduino/homie-test/homie-test.ino:1:
/Users/tutty/Documents/Arduino/libraries/homie-esp8266-2.0.0-beta.2/src/Homie/Datatypes/../../HomieNode.hpp:64:177: warning: unused parameter 'property' [-Wunused-parameter]
   HomieNode(const char* id, const char* type, const HomieInternals::NodeInputHandler& nodeInputHandler = [](const String& property, const HomieRange& range, const String& value) { return false; });
                                                                                                                                                                                 ^
/Users/tutty/Documents/Arduino/libraries/homie-esp8266-2.0.0-beta.2/src/Homie/Datatypes/../../HomieNode.hpp:64:177: warning: unused parameter 'range' [-Wunused-parameter]
/Users/tutty/Documents/Arduino/libraries/homie-esp8266-2.0.0-beta.2/src/Homie/Datatypes/../../HomieNode.hpp:64:177: warning: unused parameter 'value' [-Wunused-parameter]
/Users/tutty/Documents/Arduino/libraries/homie-esp8266-2.0.0-beta.2/src/Homie/Datatypes/../../HomieNode.hpp:64:177: warning: unused parameter 'property' [-Wunused-parameter]
/Users/tutty/Documents/Arduino/libraries/homie-esp8266-2.0.0-beta.2/src/Homie/Datatypes/../../HomieNode.hpp:64:177: warning: unused parameter 'range' [-Wunused-parameter]
/Users/tutty/Documents/Arduino/libraries/homie-esp8266-2.0.0-beta.2/src/Homie/Datatypes/../../HomieNode.hpp:64:177: warning: unused parameter 'value' [-Wunused-parameter]
In file included from /Users/tutty/Documents/Arduino/libraries/homie-esp8266-2.0.0-beta.2/src/Homie/Boot/BootNormal.hpp:8:0,
                 from /Users/tutty/Documents/Arduino/libraries/homie-esp8266-2.0.0-beta.2/src/Homie.hpp:12,
                 from /Users/tutty/Documents/Arduino/libraries/homie-esp8266-2.0.0-beta.2/src/Homie.h:4,
                 from /Users/tutty/Documents/Arduino/homie-test/homie-test.ino:1:
/Users/tutty/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266mDNS/ESP8266mDNS.h:66:8: warning: unused parameter 'ip' [-Wunused-parameter]
   bool begin(const char* hostName, IPAddress ip, uint32_t ttl=120){
        ^
/Users/tutty/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266mDNS/ESP8266mDNS.h:66:8: warning: unused parameter 'ttl' [-Wunused-parameter]
In file included from /Users/tutty/Documents/Arduino/libraries/ESPAsyncWebServer-master/src/ESPAsyncWebServer.h:448:0,
                 from /Users/tutty/Documents/Arduino/libraries/homie-esp8266-2.0.0-beta.2/src/Homie/Boot/BootConfig.hpp:9,
                 from /Users/tutty/Documents/Arduino/libraries/homie-esp8266-2.0.0-beta.2/src/Homie.hpp:13,
                 from /Users/tutty/Documents/Arduino/libraries/homie-esp8266-2.0.0-beta.2/src/Homie.h:4,
                 from /Users/tutty/Documents/Arduino/homie-test/homie-test.ino:1:
/Users/tutty/Documents/Arduino/libraries/ESPAsyncWebServer-master/src/AsyncWebSocket.h:80:10: warning: unused parameter 'i' [-Wunused-parameter]
     void operator ++(int i) { _count++; }
          ^
[... 276 lines cut ...]
/Users/tutty/Documents/Arduino/libraries/homie-esp8266-2.0.0-beta.2/src/Homie/Boot/../Datatypes/../../HomieNode.hpp:64:177: warning: unused parameter 'property' [-Wunused-parameter]
/Users/tutty/Documents/Arduino/libraries/homie-esp8266-2.0.0-beta.2/src/Homie/Boot/../Datatypes/../../HomieNode.hpp:64:177: warning: unused parameter 'range' [-Wunused-parameter]
/Users/tutty/Documents/Arduino/libraries/homie-esp8266-2.0.0-beta.2/src/Homie/Boot/../Datatypes/../../HomieNode.hpp:64:177: warning: unused parameter 'value' [-Wunused-parameter]
/Users/tutty/Documents/Arduino/libraries/homie-esp8266-2.0.0-beta.2/src/Homie/Boot/BootConfig.cpp: In member function 'virtual void HomieInternals::BootConfig::setup()':
/Users/tutty/Documents/Arduino/libraries/homie-esp8266-2.0.0-beta.2/src/Homie/Boot/BootConfig.cpp:48:39: error: passing 'const IPAddress' as 'this' argument of 'String IPAddress::toString()' discards qualifiers [-fpermissive]
   _apIpStr = ACCESS_POINT_IP.toString();
                                       ^
/Users/tutty/Documents/Arduino/libraries/homie-esp8266-2.0.0-beta.2/src/Homie/Boot/BootConfig.cpp: In member function 'void HomieInternals::BootConfig::_proxyHttpRequest(AsyncWebServerRequest*)':
/Users/tutty/Documents/Arduino/libraries/homie-esp8266-2.0.0-beta.2/src/Homie/Boot/BootConfig.cpp:245:40: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
   for (int i = 0; i < request->headers(); i++) {
                                        ^
/Users/tutty/Documents/Arduino/libraries/homie-esp8266-2.0.0-beta.2/src/Homie/Boot/BootConfig.cpp: In member function 'void HomieInternals::BootConfig::_onDeviceInfoRequest(AsyncWebServerRequest*)':
[... cut ...]
/Users/tutty/Documents/Arduino/libraries/homie-esp8266-2.0.0-beta.2/src/Homie/Boot/BootConfig.cpp:429:6: warning: unused parameter 'len' [-Wunused-parameter]
 void BootConfig::_parsePost(AsyncWebServerRequest *request, uint8_t *data, size_t len, size_t index, size_t total) {
      ^
/Users/tutty/Documents/Arduino/libraries/homie-esp8266-2.0.0-beta.2/src/Homie/Boot/BootConfig.cpp:443:21: warning: 'const String ConfigJSONError(String)' defined but not used [-Wunused-function]
 static const String ConfigJSONError(const String error) {
                     ^
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).
```

timpur, 2017-12-03:
Dup of #429

If you use latest dev this will fix that issue.

marvinroger, 2017-12-05:
Fixed in https://github.com/marvinroger/homie-esp8266/releases/tag/v2.0.0-beta.3
Thanks @timpur !

closed 2017-12-05 by marvinroger