homieiot/homie-esp8266 #444: Out of date AsyncTCP dependency?

kylegordon, 2017-12-06:
I've run into a compilation issue with #develop branch, as seen at https://travis-ci.org/kylegordon/homie-sonoff-touch/builds/312535275#L727 where it's failing with ```fatal error: freertos/semphr.h: No such file or directory```

It's apparently a solved problem, as per https://github.com/me-no-dev/AsyncTCP/issues/2 so I was wondering if Homie has an out of date dependency?

marvinroger, 2017-12-06:
> It's apparently a solved problem, as per me-no-dev/AsyncTCP#2

FreeRTOS? AsyncTCP is an ESP32 library, ESPAsyncTCP is for ESP8266, you
need this one.

euphi, 2017-12-06:
travis-ci can reproduce, see https://travis-ci.org/euphi/Homie_BareMinimum/builds/312604604 .

For this test, I created a very simple repository https://github.com/euphi/Homie_BareMinimum .
However, at my own computer, I got another error message, see #429

euphi, 2017-12-06:
One strange thing is that I have a different dependency tree at home than travis-ci:

```
Verbose mode can be enabled via `-v, --verbose` option
Collected 31 compatible libraries
Looking for dependencies...
Library Dependency Graph
|-- <Hash> v1.0
|-- <Homie> v2.0.0
|   |-- <ArduinoJson> v5.11.2
|   |-- <AsyncMqttClient> v0.8.1
|   |   |-- <ESPAsyncTCP> v1.1.3
|   |   |   |-- <ESP8266WiFi> v1.0
|   |-- <Bounce2> v2.3
|   |-- <Ticker> v1.0
|   |-- <DNSServer> v1.1.0
|   |   |-- <ESP8266WiFi> v1.0
|   |-- <Homie> v2.0.0
|   |   |-- <ArduinoJson> v5.11.2
|   |   |-- <AsyncMqttClient> v0.8.1
|   |   |   |-- <ESPAsyncTCP> v1.1.3
|   |   |   |   |-- <ESP8266WiFi> v1.0
|   |   |-- <Bounce2> v2.3
|   |   |-- <ESP Async WebServer> v1.1.1
|   |   |   |-- <ESPAsyncTCP> v1.1.3
|   |   |   |   |-- <ESP8266WiFi> v1.0
|   |   |   |-- <ESP8266WiFi> v1.0
|   |   |   |-- <ArduinoJson> v5.11.2
|   |   |   |-- <Hash> v1.0
|   |   |-- <ESPAsyncTCP> v1.1.3
|   |   |   |-- <ESP8266WiFi> v1.0
|   |   |-- <DNSServer> v1.1.0
|   |   |   |-- <ESP8266WiFi> v1.0
|   |   |-- <ESP8266HTTPClient> v1.1
|   |   |   |-- <ESP8266WiFi> v1.0
|   |   |-- <Ticker> v1.0
|   |   |-- <ESP8266WiFi> v1.0
|   |   |-- <ESP8266mDNS>
|   |   |   |-- <ESP8266WiFi> v1.0
|   |-- <ESP8266httpUpdate> v1.1
|   |   |-- <ESP8266WiFi> v1.0
|   |   |-- <ESP8266HTTPClient> v1.1
|   |   |   |-- <ESP8266WiFi> v1.0
|   |-- <ESP8266WiFi> v1.0
|   |-- <ESP8266mDNS>
|   |   |-- <ESP8266WiFi> v1.0
|   |-- <ESP8266WebServer> v1.0
|   |   |-- <ESP8266WiFi> v1.0

```
Note that <Homie> v2.0.0 is listed two times - with *different* dependencies to lower-level libraries?!
Maybe @ivankravets  has a hint?

(version of platformio is "PlatformIO, version 3.4.1" in both cases).

I saw that the examples from Homie run fine on travis-ci. And also one of my larger projects compiles fine at home with Homie v2.0.0-beta.3 (I haven't tried the others yet).

euphi, 2017-12-06:
Oh, the "larger project" is also on travis-ci - and compiles fine after upgrading Homie to latest development (==v2.0.0-beta.3): https://travis-ci.org/euphi/ESP_Homie_WS2812FX

kylegordon, 2017-12-06:
@marvinroger I agree, it shouldn't be requiring ESP32 libraries. I'm not declaring anything other than Homie though.

If you have a look at line 578, https://travis-ci.org/kylegordon/homie-sonoff-touch/builds/312535275#L578 it seems to be bringing it in via <ESP Async WebServer> v1.1.1

Is this a problem that should be referred to https://github.com/me-no-dev/ESPAsyncWebServer (again) ?

euphi, 2017-12-07:
Yeah, setting the ldf-mode of platformio to chain+ (instead of the default chain) fixes the problem. (see https://github.com/platformio/platformio-core/issues/1153)

Commit:
https://github.com/euphi/Homie_BareMinimum/commit/236bac778a715d94f360dada04b3512acc81a7c0
fixes the build (https://travis-ci.org/euphi/Homie_BareMinimum/builds/313215893).

timpur, 2017-12-16:
So whats needed to be done here ?

ThomasDevoogdt, 2017-12-28:
Hello, I'm facing the same issue, but in can't remove the wrong lib. 

First I do: 

```shell
thomas@Thomas-Aspire-V5-573G:~/MEGA/Coding/plant_care/plant_care_esp8266$ pio lib uninstall AsyncTCP
Library Storage: /home/thomas/MEGA/Coding/plant_care/plant_care_esp8266/.piolibdeps
Uninstalling AsyncTCP @ 1.0.1:  [OK]
thomas@Thomas-Aspire-V5-573G:~/MEGA/Coding/plant_care/plant_care_esp8266$ pio lib uninstall AsyncTCP
Library Storage: /home/thomas/MEGA/Coding/plant_care/plant_care_esp8266/.piolibdeps
Error: Detected unknown package 'AsyncTCP @ *'
```

But then I get: (It just reinstalls the libraries)

```
/home/thomas/.local/share/JetBrains/Toolbox/apps/CLion/ch-1/173.4127.6/bin/cmake/bin/cmake --build /home/thomas/MEGA/Coding/plant_care/plant_care_esp8266/cmake-build-debug --target PLATFORMIO_BUILD -- -j 2
[Thu Dec 28 13:46:02 2017] Processing d1_mini (platform: espressif8266; lib_deps: Hash, Adafruit Unified Sensor, https://github.com/adafruit/Adafruit_BMP085_Unified.git, https://github.com/adafruit/Adafruit_Sensor.git, https://github.com/claws/BH1750.git, https://github.com/adafruit/DHT-sensor-library.git, https://github.com/me-no-dev/ESPAsyncWebServer.git, https://github.com/arduino-libraries/NTPClient.git, https://github.com/knolleary/pubsubclient.git, https://github.com/ThomasDevoogdt/SensorFacade.git; upload_speed: 115200; board: d1_mini; framework: arduino)
--------------------------------------------------------------------------------
LibraryManager: Installing id=305
Unpacking...
LibraryManager: Installing id=1826
Unpacking...
Verbose mode can be enabled via `-v, --verbose` option
Collected 38 compatible libraries
Looking for dependencies...
Library Dependency Graph
|-- <Hash> v1.0
|-- <Adafruit Unified Sensor> v1.0.2
|-- <Wire> v1.0
|-- <Ticker> v1.0
|-- <DNSServer> v1.1.0
|   |-- <ESP8266WiFi> v1.0
|-- <DHT sensor library> v1.3.0
|   |-- <Adafruit Unified Sensor> v1.0.2
|-- <ESP Async WebServer> v1.1.1
|   |-- <ESPAsyncTCP> v1.1.3
|   |   |-- <ESP8266WiFi> v1.0
|   |-- <ESP8266WiFi> v1.0
|   |-- <AsyncTCP> v1.0.1
|   |-- <Hash> v1.0
|-- <PubSubClient> v2.6
|-- <Adafruit BMP085 Unified> v1.0.0
|   |-- <Wire> v1.0
|   |-- <Adafruit Unified Sensor> v1.0.2
|-- <BH1750>
|   |-- <Wire> v1.0
|-- <ESP8266WiFi> v1.0
|-- <SensorFacade> v1.0
|-- <NTPClient> v3.1.0
Archiving .pioenvs/d1_mini/lib/libESPAsyncTCP.a
Compiling .pioenvs/d1_mini/lib/AsyncTCP_ID1826/AsyncTCP.o
Compiling .pioenvs/d1_mini/lib/ESP Async WebServer/AsyncWebSocket.o
Compiling .pioenvs/d1_mini/lib/ESP Async WebServer/SPIFFSEditor.o
Indexing .pioenvs/d1_mini/lib/libESPAsyncTCP.a
In file included from .piolibdeps/AsyncTCP_ID1826/src/AsyncTCP.cpp:24:0:
.piolibdeps/AsyncTCP_ID1826/src/AsyncTCP.h:28:29: fatal error: freertos/semphr.h: No such file or directory
#include "freertos/semphr.h"
^
compilation terminated.
*** [.pioenvs/d1_mini/lib/AsyncTCP_ID1826/AsyncTCP.o] Error 1
========================== [ERROR] Took 2.40 seconds ==========================
CMakeFiles/PLATFORMIO_BUILD.dir/build.make:57: recept voor doel 'CMakeFiles/PLATFORMIO_BUILD' is mislukt
make[3]: *** [CMakeFiles/PLATFORMIO_BUILD] Fout 1
CMakeFiles/Makefile2:67: recept voor doel 'CMakeFiles/PLATFORMIO_BUILD.dir/all' is mislukt
make[2]: *** [CMakeFiles/PLATFORMIO_BUILD.dir/all] Fout 2
CMakeFiles/Makefile2:79: recept voor doel 'CMakeFiles/PLATFORMIO_BUILD.dir/rule' is mislukt
make[1]: *** [CMakeFiles/PLATFORMIO_BUILD.dir/rule] Fout 2
Makefile:118: recept voor doel 'PLATFORMIO_BUILD' is mislukt
make: *** [PLATFORMIO_BUILD] Fout 2 
```

ivankravets, 2017-12-28:
@ThomasDevoogdt do you use the latest PIO Core 3.5.0?

```
pio --version
pio upgrade
```

ThomasDevoogdt, 2017-12-29:
I have the newest version of PIO. First I deleted the _.pioenvs_ completely, and all the wrong libs in _.piolibdeps_. Next I tried again and got new errors. This [post](https://github.com/me-no-dev/ESPAsyncTCP/issues/62) helped me: [c2d4c16](https://github.com/me-no-dev/ESPAsyncTCP/commit/c2d4c168d41b78d9abd71a074c80eb8087c33561). I moved ESPAsyncTCP and ESP Async WebServer to the lib folder, so it won't update any more.

kylegordon, 2017-12-29:
I tried to compile https://github.com/euphi/Homie_BareMinimum again, with a fresh checkout, and it seems to fail. Using PlatformIO 3.5.0 here.

```
git clone https://github.com/euphi/Homie_BareMinimum
cd Homie_BareMinimum
pio run
```

However, if you run ```pio lib update``` prior to ```pio run```, it seems to work.

I don't know why pio run brings in ESPAsyncTCP 1.0.1, but the update seems to get 1.1.3 instead.

After 1.1.3 is installed, it will compile as expected.

ivankravets, 2017-12-29:
> I don't know why pio run brings in ESPAsyncTCP 1.0.1, but the update seems to get 1.1.3 instead.

Because

* https://github.com/marvinroger/homie-esp8266/blob/develop/library.json#L29
* https://github.com/marvinroger/async-mqtt-client/blob/master/library.json#L21

Please contact @marvinroger and ask them to resolve all issues with dependencies.

timpur, 2017-12-29:
yep i think the issue is that `async-mqtt-client` needs updating which should fix homie's sub dependencies

euphi, 2018-01-04:
@ivankravets I created a feature request that may tracking down such problems: https://github.com/platformio/platformio-core/issues/1274

timpur, 2018-01-09:
This should be fixed now ?

kylegordon, 2018-01-09:
Yes, it seems to be working as expected now. A clean checkout of some Homie based projects results in a clean build from 'pio run'.

Many thanks for everyone's help here!

closed 2018-01-09 by kylegordon