homieiot/homie-esp8266 #446: Configuration v2 fails due to CORS

jalmeroth, 2017-12-13:
I am trying to configure the latest dev-version of Homie 2 (f605adf).

When using the latest Configurator page v2 (03034c6) I am receiving an error in Chrome 63.0.3239.84 (Official Build) (64-bit) and other browsers:
`Failed to load http://192.168.123.1/heart: No 'Access-Control-Allow-Origin' header is present on the requested resource. Origin 'http://localhost:8000' is therefore not allowed access.`

I could temporarily fix this issue by adding the following default headers after this line https://github.com/marvinroger/homie-esp8266/blob/develop/src/Homie/Boot/BootConfig.cpp#L18:
```
DefaultHeaders::Instance().addHeader(F("Access-Control-Allow-Origin"), F("*"));
DefaultHeaders::Instance().addHeader(F("Access-Control-Allow-Methods"), F("PUT, GET"));
DefaultHeaders::Instance().addHeader(F("Access-Control-Allow-Headers"), F("Content-Type, Origin, Referer, User-Agent"));
`
``
_Please note: This is no the fix, it's just a proof to make it work temporarily._

Some of the following requests are also failing then as calls to __sendCORS() will send these headers again.

timpur, 2017-12-14:
Ill look into a fix for CORS, but adding those lines wont help you here, because your origin is from 8000 (Self hosted webserver?) not the esp. Thus you need 8000 to allow CORS not the esp, because you cant even get to the esp in the first place? (in my understanding,  might be wrong?)

Ill look into a solution for this. It would be nice to use the online config and not save to rely on the build in esp config, even though having it on the esp has other advantages, but thats up to you to decide and thats why ill try fix this.

jalmeroth, 2017-12-14:
> Ill look into a fix for CORS, but adding those lines wont help you here, because your origin is from 8000 (Self hosted webserver?) not the esp. Thus you need 8000 to allow CORS not the esp, because you cant even get to the esp in the first place? (in my understanding, might be wrong?)

That's wrong, `Access-Control-Allow-Origin: *` allows access from everywhere, including my local webserver (not the esp itself).

> Ill look into a solution for this. It would be nice to use the online config and not save to rely on the build in esp config, even though having it on the esp has other advantages, but thats up to you to decide and thats why ill try fix this.

Yeah, the online configurator is nice, but please keep the compatibility to checkout the gh-pages branch on your local maschine to configure homie devices, when you are offline or connected to the Homie Device AP. ;)

marvinroger, 2017-12-14:
Right, there’s a CORS issue. What I don’t get, though, is why it used to work.

The code was only handling preflight CORS request (OPTIONS), so the config endpoint was the only one that was supposed to work... anyway, the fix is pretty simple: it’s just a matter of wrapping each request into something that sends the CORS header, and it’lol work everywhere.

timpur, 2017-12-14:
> That's wrong, Access-Control-Allow-Origin: * allows access from everywhere, including my local webserver (not the esp itself).

but you want the webserver to tell the client (chrome) that its okay to go to the esp for requests ??? It wont work is you load the website from one server and then try access another webserver (ESP) with out the origin server giving CORS access??

jalmeroth, 2017-12-14:
@timpur well… as long as the AsyncWebserver, running on the ESP, is not sending these Access-Control-Allow-Headers, modern clients (e.g. Chrome) will not let you execute the JS XHR, which is in this case hosted on my local webserver, but could be anywhere else like the online configurator, and therefore the queries to the URLs necessary to configure your device will fail.

timpur, 2017-12-14:
Seems my understanding of CORS was totally wrong.

timpur, 2017-12-14:
But if you want @marvinroger i can fix this issue ?

marvinroger, 2017-12-14:
Yes @timpur, please, go ahead. 😉

timpur, 2017-12-14:
@marvinroger assign this to me ? (how do you feel about making me a Collaborator ?)

benzino77, 2017-12-14:
I think this is a great idea. @marvinroger you don't have so much time for the projects as before and @timpur is one of the most active contributor which keep this project alive.

marvinroger, 2017-12-14:
Well, believe me or not, I was about to. ☺️
One condition, though: add your name on the manifests. 😌

tripflex, 2017-12-21:
@timpur thank you for all your hard work on this, it's much appreciated!  Hope to be able to start contributing myself very soon

tripflex, 2018-01-11:
@jalmeroth thank you for this! I was having this issue while trying to call API endpoints using a Cordova mobile app, which does not work without the device handling CORS.

Yes I did look at potentially using a proxy to do this, but really CORS should be handled via the server, in the case being the device.

Being as though this is for the API endpoints, I don't see any reason to not be handling this on the device, @timpur can you please make sure to include this in the 2.1.0 release?  I see it was added to the milestone, and think this is something we should def have in the codebase

tripflex, 2018-01-11:
Looking at the code itself, it appears that CORS handling already exists for `/config` and `/wifi/connect` ... so probably just need to set it for the other endpoints as well. 

Adding those default headers makes it possible to get past CORS restrictions for `GET` requests without having to respond to `OPTIONS`

> NOTE: You will still need to respond to the OPTIONS method for CORS pre-flight in most cases. (unless you are only using GET)

^ from [ESPAsyncWebServer](https://github.com/me-no-dev/ESPAsyncWebServer)
[Mozzila CORS Docs](https://developer.mozilla.org/en-US/docs/Web/HTTP/CORS#Functional_overview)

timpur, 2018-01-12:
@tripflex its fixed, did it last night (for me) all works, just haven't pushed to my repo, will do tonight.

All request have been opened to CORS, not really worried about security issues with that. Unless ive missed something?

tripflex, 2018-01-19:
@timpur yup I tested myself using your 2.1 branch fork and no CORS issues at all! Thanks for fixing that so quick!

And no this is not an issue security wise, as CORS is really useful for preventing malicious scripts, browser extensions/plugins, or even other JS code from making requests to a server it should not be.  Being as though this is for iot device, this should be open as the requests are almost always going to be coming from somewhere other than the device, not from the device itself.

boneskull, 2018-01-20:
ha, I was just running into this.  thanks @timpur

commit 95d95669 (2018-03-18): Fix #446 #477 (#501)

* Fix #446 CORS Issue

* Fix for #477

* Fix Lint
closed 2018-03-18 by timpur