homieiot/homie-esp8266 #447: Configuration v2 fails due to Mixed Content

jalmeroth, 2017-12-13:
I am trying to configure the latest dev-version of Homie 2 (f605adf).

When using the [Configurator v2](https://marvinroger.github.io/homie-esp8266/configurators/v2/) (probably 03034c6) I am receiving an error in Chrome 63.0.3239.84 (Official Build) (64-bit) and other browsers:

```
Mixed Content: The page at 'https://marvinroger.github.io/homie-esp8266/configurators/v2/' was loaded over HTTPS, but requested an insecure XMLHttpRequest endpoint 'http://192.168.123.1/heart'. This request has been blocked; the content must be served over HTTPS.
```

timpur, 2017-12-14:
try load it over http ? http://marvinroger.github.io/homie-esp8266/configurators/v2/

jalmeroth, 2017-12-14:
Well, this is not a Chrome issue, it's common, modern browser security. Read more about this here: <https://developer.mozilla.org/en-US/docs/Web/Security/Mixed_content#Mixed_active_content>.
The HTTPS site is linked here for example: <https://marvinroger.github.io/homie-esp8266/>.
I will test the non-HTTPS site later, but it will presumably lead to #446. :)

marvinroger, 2017-12-14:
Well that’s right, the configurator MUST be loaded over HTTP. I thought it was written in the docs, I might have forget that.

stritti, 2020-01-03:
No progress for long time. Closing issue.

closed 2020-01-03 by stritti

danielwegener, 2020-01-17:
Well, a user-friendly fix would be to let the configurator page to reload itself via http if it detects that it is loaded via https. E.g. via `document.location.protocol`. Or to simply display a warning with a generated link to the current location via http (many hosting locations may decide to enforce https via HSTS).

stritti, 2020-01-18:
@danielwegener PR is welcome 😀