homieiot/homie-esp8266 #449: WIFI_CONNECTED does not get fired after a WiFi reconnect

T94T, 2017-12-17:
Hi,
I use the latest version of homie-esp8266. When I turn on the device HomieEventType::WIFI_CONNECTED event gets fired. After turning off and on my WiFi router, this event does not get fired anymore. Is this a bug, or do I something wrong?

timpur, 2017-12-17:
Hard to say 
How long do you leave your router off for ?

T94T, 2017-12-18:
Just wanted to test the events. So I turned off the router and switched it back on immediately.

T94T, 2017-12-18:
btw: The device reconnects to the WiFi AP but the event does not get fired.

timpur, 2017-12-18:
Might not to do with homie more the the esp Arduino sdk

T94T, 2017-12-18:
Thank you for this hint. Just made a git pull on esp8266/Arduino.

Even
`WiFi.isConnected()`
returns false after the device is reconnected. Very strange...

T94T, 2017-12-18:
@marvinroger: Could this be the problem? https://github.com/esp8266/Arduino/issues/2735#issuecomment-264711139 How is your code implemented?

T94T, 2018-01-16:
Problem seems to be solved with the latest master branch of esp8266/Arduino. Please have a look at this: https://github.com/esp8266/Arduino/issues/4161

closed 2018-01-16 by T94T

timpur, 2018-01-16:
Cool :)