homieiot/homie-esp8266 #450: Question regarding IteadSonoffButton.ino and offline usage

abbe79, 2017-12-17:
Hi!
Thanks for this great project, looks like I won't have to code everything I need from scratch. :-)

I plan to use an itead Sonoff4ch to control my ventilation system. I looked into your example `IteadSonoffButton.ino` but have one question:
You are registering the function that reads the button status like this
`Homie.setLoopFunction(loopHandler);`
If I read the docs correctly this means that this is called only while in normal state, not in configuration.
It would be fantastic if controlling by button would also work while the connection to WiFi/MQTT is lost.

Since the function is written in a clever way (saving the millis of button press changes without an `delay()`), would it be possible to move the call to the `loop()`-function?

timpur, 2017-12-17:
Good point I'll look into that

Also might be better to use debounce2 which is used by homie for the reset button anyways so the lib is included in the firmware so may as well reuse it. 

Might fix up that example and others in my next push

tripflex, 2017-12-26:
@abbe79 here's what i'm using, found this in one of the older issues, and since `Bounce` is already included in homie, no need to include extra libs:

```c++
Bounce cycleBounce = Bounce();

void setup(){
    cycleBounce.attach(PIN_CYCLE);
    cycleBounce.interval(200);
}

void loop(){
    if ( cycleBounce.fell() ) {
        Serial.println("PIN Toggle Relay");
    }

    cycleBounce.update();

   homie.loop();
}
```

Here's the Wiki for Bounce2:
https://github.com/thomasfredericks/Bounce2/wiki

tripflex, 2018-01-19:
@abbe79 one thing to mention, check here:
https://github.com/marvinroger/homie-esp8266/pull/458#issuecomment-357233820

Homie may be moving to onebutton, so just something to keep in mind

abbe79, 2018-01-30:
Thanks for the notice, not a problem for me: Move and I'll follow! :-D