homieiot/homie-esp8266 #451: Set default device setting without using MQTT?

tripflex, 2017-12-19:
Is there a way to set default device property, specifically `$stats/interval` in setup code? can’t find anything online that details it and would like to change it to 5 seconds by default for all devices instead of just 1 second

so I found `const uint32_t STATS_SEND_INTERVAL = 1 * 60 * 1000;` in `Constants.hpp` but still can’t seem to figure out the correct way to set this without modifying the lib code

timpur, 2017-12-19:
Nope (that I know of) will add in my next release.

tripflex, 2017-12-19:
Awesome thank you, just the ability to configure or override any default settings would be perfect 👍  thanks!!

tripflex, 2017-12-19:
@timpur now that i'm looking at this ... is it even possible to set interval via MQTT?  I can't seem to figure out any way to change the interval value used ... which maybe I don't understand the purpose of it, but my understanding is that interval would be something settable so you can change it as needed ... but now it looks like it's just a static value?

timpur, 2017-12-20:
The interval is used to send info like signal and up time

It would be nice to be able to configure this. Thinking of doing this through the config, thus you can also update over mqtt via incremental json

luebbe, 2017-12-20:
But isn't it also important for the MQTT server to receive this kind of "ping" message in a not "too long" interval? Kind of keepalive. Otherwise he will consider the device offline?!?
I don't know enough about MQTT (whether and how it is possible to connect *and* tell the server at the same time how often you will ping it). I only remember having problems with one (non C) MQTT library where I couldn't set a keepalive interval and had to make sure to ping the server once every minute.

timpur, 2017-12-20:
The keep alive period is handled by the mqtt lib and has nothing to do with homie. By protocol (my understanding) when the client connects it tells the broker the keep alive period and if that period is broken then the broker considers the connection broken....

Nothing to do with this stats interval with homie.

tripflex, 2017-12-20:
@timpur yeah i'll probably start working on this and submit a PR if you haven't already ... did you have any specific ideas in mind of how to set it up?  I can probably work on the code if you wanted it done in a specific way, but I will need this for my current project.

Reason being ... use case scenario would be outputting data on a dashboard ... so when the user is viewing dashboard for a device, we would want to update the interval to say, every 15 seconds to give a better UI experience, whereas, when just storing stats, we would only want it set to every 300 seconds (5 minutes) instead

timpur, 2017-12-21:
Okay well that's not a big change let me do this today. I'll get onto it now

timpur, 2017-12-21:
Closed by #455

You can now add the interval in seconds in the config file, which means you can also use incremental json to change this via mqtt if you like. Node not yet set this in the web UI configurator

closed 2017-12-21 by timpur

tripflex, 2017-12-21:
Awesome, thank you!!  Yeah I doubt it would need to be set in the web UI, but if someone really needed it they could just setup a "custom" setting to allow for that, thanks again, will test and let you know if i find any issues!

tripflex, 2017-12-21:
Any possibility of adding a function similar to Homie_setFirmware for the interval, say, Homie_setStatsInterval to allows configuring initial interval in the setup function?

I did see you added the ability in the config file, just wasn't sure if Homie would read incremental config.json from spiff or not ... as I do still want to use the Web UI setup (interval not necessary to set in web UI)

Also posted this in #454

tripflex, 2017-12-22:
@timpur yeah understandable, but I meant that I would like to be able to set the default value configured for the device even before it's setup and ready to go. 

Mainly to set default to around 5 minutes, and then I can adjust via MQTT once it's connected to the broker via incremental json.  I guess for now I can just immediately send an incremental json to the device once it connects to the broker and change it from 1 minute to 5 minutes ... which just be much cleaner to be able to set the default value in the setup function, just like we would with a normal HomieSetting

timpur, 2017-12-22:
i think you miss understand, you can upload the config file from your computer with the 5min value with out the need of mqtt. See http://marvinroger.github.io/homie-esp8266/docs/develop/configuration/json-configuration-file/ .You also dont need to use configurator if you wait to just upload the config file via your computer. [Uploading files to file system](http://esp8266.github.io/Arduino/versions/2.3.0/doc/filesystem.html#uploading-files-to-file-system)

also see https://github.com/marvinroger/homie-esp8266/tree/develop/data/homie