homieiot/homie-esp8266 #453: Feature - Battery Management

timpur, 2017-12-20:
Wondering if you guys think if battery management would be usefully / belongs as part of homie ?

Idea would be that you can enable battery management in homie and configure it. 

It would send status updates like the uptime and signal as part of the interval 

It could also be used to change a sleep time for deep sleep if needed. Useful to extend battery life by reducing the interval its on for.

I already have this module, as I use it in my solar battery devices.

Also any other ideaa for this ?

Gulaschcowboy, 2017-12-20:
+1 from me, but no further ideas so far. At the moment I made my own outside of homie, which just does a deepSleep forever if it detects low batteries. So any improvement welcome.

euphi, 2017-12-20:
I would see this as feature for a version 3.1.

- So, first, there should be stable release with current features. (Version 2.0 or 2.1)
- Then there should be a release 3.0 based on the homie convention 3.0 (or however this release will be called).

Even then, I would prefer to allow Homie-Modules that extend core functionality. The battery management then would be a very good example for such a module.

A modularization could be achieved by allowing the addition of callback functions for internal tasks like sending device attributes.

Another argument for "modularization": My most important current project (a touch controlled thermostat) already is very close to the 480kb "size limit" (ESP-12F with 1MB flash. 64kb for SPIFFS, so there is only 480kb left if you want to update via OTA).

timpur, 2017-12-20:
Yeah I thought it might not be part of core, I do like the idea of add-on modules that can hook onto core. Would be good for other bits later also.

closed 2019-07-15 by stritti