homieiot/homie-esp8266 #454: Stable Release 2.0

timpur, 2017-12-20:
How do we feel about current Dev being released as stable, in the near future?

Anyone have bugs but havent reported it ?

tripflex, 2017-12-21:
I've been heavily testing it myself for the past week or so and haven't found any major issues, but i should know within the next few days once i'm able to release a beta, how things are going.

Any possibility of adding a function similar to `Homie_setFirmware` for the interval, say, `Homie_setStatsInterval` to allows configuring initial interval in the setup function?

I did see you added the ability in the config file, just wasn't sure if Homie would read incremental `config.json` from `spiff` or not ... as I do still want to use the Web UI setup (interval not necessary to set in web UI)

timpur, 2018-03-12:
Who wants to help write up some release notes?

euphi, 2018-03-12:
The documentation is in a very good state so there is no need for long release notes.

So I would just write that the beta has been proven as stable and there is more exciting stuff to come with releases 2.1 and 3.0.

The most important new features have already been summarized in the release notes for [beta.1](https://github.com/marvinroger/homie-esp8266/releases/tag/v2.0.0-beta.1), so they can be used as base for the release notes.

What do you think about some small "press release", so Homie-ESP8266 may get some advertisement? Such a release could be send to some maker magezine / technical news portals!?

bodiroga, 2018-03-12:
Hi @timpur!

Great to see that you are going to release Homie-ESP8266 2.0 as a stable version!

One question though, is this (https://github.com/marvinroger/homie-esp8266/issues/485) problem fixed in the 2.0 branch? I know that you have fixed it in your 2.1 branch, but is there any option to backport it to the 2.0 branch?

For new and inexperienced users using the web ui to configure the device is a much more pleasant experiencience and we should offer it as an option. Every documentation page talks about this method and was the prefered one, so please, keep it in the 2.0 stable version :pray: 

Thanks you so much for your work!

timpur, 2018-03-12:
I think you have a good point there actually, I think I will merge the bug fixes (that make sense) back to 2.0

timpur, 2018-03-12:
I shall do these things over the week and, we should be ready this weekend.

bodiroga, 2018-03-13:
Awesome! Thanks for taking my comment into account :+1: 

I will help debugging before the final release, keep up with the good job!

timpur, 2018-03-18:
merged some issues #501 back to 2.0 from 2.1

@bodiroga can you check the CORS issue #446 
@Gulaschcowboy can you test #477 

Thanks guys.

bodiroga, 2018-03-19:
Hi @timpur!

I have deleted the previously installed homie-esp-2.1 library, downloaded the "develop" branch as a .zip file, imported it to my Arduino IDE, reflashed the NodeMCU and... it works! The configuration page make its job, so I can confirm that the CORS issue is resolved :+1: 

Thank you so much for the effort!

timpur, 2018-03-19:
Thank you for testing :)

bodiroga, 2018-03-19:
Good morning again @timpur!

This is not very important, but I'm not sure if you could clean the different warning messages that appear in the compilation stage before the final version. It is just an aesthetic thing, nothing serious :wink: 

Here you have what I get when compiling a Homie sketch:

> /home/aitor/Arduino/libraries/homie-esp8266-develop/src/Homie.cpp: In constructor 'HomieInternals::HomieClass::HomieClass()':
/home/aitor/Arduino/libraries/homie-esp8266-develop/src/Homie.cpp:13:30: warning: 'BUILTIN_LED' is deprecated (declared at /home/aitor/.arduino15/packages/esp8266/hardware/esp8266/2.4.1/variants/nodemcu/../generic/common.h:78) [-Wdeprecated-declarations]
   Interface::get().led.pin = BUILTIN_LED;
                              ^
/home/aitor/Arduino/libraries/homie-esp8266-develop/src/Homie/Logger.cpp: In member function 'virtual size_t HomieInternals::Logger::write(uint8_t)':
/home/aitor/Arduino/libraries/homie-esp8266-develop/src/Homie/Logger.cpp:20:1: warning: no return statement in function returning non-void [-Wreturn-type]
 }
 ^
/home/aitor/Arduino/libraries/homie-esp8266-develop/src/Homie/Logger.cpp: In member function 'virtual size_t HomieInternals::Logger::write(const uint8_t*, size_t)':
/home/aitor/Arduino/libraries/homie-esp8266-develop/src/Homie/Logger.cpp:24:1: warning: no return statement in function returning non-void [-Wreturn-type]
 }
 ^
/home/aitor/Arduino/libraries/homie-esp8266-develop/src/Homie/Timer.cpp: In member function 'uint32_t HomieInternals::Timer::getInterval()':
/home/aitor/Arduino/libraries/homie-esp8266-develop/src/Homie/Timer.cpp:20:12: warning: statement has no effect [-Wunused-value]
   _interval;
            ^
/home/aitor/Arduino/libraries/homie-esp8266-develop/src/Homie/Timer.cpp:21:1: warning: no return statement in function returning non-void [-Wreturn-type]
 }
 ^
/home/aitor/Arduino/libraries/homie-esp8266-develop/src/Homie/Boot/BootConfig.cpp: In member function 'void HomieInternals::BootConfig::_proxyHttpRequest(AsyncWebServerRequest*)':
/home/aitor/Arduino/libraries/homie-esp8266-develop/src/Homie/Boot/BootConfig.cpp:285:40: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
   for (int i = 0; i < request->headers(); i++) {
                                        ^
/home/aitor/Arduino/libraries/homie-esp8266-develop/src/Homie/Boot/BootConfig.cpp: In member function 'void HomieInternals::BootConfig::_onDeviceInfoRequest(AsyncWebServerRequest*)':
/home/aitor/Arduino/libraries/homie-esp8266-develop/src/Homie/Boot/BootConfig.cpp:339:32: warning: comparison with string literal results in unspecified behaviour [-Waddress]
     if (iSetting->getType() != "unknown") {
                                ^
/home/aitor/Arduino/libraries/homie-esp8266-develop/src/Homie/Boot/BootConfig.cpp: In static member function 'static void HomieInternals::BootConfig::__parsePost(AsyncWebServerRequest*, uint8_t*, size_t, size_t, size_t)':
/home/aitor/Arduino/libraries/homie-esp8266-develop/src/Homie/Boot/BootConfig.cpp:425:41: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
     void* buff = request->_tempObject + index;
                                         ^
/home/aitor/Arduino/libraries/homie-esp8266-develop/src/Homie/Boot/BootConfig.cpp:428:43: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
       void* buff = request->_tempObject + total;
                                           ^
/home/aitor/Arduino/libraries/homie-esp8266-develop/src/Homie/Boot/BootConfig.cpp: At global scope:
/home/aitor/Arduino/libraries/homie-esp8266-develop/src/Homie/Boot/BootConfig.cpp:433:21: warning: 'const String ConfigJSONError(String)' defined but not used [-Wunused-function]
 static const String ConfigJSONError(const String error) {
       
Low priority feature request, take your time! Thank you!

timpur, 2018-03-19:
@bodiroga  can you test my dev branch for warnings ?

bodiroga, 2018-03-19:
Perfect @timpur, no more warning messages! You are really fast!

timpur, 2018-03-19:
Again thanks for testing.

stritti, 2019-05-07:
@bodiroga could we close this issue?

closed 2019-05-07 by timpur