homieiot/homie-esp8266 #456: Node property settable callback, call method in class?

tripflex, 2017-12-21:
Forgive me for my C++ knowledge is fairly limited, but spent the past 3-4 hours trying to figure this out with no results ... i've basically created a custom node class and wanted to be able to handle setting the properties from inside the same class, but am unable to do so.

Here's what I was trying to do:

```c++
CustomNode::CustomNode(const char* name)
    : HomieNode(name, "MisterNode") {
             advertise("on").settable( handleOn ); // Does not work
    }

bool MisterNode::handleOn(const HomieRange& range, const String& value){
    // Handle On property
    return true; // or false 
}

void MisterNode::setup() {

    // Also tried doing this inside the setup method as well, but does not work either
    // advertise("on").settable( handleOn ); 

}
```

Now I can do this using the Node handler, but I was trying to separate each property (as there are a few of them), into separate methods to handle each one

http://marvinroger.github.io/homie-esp8266/docs/2.0.0-beta.2/advanced-usage/input-handlers/

Can someone please help me with this?  I don't think i'll have any hair left by the end of the day if I can't figure something out.  Thank you!!!

euphi, 2017-12-21:
I haven't tried, but you could use a lambda function for this, e.g. something like this:

```C++
advertise("on").settable([this](const HomieRange& range, const String& value) {this->handleOn(range,value);});
```

timpur, 2017-12-22:
confirmed 
something like this should work (compiles)
http://en.cppreference.com/w/cpp/language/lambda
`
[this](const HomieRange& range, const String& value) -> bool{
		return true;
	}
`

capturing this (`[this]`) means you can use this: `this->function()`

also looks like you class names are inconsistent ?
`CustomNode::CustomNode`
`MisterNode::handleOn`

euphi, 2017-12-22:
@tripflex  I forgot the return value (Thats the `-> bool` part in @timpur 's explanation.

timpur, 2017-12-22:
@tripflex please close if all is working, thanks.

tripflex, 2017-12-26:
Awesome, thanks for all the help @euphi and @timpur

closed 2017-12-26 by tripflex