homieiot/homie-esp8266 #457: Saving custom setting from device

tripflex, 2017-12-22:
As mentioned in my other issues/comments, I have a custom node that relies on a custom config value that is stored by Homie.

The issue is that, I also have exposed this setting in a node property, to allow configuring that one property via MQTT, but can't find any way to save a custom setting value (without using incremental json to do so).

Is there no way to set a custom setting value and have that save to the config file ... from inside my code base?  IF not, i'll just have to use a lib I created from my old project to save json values to a file ... would just be nicer to be able to save that value in the same homie config to not have to use multiple configuration files.

Case and point, I have a RelayNode (which controls a pump), and I have a custom setting setup for  `runDuration` ... this `runDuration` is also exposed as a property of the node, ie:
`/homie/test-device/pump/duration` which is settable, and when that value is set, I need it to update the `runDuration` saved to the HomieConfiguration, that way that value that was set is used instead of the default value.

Yes when the device connects back via MQTT and it subscribes it will receive the last set value, but if the device does not have internet, I still need it to be able to run and use the last set value, thus, the reason I need to be able to save the value to the configuration.

Is this something that would be possible already with the existing Homie codebase, or do I need to just go ahead and setup another way to handle saving my own custom configurations?

timpur, 2017-12-23:
Im working on alot to enable using settings better with homie, ill add this to the list ( #458 )

tripflex, 2017-12-26:
@timpur awesome 👍  thank you!!

timpur, 2017-12-29:
Added this to my fork. You can test it if you like

tripflex, 2018-01-19:
Yup I noticed that, i've actually been using your 2.1 branch fork for a while now, haven't had a chance to test this part yet, but will do so here this weekend and let you know

timpur, 2018-01-20:
@tripflex
Awsome, on hols for a week so will be smashing out some homie stuff while I have a bit more time... So report issues to me if they arise about V2.1

tripflex, 2018-01-20:
@timpur will do .. is the plan to keep the `/wifi/connect` as `POST`?  Reason I ask I create a Meteor plugin to handle Homie API (and anything else I can add later) and noticed you changed it to POST in your fork:
https://github.com/tripflex/meteor-homie

timpur, 2018-01-20:
Yes, I believe put should be for putting a file on the server and not setting a setting. So I think post makes more sense here